MichaIng/DietPi #1401: Cannot Install Oracle JDK 8

aaro668, 2018-01-16:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | 159
- SBC Device :Orange Pi Zero
- Power supply used : USB
- SD card used Kingston 16GB
- Distro  | Linux DietPiOrange 3.4.113-sun8i #2 SMP PREEMPT Fri Mar 10 18:24:50 GMT 2017 armv7l GNU/Linux

#### Additional Information (if applicable):
- Software title: Oracle JDK 8
- Can this issue be replicated on a fresh installation of DietPi? yes

#### Expected behaviour:
I am trying to do automated Install of Dietpi on Orange Pi Zero, with node-red, Mosquitto and Blynk server set up to install automatically.
Here is the dietpi.txt file that I used:

`#Modifications to /boot/* files, will not be saved when powered on.
#Please ensure you edit from the DietPi-RamDisk location /DietPi/*

#------------------------------------------------------------------------------------------------------
# D I E T - P I
# 1st run settings
# All settings below will be applied on the 1st boot of DietPi, ONCE.
#------------------------------------------------------------------------------------------------------
# >> Networking Options -----------------------------
#If both Ethernet and Wifi are enabled, Wifi will take priority and Ethernet will be disabled.
#   1=enabled
Ethernet_Enabled=0
Wifi_Enabled=1

#Enter your Wifi details below, if applicable (Case Sensitive).
Wifi_SSID=AaronsNetwork
Wifi_KEY=electricity
#	available | WPA-PSK / NONE | WEP=No longer supported
Wifi_KEYMGR=WPA-PSK

#Enter your Static Network details below, if applicable.
Use_Static=1
Static_IP=192.168.0.14
Static_Mask=255.255.255.0
Static_Gateway=192.168.0.33
Static_DNS=8.8.8.8

#Hostname
Hostname=DietPi

#Force ethernet speeds
#NB: This is mainly aimed towards Pine A64's which may have a HW issue that causes unstable 1Gbit link.
#	0=automatic speed | 10 = 10mbit, 100 = 100mbit etc
Ethernet_Force_Speed=0

# >> Misc Options -----------------------------------
#Size of dphys-swapfile to generate (MB)
#   0=Disabled | 1=auto (1GB-RAM = size) | 2+=manual
Swapfile_Size=1567

# >> Automation Options -----------------------------
#NB: Following options are only executed on 1st run of DietPi.

#Fully automate installation
#   1=Automated installation with no user inputs.
AUTO_Install_Enable=1

#Global Password to be applied for the system
#  Affects user "root" and "dietpi" login passwords, and, all software installed by dietpi-software, that requires a password
#  eg: MySQL, Transmission, Deluge etc.
#  WARN: Passwords with the any of the following characters are not supported: \"$
#  WARN: Do NOT change this entry, after 1st run setup of DietPi has been completed. It is always scraped by dietpi-software.
AUTO_Global_Password=dietpi
[... 237 lines cut ...]

download failed
Oracle JDK 8 is NOT installed.
dpkg: error processing package oracle-java8-installer (--configure):
 subprocess installed post-installation script returned error exit status 1
Errors were encountered while processing:
 oracle-java8-installer
E: Sub-process /usr/bin/dpkg returned an error code (1)

 [Info] Please try:
 [Info] - Fixing apt package errors with 'apt-get install -f'
 [Info] - Change apt mirror: 'dietpi-config' > 'Networking: NAS/MISC' > 'Apt Mirror'
 [Info] - Clear and refresh apt 'apt-get update'
 [Info] - If problems persist, please create a Git ticket so DietPi can investigate the issue:
 - https://github.com/Fourdee/DietPi/issues
`

#### Steps to reproduce:
Flash image of dietpi for orange pi zero onto fresh SD card with my dietpi.txt file in root directory

#### Did you submit a ```dietpi-bugreport```?
No

#### Extra details:
I am having a hard time getting dietpi installed on any of my SBC's with node-red and Mosquitto as well as local Blynk Server, it seems there are errors all along and spend the whole day with no luck...Does anyone have a simple foolproof script to get node-red, mosquitto and blynk server installed onto dietpi??

MichaIng, 2018-01-17:
@aaro668 
- We use this PPA: https://launchpad.net/~webupd8team and installation breaks before we do any configuration/adjustments.
- The installer tries to download the SDK from oracle.com, but gets redirected several times and finally: `HTTP request sent, awaiting response... 404 Not Found` breaks it.
- https://www.oracle.com/index.html got some heavy optical rework since the last time I saw it. Maybe also the internal download paths were changed, causing the issue. In this case we would need to wait for a fix by `“WebUpd8” team`. Although I don't see open bugs on their launchpad or a possibility to add a bug report: https://launchpad.net/~webupd8team
- I will try to replicate tomorrow.

aaro668, 2018-01-17:
@MichaIng 
OK, so this is basically either a problem with Oracle download page or WebUpd8?
Are there alternative ways to getting these 3 packages installed in dietpi? (node-red, blynk server and mosquitto)

aaro668, 2018-01-17:
I tried installing only blynk server tonight, and got the error again:
[image]

and here is the log:

`[Info] Apt Log:
Setting up oracle-java8-installer (8u151-1~webupd8~0) ...
Using wget settings from /var/cache/oracle-jdk8-installer/wgetrc
Downloading Oracle Java 8...
--2018-01-16 21:21:14--  http://download.oracle.com/otn-pub/java/jdk/8u151-b12/e758a0de34e24606bca991d704f6dcbf/jdk-8u151-linux-arm32-vfp-hflt.tar.gz
Resolving download.oracle.com (download.oracle.com)... 23.56.202.80
Connecting to download.oracle.com (download.oracle.com)|23.56.202.80|:80... connected.
HTTP request sent, awaiting response... 302 Moved Temporarily
Location: https://edelivery.oracle.com/otn-pub/java/jdk/8u151-b12/e758a0de34e24606bca991d704f6dcbf/jdk-8u151-linux-arm32-vfp-hflt.tar.gz [following]
--2018-01-16 21:21:20--  https://edelivery.oracle.com/otn-pub/java/jdk/8u151-b12/e758a0de34e24606bca991d704f6dcbf/jdk-8u151-linux-arm32-vfp-hflt.tar.gz
Resolving edelivery.oracle.com (edelivery.oracle.com)... 104.93.179.190, 2600:140a:0:38c::2d3e, 2600:140a:0:382::2d3e
Connecting to edelivery.oracle.com (edelivery.oracle.com)|104.93.179.190|:443... connected.
HTTP request sent, awaiting response... 302 Moved Temporarily
Location: http://download.oracle.com/otn-pub/java/jdk/8u151-b12/e758a0de34e24606bca991d704f6dcbf/jdk-8u151-linux-arm32-vfp-hflt.tar.gz?AuthParam=1516155801_dfe8adb3289e8695a230f94b67ffb4b1 [following]
--2018-01-16 21:21:21--  http://download.oracle.com/otn-pub/java/jdk/8u151-b12/e758a0de34e24606bca991d704f6dcbf/jdk-8u151-linux-arm32-vfp-hflt.tar.gz?AuthParam=1516155801_dfe8adb3289e8695a230f94b67ffb4b1
Connecting to download.oracle.com (download.oracle.com)|23.56.202.80|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2018-01-16 21:21:23 ERROR 404: Not Found.

download failed
Oracle JDK 8 is NOT installed.
dpkg: error processing package oracle-java8-installer (--configure):
 subprocess installed post-installation script returned error exit status 1
Errors were encountered while processing:
 oracle-java8-installer
E: Sub-process /usr/bin/dpkg returned an error code (1)

 [Info] Please try:
 [Info] - Fixing apt package errors with 'apt-get install -f'
 [Info] - Change apt mirror: 'dietpi-config' > 'Networking: NAS/MISC' > 'Apt Mirror'
 [Info] - Clear and refresh apt 'apt-get update'
 [Info] - If problems persist, please create a Git ticket so DietPi can investigate the issue:
 - https://github.com/Fourdee/DietPi/issues
`

Being that this is software that is supposed to be optimized to run on dietpi and should "just work" then I consider this a dietpi issue??

MichaIng, 2018-01-17:
@aaro668
> OK, so this is basically either a problem with Oracle download page or WebUpd8?

Yes looks like, especially if other apt installations work fine. Temporarily we can try to find an alternative source for Java 8 SDK, e.g. Raspbian had it's own package included.

But I need to sleep now. Will try to replicate tomorrow and see if I find a solution.

gotttlieb, 2018-01-17:
I'am in the same situation
``` DietPi-Software
─────────────────────────────────────────────────────
 Mode: Install software
 Please wait...

 ALSA: linux sound system
 [Info] Please wait...

 #               processing | D i e t  P i             #

 [Info] APT is processing, please wait...

Configurando oracle-java8-installer (8u151-1~webupd8~0) ...
Using wget settings from /var/cache/oracle-jdk8-installer/wgetrc
Downloading Oracle Java 8...
--2018-01-17 20:07:20--  http://download.oracle.com/otn-pub/java/jdk/8u151-b12/e758a0de34e24606bca991d704f6dcbf/jdk-8u151-linux-arm32-vfp-hflt.tar.gz
Resolviendo download.oracle.com (download.oracle.com)... 104.104.142.192
Conectando con download.oracle.com (download.oracle.com)[104.104.142.192]:80... conectado.
Petición HTTP enviada, esperando respuesta... 302 Moved Temporarily
Localización: https://edelivery.oracle.com/otn-pub/java/jdk/8u151-b12/e758a0de34e24606bca991d704f6dcbf/jdk-8u151-linux-arm32-vfp-hflt.tar.gz [siguiendo]
--2018-01-17 20:07:20--  https://edelivery.oracle.com/otn-pub/java/jdk/8u151-b12/e758a0de34e24606bca991d704f6dcbf/jdk-8u151-linux-arm32-vfp-hflt.tar.gz
Resolviendo edelivery.oracle.com (edelivery.oracle.com)... 104.83.55.184, 2a02:26f0:129:397::2d3e, 2a02:26f0:129:392::2d3e
Conectando con edelivery.oracle.com (edelivery.oracle.com)[104.83.55.184]:443... conectado.
Petición HTTP enviada, esperando respuesta... 302 Moved Temporarily
Localización: http://download.oracle.com/otn-pub/java/jdk/8u151-b12/e758a0de34e24606bca991d704f6dcbf/jdk-8u151-linux-arm32-vfp-hflt.tar.gz?AuthParam=1516216160_7cbc4fca3f0596208f6c147e4c69b986 [siguiendo]
--2018-01-17 20:07:20--  http://download.oracle.com/otn-pub/java/jdk/8u151-b12/e758a0de34e24606bca991d704f6dcbf/jdk-8u151-linux-arm32-vfp-hflt.tar.gz?AuthParam=1516216160_7cbc4fca3f0596208f6c147e4c69b986
Conectando con download.oracle.com (download.oracle.com)[104.104.142.192]:80... conectado.
Petición HTTP enviada, esperando respuesta... 404 Not Found
2018-01-17 20:07:21 ERROR 404: Not Found.

download failed
Oracle JDK 8 is NOT installed.
dpkg: error al procesar el paquete oracle-java8-installer (--configure):
 el subproceso instalado el script post-installation devolvió el código de salida de error 1
Se encontraron errores al procesar:
 oracle-java8-installer
E: Sub-process /usr/bin/dpkg returned an error code (1)
 [Failed] The apt cache may be corrupt, apt mirror offline, or you have held broken packages. DietPi-Software will now exit.

 [Info] Apt Log:
Configurando oracle-java8-installer (8u151-1~webupd8~0) ...
Using wget settings from /var/cache/oracle-jdk8-installer/wgetrc
Downloading Oracle Java 8...
--2018-01-17 20:07:20--  http://download.oracle.com/otn-pub/java/jdk/8u151-b12/e758a0de34e24606bca991d704f6dcbf/jdk-8u151-linux-arm32-vfp-hflt.tar.gz
Resolviendo download.oracle.com (download.oracle.com)... 104.104.142.192
Conectando con download.oracle.com (download.oracle.com)[104.104.142.192]:80... conectado.
Petición HTTP enviada, esperando respuesta... 302 Moved Temporarily
Localización: https://edelivery.oracle.com/otn-pub/java/jdk/8u151-b12/e758a0de34e24606bca991d704f6dcbf/jdk-8u151-linux-arm32-vfp-hflt.tar.gz [siguiendo]
--2018-01-17 20:07:20--  https://edelivery.oracle.com/otn-pub/java/jdk/8u151-b12/e758a0de34e24606bca991d704f6dcbf/jdk-8u151-linux-arm32-vfp-hflt.tar.gz
Resolviendo edelivery.oracle.com (edelivery.oracle.com)... 104.83.55.184, 2a02:26f0:129:397::2d3e, 2a02:26f0:129:392::2d3e
Conectando con edelivery.oracle.com (edelivery.oracle.com)[104.83.55.184]:443... conectado.
Petición HTTP enviada, esperando respuesta... 302 Moved Temporarily
Localización: http://download.oracle.com/otn-pub/java/jdk/8u151-b12/e758a0de34e24606bca991d704f6dcbf/jdk-8u151-linux-arm32-vfp-hflt.tar.gz?AuthParam=1516216160_7cbc4fca3f0596208f6c147e4c69b986 [siguiendo]
--2018-01-17 20:07:20--  http://download.oracle.com/otn-pub/java/jdk/8u151-b12/e758a0de34e24606bca991d704f6dcbf/jdk-8u151-linux-arm32-vfp-hflt.tar.gz?AuthParam=1516216160_7cbc4fca3f0596208f6c147e4c69b986
Conectando con download.oracle.com (download.oracle.com)[104.104.142.192]:80... conectado.
Petición HTTP enviada, esperando respuesta... 404 Not Found
2018-01-17 20:07:21 ERROR 404: Not Found.

download failed
Oracle JDK 8 is NOT installed.
dpkg: error al procesar el paquete oracle-java8-installer (--configure):
 el subproceso instalado el script post-installation devolvió el código de salida de error 1
Se encontraron errores al procesar:
 oracle-java8-installer
E: Sub-process /usr/bin/dpkg returned an error code (1)```

Already tryed to chanfe the APT Mirror

MichaIng, 2018-01-17:
@aaro668 @gotttlieb 
Jep, I face exactly the same on following the official installation guide of the PPA maintainers: http://www.webupd8.org/2014/03/how-to-install-oracle-java-8-in-debian.html

Thus it's no DietPi issue and should be addressed to http://www.webupd8.org/.

Fourdee, 2018-01-18:
@MichaIng 

> Was my guess right, that using the mentioned PPA is because of outdated Java version on Debian repo? As mentioned, we can use jessie-backports, even version 9 is available on buster+stretch-backports: https://packages.debian.org/de/stretch-backports/openjdk-9-jdk. Or was the reason indeed another?

We use the PPA to install Oracle Java, which is proprietary. OpenJDK is not.
The reason for this, was me assuming the proprietary software should perform better/faster, than the open source version, however turns out I may of been wrong:
https://stackoverflow.com/a/38685948

So yep, using openjdk via APT sounds like a better solution.

aaro668, 2018-01-18:
@MichaIng 
Ok, so when I try to install the software (blynk server) from dietpi-software, I got the error....
Should I install the backport(apt install -t jessie-backports openjdk-8-jdk) before installing the blynk server via dietpi-software and it should then work without the error?

aaro668, 2018-01-18:
just a FYI:
I am also getting this error on fresh install of dietpi on OPi zero:
` 
DietPi-Software
─────────────────────────────────────────────────────
 Mode: Install software
 Please wait...

 Build-Essentials: common packages for compile
 [Info] Please wait...

 #               processing | D i e t  P i             #

 [Info] APT is processing, please wait...

Extracting templates from packages: 100%
Preconfiguring packages ...
(Reading database ... 16105 files and directories currently installed.)
Preparing to unpack .../libc6_2.19-18+deb8u10_armhf.deb ...

LD_LIBRARY_PATH contains the traditional /lib directory,
but not the multiarch directory /lib/arm-linux-gnueabihf.
It is not safe to upgrade the C library in this situation;
please remove the /lib/directory from LD_LIBRARY_PATH and
try again.

dpkg: error processing archive /var/cache/apt/archives/libc6_2.19-18+deb8u10_armhf.deb (--unpack):
 subprocess new pre-installation script returned error exit status 1
Errors were encountered while processing:
 /var/cache/apt/archives/libc6_2.19-18+deb8u10_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
 [Failed] The apt cache may be corrupt, apt mirror offline, or you have held broken packages. DietPi-Software will now exit.

 [Info] Apt Log:
Extracting templates from packages: 100%
Preconfiguring packages ...
(Reading database ... 16105 files and directories currently installed.)
Preparing to unpack .../libc6_2.19-18+deb8u10_armhf.deb ...

LD_LIBRARY_PATH contains the traditional /lib directory,
but not the multiarch directory /lib/arm-linux-gnueabihf.
It is not safe to upgrade the C library in this situation;
please remove the /lib/directory from LD_LIBRARY_PATH and
try again.

dpkg: error processing archive /var/cache/apt/archives/libc6_2.19-18+deb8u10_armhf.deb (--unpack):
 subprocess new pre-installation script returned error exit status 1
Errors were encountered while processing:
 /var/cache/apt/archives/libc6_2.19-18+deb8u10_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

 [Info] Please try:
 [Info] - Fixing apt package errors with 'apt-get install -f'
 [Info] - Change apt mirror: 'dietpi-config' > 'Networking: NAS/MISC' > 'Apt Mirror'
 [Info] - Clear and refresh apt 'apt-get update'
 [Info] - If problems persist, please create a Git ticket so DietPi can investigate the issue:
 - https://github.com/Fourdee/DietPi/issues
`

I am able to fix it like this:

echo $LD_LIBRARY_PATH
unset LD_LIBRARY_PATH
echo $LD_LIBRARY_PATH
ldconfig -v

And then running dietpi-software again

MichaIng, 2018-01-18:
@Fourdee 
Ah yes I see. The current issues with the Oracle JDK PPA might be temporary, but as of your link, it might be at least not worse, to switch to OpenJDK and it would be easier to handle/install.

@aaro668 
Ah yeah sorry, was not thinking too far. You can install OpenJDK by the method I provided, but of course DietPi-Software, when choosing dependant software, still expects Oracle JDK to be installed/tries to do this. We need to adjust this behaviour. This can be done for v6.0 release, but we decided to only provide this major update as new image. So depending on how urgent your request is and how willing you are to flash the new image, once released, I can start to implement the adjustment and then provide the hopefully not too vast changes you need to do to `dietpi-software` manually as temporary fix.

aaro668, 2018-01-18:
@MichaIng 

I am basically on a quest to try and find the best/fastest way for setting up a minimal OS on an Orange Pi Zero, running Blynk Server, Node-red and Mosquitto MQTT broker.
This is a handy/cheap/small and flexible board with ethernet and wifi connectivity,I think the potential for IOT applications is enormous.

I have a few boards here that I am just playing around with to try and find the best way to get these 3 softwares running.
Once you get V6 released please do let me know, then I can test the image on a board here and see how it goes!

Thanks for all your help!

MichaIng, 2018-01-18:
@aaro668 
This is great news. As we made huge/deep changes with v6.0, intensive testing/willing testers are highly appreciated.

I made a first PR for the Oracle JDK => OpenJDK switch: https://github.com/Fourdee/DietPi/pull/1405
Most of our software seem to get the related binaries/lib paths by themself, e.g. via `which java`, but testing is needed.
- On Blynk installation I just faced another issue which might be related to v6.0 changes:
```
gyp WARN EACCES user "root" does not have permission to access the dev dir "/usr/local/lib/node_modules/onoff/node_modules/epoll/.node-gyp/9.4.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/usr/local/lib/node_modules/onoff/node_modules/epoll/.node-gyp"
```
- Bizarre as root of course has full root permissions... However manual `npm install -g onoff` worked well and allowed me to finish installation afterwards.
- But Blynk server does not start up because of not recent enough Java version, if I get the log right. Sadly seems to be not solved for Jessie that easy. Even the backport is on Java `8u131`, where on Stretch it's `8u151`
- We'll see how to solve it.

aaro668, 2018-01-18:
@MichaIng 

I have figured out how to install the BlynkServer manually:
#1 Download this file to "/mnt/dietpi_userdata/downloads": https://github.com/blynkkk/blynk-server/releases/download/v0.29.7/server-0.29.7-java8.jar
#2 Run these commands: 
sudo tar zxvf jdk-8u161-linux-arm32-vfp-hflt.tar.gz -C /opt
sudo update-alternatives --install /usr/bin/javac javac /opt/jdk1.8.0_161/bin/javac 1
sudo update-alternatives --install /usr/bin/java java /opt/jdk1.8.0_161/bin/java 1
sudo update-alternatives --config javac
sudo update-alternatives --config java
#3 Make a directory: /mnt/dietpi_userdata/Blynk
cd /mnt/dietpi_userdata/Blynk
wget "https://github.com/blynkkk/blynk-server/releases/download/v0.29.7/server-0.29.7-java8.jar"
copy mail.properties and server.properties files into "/mnt/dietpi_userdata/Blynk/"         directory

#4 Set up to run at boot:
sudo nano /etc/rc.local

Add the following command mentioned before "exit" 
java -jar /mnt/dietpi_userdata/Blynk/server-0.29.7-java8.jar -dataFolder /mnt/dietpi_userdata/Blynk -serverConfig /mnt/dietpi_userdata/Blynk/server.properties -mailConfig /mnt/dietpi_userdata/Blynk/mail.properties &

sudo reboot

when it has rebooted the blynk server should be running

MichaIng, 2018-01-18:
@aaro668 
Many thanks for testing. Yes we somehow missed to switch to the Java 8 Blynk version. I also got it running now, also there was a typo, leading to en empty server.properties, where the actually intention is to pull it also from Blynk github. Fixed with: https://github.com/Fourdee/DietPi/pull/1406

Service runs, web ui opens at `https://<yourURL/IP>:9443/admin`. I just cannot access configuration page, see PR.

aaro668, 2018-01-18:
@MichaIng 
I cannot access config page either, but I don't really need it, and have never used so I am not sure if it is even needed?
Possibly get someone involved that has experience with local Blynk server to help with this.

gotttlieb, 2018-01-18:
Hi @MichaIng first of all thanks for your efforts!
To me it's happen the same, I was able to intall the OpenJDK, but for me it's failing the instalation of packages via dietpi-software as you mention.
The thing is, whe can know witch dependencies are needed? I take a look in the Forum but they are not listed.

I take a look at you PR, but i see that is only for blick, can you provide the file path to check it myself?
Thanks!

MichaIng, 2018-01-18:
@gotttlieb 
- For some software titles the dependencies/installation method also depends on your machine and Debian/Raspbian version.
- Easiest I guess is, if you adjust your `/DietPi/dietpi/dietpi-software` software according to this PR: https://github.com/Fourdee/DietPi/pull/1405/files
- If you don't need Tomcat and Blynk, you just need to adjust the ~middle part where it's about the actual Java installation. The file lines do not match, in your version of dietpi-software it should be the part after line 8325: https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L8237-L8258

MichaIng, 2018-01-18:
@aaro668 
Yeah, I guess the config page error is not related to Java anymore. I just asked at their github page, someone else had the same error on some sub page of config and there the info was, that web ui is under rework: https://github.com/blynkkk/blynk-server/issues/658#issuecomment-358778423

gotttlieb, 2018-01-18:
OK, I manage to get something "working"  ( at least no errors apeared in my sistem)

Next Snipet is Tested in **Odroid XU 4, use under your own risk. **
First i executed as @MichaIng advised:
```dpkg -P oracle-java8-installer && apt-get autoremove --purge```
And replaced next lines are replaced in the  **/DietPi/dietpi/dietpi-software** file.
Remove Line number before pasting. **DO NOT COPYPAST WITHOUT EDIT, DO A BACKUP FIRST**

```8237                         # - Raspbian
 8238                         if (( $HW_MODEL < 10 )); then
 8239
 8240                                 AGI openjdk-8-jdk
 8241
 8242                         # - Debian. Add repo
 8243                         else
 8244                                 #Add Manual install of Jdk
 8245                                 apt install -t jessie-backports openjdk-8-jdk
 8246                                 AGI openjdk-8-jdk
 8247
 8248                         fi ```

MichaIng, 2018-01-18:
@gotttlieb 
Ah sorry forgot, that the `G_` commands/variables are not yet available in v159 😅.
So this should work then as expected on every system:
```
			if (( $DISTRO == 3 )); then

				cat << _EOF_ > /etc/apt/preferences.d/99-dietpi-openjdk-8-jdk
Package: openjdk-8-jdk
Pin: release a=jessie-backports
Pin-Priority: 990
_EOF_

				AGI openjdk-8-jdk -t jessie-backports

			else

				AGI openjdk-8-jdk

			fi
```
- AGI is just a wrapper of `apt-get install` with some automation options, so `apt install` and `AGI` is doubled in your snippet.
- Important on Jessie (`$DISTRO == 3` means Debian/Raspbian Jessie) is that it pulls the package and it's dependencies from backports: `-t jessie-backports`
- The `preferences.d` addition allows apt(-get) (dist-)upgrade to pull upgrades from backports as well.
- No check for `$HW_MODEL` necessary anymore, as we now handle it similar for all machines.

MichaIng, 2018-01-22:
Mark as closed, as fixed with v6.0 and manual fix for v159 provided.

closed 2018-01-22 by MichaIng