MichaIng/DietPi #1408: Retropie?

symbios24, 2018-01-20:
Hello, there i`m using latest Dietpi v158 in a Raspberry Pi 3 but i cannot see the Retropie in the install software menu, is Retropie removed?

Thanks

symbios24, 2018-01-20:
ps. i tried both DietPi jessie and stretch its not there in any of them

Fourdee, 2018-01-20:
@symbios24 
Hi,

We removed Retro-pie from DietPi a while back. Reasons for this:
- RetroPie expects, and, is developed based on the user having a stock Raspbian image system for installation.
- Constant Retro-pie devolpment is impossible for us to maintain. We had to provide many fixes and changes over the course of its availability in DietPi.
- We are a very small team, with 150+ software titles and 20+ devices to maintain.

As much as I would love to have Retro-pie available in DietPi (I am a retro gamer), its simply not fee-sable at this time, to achieve true justice with DietPi + Retro-pie.

Regardless, the official Retro-pie installation method should work, if you experience any issues, post here and we'll try to assist.

closed 2018-01-20 by Fourdee

symbios24, 2018-01-21:
ok thanks

i have some other ideas for extra software on DietPi if you are interested

JDownloader Server
a simple minecraft server without gui just the normal minecraft server
assaultcube server
openarena game
the battle for wesnoth game
minetest server
manaplus client
chromium browser OS (just the chromium browser standalone so you can run it on startup)
terraria game server
screenly ose

symbios24, 2018-02-05:
i still believe dietpi is 10 times faster/lighter than the raspbian lite that retropie has, dietpi size 600mb raspbian lite size 4Gb its obvious which one is lighter, it will be great to include retropie in the gaming section anyway.