MichaIng/DietPi #1410: DietPi Software | Amiberry next version

midwan, 2018-01-20:
## Creating a bug report/issue:
Prepare DietPi for the upcoming release of Amiberry

#### Required Information:
- SBC Device: RPI1/2/3, XU4, OrangePi (maybe more can run it, but those are confirmed for now)

#### Additional Information (if applicable):
- Extra lib dependencies to support CD32 FMV
- SDL2, SDL2_image, SDL2_ttf (compiled from source to support KMS!)
- SDL1 version is also supported, both SDL1 and SDL2 build from the same Makefile
- new file dependencies included in the data/ folder
- No need to launch from under X11 anymore

#### Extra details:
Full list of required packages to build from source:
`libxml2-dev libflac-dev libmpg123-dev libpng-dev libmpeg2-4-dev`

Additionally, for the SDL1 version:
`libsdl-dev libguichan-dev libsdl-ttf2.0-dev libsdl-gfx1.2-dev`

Or for the SDL2 version:
- SDL2-2.0.x compiled from source to support the KMSDRM driver (or RPI driver)
- SDL2_ttf
- SDL2_image

I wrote a wiki on how to configure/compile SDL2 from source here: https://github.com/midwan/amiberry/wiki/Compile-SDL2-from-source

Fourdee, 2018-01-20:
@midwan 

Excellent, see you've been very busy with Git and great to see this project evolve. 
Many thanks for the info. I'll take a look at this over the next few days. Should be able to get it in for DietPi v6.0 release if all goes to plan 😄 

ref: https://github.com/Fourdee/DietPi/issues/1385

Fourdee, 2018-01-21:
Notes:
- ARMv8 (Odroid C2) Jessie:
```
root@DietPi:~#  apt-get install gir1.2-ibus-1.0 libdbus-1-dev libegl1-mesa-dev libibus-1.0-5 libibus-1.0-dev libice-dev libsm-dev libsndio-dev libwayland-bin libwayland-dev libxi-dev libxinerama-dev libxkbcommon-dev libxrandr-dev libxss-dev libxt-dev libxv-dev x11proto-randr-dev x11proto-scrnsaver-dev x11proto-video-dev x11proto-xinerama-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package libwayland-bin
```

midwan, 2018-01-21:
We don't really use wayland anyway, so this shouldn't be a problem. The list is what I got if I tried to install the package from the official repo (which you normally don't want, because their binaries are older and don't have the RPI or KMSDRM drivers enabled)

Fourdee, 2018-01-21:
@midwan 

```./configure --disable-video-rpi --disable-video-x11 --enable-video-kmsdrm; make -j $(nproc --all); sudo make install```

```make -j $(nproc --all)``` compiles using all available CPU cores, much quicker 😃

midwan, 2018-01-21:
yeah, I usually specify the number of cores manually  here, but I noticed that the Pi can easily overheat if using all 4, and then it throttles down (so it becomes sloweeeeeer). I found that using up to 3 is ok.

Fourdee, 2018-01-21:
@midwan 

> yeah, I usually specify the number of cores manually here, but I noticed that the Pi can easily overheat if using all 4, and then it throttles down (so it becomes sloweeeeeer). I found that using up to 3 is ok.

Good point, she does run a little hot lol 👍 

Want me to revert the changes: https://github.com/midwan/amiberry/wiki/Compile-SDL2-from-source ?

Fourdee, 2018-01-21:
> ...and SDL2_ttf:

```
configure: error:
*** Unable to find FreeType2 library (http://www.freetype.org/)
```

🈯️ ```libfreetype6-dev``` pre-req
```
apt-get install libfreetype6-dev -y
```

Fourdee, 2018-01-26:
@midwan 

Apologies, will need to nudge this to v6.1 release. We have too many outstanding items for v6.0 release on Sunday.

In the mean time, i'll create a new AmiBerry image, which will use the existing installation. I'll also update you with the details once its completed as a few things have changed in DietPi (eg: automation options).

midwan, 2018-01-26:
@Fourdee 
No worries, we're also busy with a few things :)

midwan, 2018-01-30:
@Fourdee 
We also have a new target: ASUS Tinker board.
However, it's still a little tricky to get it working there, as you probably noticed also. Due to the lack of proper support, we ended up compiling our own kernel (based on Armbian), using the GPU blobs and several other modifications to get SDL2 to compile with support for OpenGL ES2.

The Makefile includes the target, and the code has the necessary logic to accommodate for it (e.g. different timings, custom cursor workaround for Mali GPUs, etc). We can also use multi-threaded rendering in SDL2, which improves the performance a lot.

Sadly, multi-threaded rendering does not work on the current implementation of the KMSDRM backend for the Pi. It does work if you're running under x11 only (and you've compiled SDL2 without disabling x11 of course). So the fastest target for the Pi is still using Dispmanx directly.

Fourdee, 2018-02-01:
@midwan 

Thanks for the info 👍  

I'll try and pick this back up over next few days, got side tracked with bug fixes etc.

midwan, 2018-02-01:
@Fourdee 
I've asked the person who worked on the port to get in touch with you about using the custom kernel and extra modifications used on the Tinkerboard, since we're using DietPi as a basis now :)

I think getting GPU support on the Tinkerboard using DietPi would be beneficial to all...

Fourdee, 2018-02-02:
@midwan 
> I think getting GPU support on the Tinkerboard using DietPi would be beneficial to all...

Agree.

We'll focus on RPi 1st, then Odroid and ASUSTB.

midwan, 2018-02-02:
@Fourdee 
BTW, here's the repo with the kernel used, config and info about the extra stuff needed (gpu blobs etc):
https://github.com/alynna/tinker-kdev

Fourdee, 2018-02-03:
@midwan 

#### Testing previous and dev AmiBerry binaries on RPi3 Stretch, with SDL2 installed as per your guide.
```
INFO: SDL could not initialize! SDL_Error: No Available video device
```
Both with/without ```xinit```

#### Under RPi OpenGL, using compiled SDL2 as per your guide:
- AmiBerry-sdl2 (old) is fine under ```xinit```
- AmiBerry-sdl2 (dev) only under xinit, ```MESA-LOADER: failed to retrieve device info```

---

@midwan 
```./configure --disable-video-rpi --enable-video-kmsdrm```, should this not be ```./configure  --disable-video-opengl``` only for RPi?

---

Heres what I have so far:
```
#AmiBerry compile  -----------------------------------------------------------------

#SDL2
G_AGI build-essential libfreetype6-dev gir1.2-ibus-1.0 libdbus-1-dev libegl1-mesa-dev libgl1-mesa-dev libibus-1.0-5 libibus-1.0-dev libice-dev libsm-dev libsndio-dev libwayland-dev libxi-dev libxinerama-dev libxkbcommon-dev libxrandr-dev libxss-dev libxt-dev libxv-dev x11proto-randr-dev x11proto-scrnsaver-dev x11proto-video-dev x11proto-xinerama-dev
#libwayland-bin

cd "$HOME"
wget https://www.libsdl.org/release/SDL2-2.0.7.tar.gz
tar zxvf SDL2-2.0.7.tar.gz
cd SDL2-2.0.7
./configure --disable-video-rpi --enable-video-kmsdrm #--host=arm-raspberry-linux-gnueabihf #--disable-video-opengl
G_RUN_CMD make -j $(nproc --all)
make install

cd "$HOME"
wget https://www.libsdl.org/projects/SDL_image/release/SDL2_image-2.0.2.tar.gz
tar zxvf SDL2_image-2.0.2.tar.gz
cd SDL2_image-2.0.2
mkdir build
cd build
../configure
G_RUN_CMD make -j $(nproc --all)
make install

cd "$HOME"
wget https://www.libsdl.org/projects/SDL_ttf/release/SDL2_ttf-2.0.14.tar.gz
tar zxvf SDL2_ttf-2.0.14.tar.gz
cd SDL2_ttf-2.0.14
mkdir build
cd build
../configure
G_RUN_CMD make -j $(nproc --all)
make install

#	LDconfig for /usr/local/lib, else, fails to load SDL2-ttf.so binary
echo -e "/usr/local/lib" > /etc/ld.so.conf.d/usr-local-lib.conf
ldconfig

#AmiBerry

G_AGI build-essential git \
libxml2-dev libflac-dev libmpg123-dev libpng-dev libmpeg2-4-dev
#libxml2-dev libflac-dev libmpg123-dev libpng-dev libmpeg2-4-dev libsdl-dev libguichan-dev libsdl-ttf2.0-dev libsdl-gfx1.2-dev \
#libsdl2-dev libsdl2-image-dev libsdl-image1.2-dev

cd "$HOME"
git clone https://github.com/midwan/amiberry -b dev amiberry-dev
cd amiberry-dev

make clean
G_RUN_CMD make -j $(nproc --all) all PLATFORM=rpi3-sdl2

#Run pre-req binaries
G_AGI libmpg123-0 libxml2 mpeg2dec libmpeg2-4

#Run/test binary

#AmiBerry compile  -----------------------------------------------------------------
```

midwan, 2018-02-04:
@Fourdee 
Very important question: What video driver do you have enabled in `/boot/config.txt` ?
KMSDRM will only work with **fkms** or **KMS**.

There's no need to run it under `xinit `from my tests here, but there are a few important details:

Legacy driver: **Dispmanx** can be used, but there's not OpenGL(ES) support (not that we use it in Amiberry)
FKMS driver: **Dispmanx** can be used, and there's also OpenGL(ES) support (like a hybrid).
KMS driver: **Dispmanx** is disabled and **cannot** be used, there's only OpenGL(ES) support.

Considering the above, I have multiple targets available in the Makefile to cover all possible combinations:

If you have Legacy driver: Use the **SDL1 + DispmanX** target
If you have the FKMS driver (Recommended): You can use both **SDL1 + Dispmanx** or **SDL2 + Dispmanx** versions. You can also use the pure SDL2 version, but it is currently slower than Dispmanx on the Pi, so it's not recommended.
If you have the full KMS driver and/or running under x11: Use the **pure SDL2** target (it's the only one that works)

Of course, how you configure SDL2 is important also:

- Use `--enable-video-kmsdrm` and `--disable-video-rpi` only if you're using FKMS or KMS driver. Otherwise it doesn't work anyway. It's important to disable the "rpi" video backend, otherwise it doesn't seem to work with Dispmanx (since we open it directly from Amiberry).

- You don't have to disable x11 or OpenGL if you're using the `fkms` driver, since the `kmsdrm` backend will take higher priority anyway and it can open a screen from the console.

I will update my Wiki to hopefully make this more clear also, but please let me know if the above makes sense and if you run into problems?

midwan, 2018-02-04:
@Fourdee 
Please note that the `dev` branch is now merged into `master` on my repo. :)
I've also made binaries available for multiple configurations (`sdl1` and `sdl2`). I've compiled the `sdl1` versions on DietPi and they seem to work fine.

Please let me know if you run into any problems! :)

Fourdee, 2018-02-05:
@midwan 

Was using ```vc4-kms-v3d```, for SDL2. Worked fine with the older AmiBerry binaries, but not the dev compiled one.

Thanks for the info 👍 , i'll recompile and retest from master.

midwan, 2018-02-05:
@Fourdee 
You can start using the master branch from now on, since the latest stuff is merged there. :)
If you want binaries, those are also available: https://github.com/midwan/amiberry/releases

Fourdee, 2018-02-06:
@midwan 

Many thanks 👍 

Getting somewhere, user error my end, was using ```dtoverlay=vc4-kms-v3d```, instead of:
```
dtoverlay=vc4-fkms-v3d
```

midwan, 2018-02-06:
@Fourdee 
Great, let me know if you get stuck anywhere and I can help!

Fourdee, 2018-02-06:
@midwan 

Yay, on screen AmiBerry 😃 . I should be good to go now, just need to update install in DietPi, will let you know if I get stuck 👍

Fourdee, 2018-02-06:
RPi 3: Post test install notes:
- KMS (without X11) Using our previous lower FB resolution (optimization), results in 3/4 of screen mouse reach and cursor offset to actual result.
- Elite using 1/5 of horz space. Probably due to using existing (old) autostart.uae, will create a new default.

Fourdee, 2018-02-06:
@midwan 

Any chance you have a stock autostart.uae for the latest version of AmiBerry I can test/distribute for our install? I tried clearing our previous config (to only leave in the paths), however, few issues with CPU speed being reset each restart:
https://github.com/Fourdee/DietPi/issues/1410#issuecomment-363589044

midwan, 2018-02-07:
@Fourdee 
That's why we have "Quickstart" options now, in the GUI itself... :)
If there is no bundled config, the Quickstart options should be enough to get you started with a pre-defined machine configuration, the default being the most common (A500 with 512Kb expansion RAM).

Also, remember to delete any old `adfdir.conf` file if you had it, it will be regenerated by the emulator anyway. We added more stuff there as well.

Fourdee, 2018-02-07:
Initial 1st pass installation for RPi 2/3 is completed.

Outstanding:
- General testing, identify/resolve any DietPi specific config issues. If not, i'll create Git Ticket on AmiBerry.
- Implement for other devices (i've got x86_64 PC on my personal to-do list @midwan  ;) )

midwan, 2018-02-07:
Great news!

I'm currently preparing an Android build, following that I'll start on a x86 target as well :)

Fourdee, 2018-02-07:
@midwan 

> I'm currently preparing an Android build

💘. AmiBerry on the go? Count me in 👍 
I can test on my https://www.gsmarena.com/xiaomi_redmi_note_4x-8580.php (-pink), just let me know.

midwan, 2018-02-07:
@Fourdee 
Thanks! I'll let you know once I've got something running that's not too embarrassing to share. :-)

midwan, 2018-02-07:
@Fourdee 
Actually, you can already have Amiberry on the go even without Android:
[image]

[image]

That's running a Raspberry Pi CM3 module ;)

Fourdee, 2018-02-08:
@midwan 

> Actually, you can already have Amiberry on the go even without Android:
That's running a Raspberry Pi CM3 module ;)

Simply epic!

midwan, 2018-02-11:
@Fourdee 
Heads up, just pushed out v2.13 out of beta. :)
The master branch is updated, and binaries are available for all supported distros (Jessie and Stretch) here: https://github.com/midwan/amiberry/releases

Fourdee, 2018-02-11:
@midwan 
Legend 👍 Many thanks.

RPi1-sdl2 aswell, nice one 😄 

I'll pick this back up next few days. v6.2 release planned before 18th

midwan, 2018-02-11:
@Fourdee 
Great!
Just a question: Will users of older versions of DietPi (e.g. <6.x) be able to upgrade their Amiberry installations through `dietpi-upgrade`, or is that not supported and they'll have to re-flash a v6.x image first?

Fourdee, 2018-02-11:
@midwan 

> Just a question: Will users of older versions of DietPi (e.g. <6.x) be able to upgrade their Amiberry installations through dietpi-upgrade, or is that not supported and they'll have to re-flash a v6.x image first?

Bit tricky, due to v159 requiring new image, and, AmiBerry image is Jessie, new one will be Stretch (as per v6.x)
If they are running DietPi v159 (actually v5.59) or lower, they will need to fresh install via latest image regardless. The existing code is not compatible with DietPi in v6.x. 

The other issue is the previous AmiBerry DietPi image is Jessie, the new one will be Stretch only for RPi. We now only support 1 image per device (RPi being Stretch as of v6.0). The Jessie AmiBerry image was offered as a way for users to install AmiBerry then, whilst waiting for the full RPi Stretch upgrade.

Due to the above, I believe patching is not viable. Its looking like fresh installs in this case, at the moment.

Fourdee, 2018-02-13:
Verified SDL2 installs and AmiBerry displaying on screen:
- 🈯️ RPi 3
- 🈯️ RPi Zero

ToDo:
- General usage tests
- Update online docs
- Create new Stretch Amiberry image
- Implement for non-rpi devices (will be done for v6.3, won't make it for v6.2 release this week)

midwan, 2018-02-18:
@Fourdee 
I'm pushing a new version into master today, v2.14. :)

Binaries will also be released with it, as usual. This time, I'll be creating the binaries using profiler-assisted optimizations, so the performance on those will be as good as possible. This also means that there will be a difference in performance between the binaries (best performance) and compiling from source directly, unless you also use the profiler of course.

To use the profiler, you need to compile the binary twice:
- First, enable the `GEN_PROFILE=1` commented out line in the Makefile, compile as normal, then run the generated binary and use it in a scenario you want to optimize for. I render a benchmark scene in Lightwave for example.
- Then you'll need to do a `make clean`, change the Makefile again to disable `GEN_PROFILE=1` and enable `USE_PROFILE=1` instead. Recompile for the same target and you'll get an optimized binary.
- To clean out the profiler generated files, you can do a `make cleanprofile`.
- Repeat the above process for each target you want to create a binary for.

Fourdee, 2018-02-18:
@midwan 

> I'm pushing a new version into master today, v2.14. :)
Binaries will also be released with it, as usual. This time, I'll be creating the binaries using profiler-assisted optimizations, so the performance on those will be as good as possible.

Excellent 👍 

I'am just wrapping up v6.2 for release today. I'll make sure to include (and test) the 2.14 binaries before we do.

Fourdee, 2018-02-18:
@midwan 

2.14 👍 

> Starting with v2.14, these also include the capsimg.so library, to enable IPF support.

Quick question, I assume ```capsimg.so``` is specific to the binary and/or CPU architecture its bundled with?

Fourdee, 2018-02-18:
@midwan

Some notes with testing, could be an issue DietPi end or not. Unsure at the moment.
 
#### Minor Issue 1:
RPi 3, SDL2:
- Run elite
- F12 
- Select restart
- Reload autostart.uae with previous setting and elite
- selecting "start" hangs with black screen for 10-20 seconds, before starting. Same with 4.9 and 4.14 kernels.

#### Minor issue 2:
- autostart.uae is not loaded upon 1st run or restart of AmiBerry. Manually selecting "load" is required afterwards.

#### Minor issue 3:
- Lack the RPi1 2.14 binary. We'll use the 2.13 for now. I won't get the time to recompile prior to release of DietPi today.

Apart from the above, Elite runs flawless and at a higher FPS than the previous AmiBerry version, great work 👍 🎖 

---

@midwan 
I'll update you with more details on the installation and the RPi AmiBerry image, once DietPi v6.2 is released.

midwan, 2018-02-18:
@Fourdee 
The RPI1 build has some issues with JIT (segfaults), so I didn't provide a binary until I find a solution.

What does "autostart.uae" contain? Was it generated from Quickstart? If not, I'd advise that it's generated from scratch, using a Quickstart config as a template, since some config values have been changed from older versions of the emulator and perhaps something in there is incorrect...

Fourdee, 2018-02-18:
@midwan 

Image done for RPi, auto installation of latest AmiBerry:
- New DL link (no changes needed to amiberry.com info, only the link change) http://dietpi.com/downloads/images/DietPi_RPi-ARMv6-Stretch_AmiBerry.7z
- Brings image up to Debian Stretch
- SDL2 is the default and only installation method (I've opted to remove SDL1, as SDL2 seems stable and offers improved performance, newer library etc). If there is any major reason to include SDL1 versions again, please let me know and we can review
- RPi2/3 = 2.1.4, RPi1 = 2.1.3. But we can now patch during DietPi updates with ease.
- 🈯️ RPi 3 test install
- 🈴 Outstanding issue with ```autostart.uae``` not loading automatically. I'll review/retest when I can.

Any problems, please let me know.

Great work again on 2.1.4, amazing achievement. When I get 5 minutes, i'll be having a good play of Elite 👍 😃

midwan, 2018-02-19:
@Fourdee 
Thanks, I've updated the link to the download webpage. :)

midwan, 2018-03-18:
@Fourdee 
Amiberry v2.18 is now online! :)
https://github.com/midwan/amiberry/releases

- Improved audio latency
- Unique new feature: WHDLoad booter! (check the wiki for details)
- Bugfixes in various places

Fourdee, 2018-03-20:
@midwan 

> Improved audio latency

👍 💘 

Great work as always and thanks for the heads up. We'll patch the 2.18 update for existing installations, in our v6.5 release (ETA 1 week).

MichaIng, 2018-06-11:
v2.19 up: https://github.com/midwan/amiberry/releases

Fourdee, 2018-06-29:
> v2.19 up: https://github.com/midwan/amiberry/releases

Was added in v6.8

midwan, 2019-01-09:
@Fourdee 
Regarding the Tinkerboard, we made some good progress recently and we have a minimal image (based on Armbian) with GPU support that runs very quickly. There's a wiki on how to set it up from scratch here: https://github.com/midwan/amiberry/wiki/Compiling-for-the-Tinker-Board

I think you should be able to use it to get GPU support for DietPi as well, and then the Amiberry version for the Tinkerboard can be included in the list of platforms also :)

Fourdee, 2019-01-10:
@midwan 

Great news, good to hear 👍 

I'll take a look when I can.

Fourdee, 2019-02-28:
Will mark this as closed and create new tickets as needed.

closed 2019-02-28 by Fourdee