MichaIng/DietPi #1411: Linux Dash didn't show correct info

RmG152, 2018-01-20:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | 159
- SBC Device: Orange Pi Zero / Raspberry Pi 3
- Power supply used: Usb / usb hub Anker 60w
- SD card used: Samsung evo / Sandisk Ultra
- Distro:
Linux DietPiVoice1 3.4.113-sun8i #2 SMP PREEMPT Fri Mar 10 18:24:50 GMT 2017 armv7l GNU/Linux
Linux DietPi 4.9.62-v7+ #2 SMP Fri Nov 17 23:52:26 GMT 2017 armv7l GNU/Linux

#### Additional Information (if applicable):
- Software title? Linux Dash
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Expected behaviour:
Show Current system information (OS, processes, hardware info, etc(.

#### Actual behaviour:
Board is showing demo info:
https://afaqurk.github.io/linux-dash/#/system-status

#### Steps to reproduce:
Download DietPI
Install
Install LinuxDash from dietpi-config
Enter LinuxDash

#### Did you submit a ```dietpi-bugreport```?
No

#### Extra details:
NA

gotttlieb, 2018-01-20:
I confirm this behaviour in Odroid XU4

RmG152, 2018-01-20:
This is working well:
/linuxdash/app/

This one not:
/linuxdash/

gotttlieb, 2018-01-20:
I suggest to change the Documentation to avoid this confusion  and delete this dummy data form dietpi instalation. 
http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=20#p108

Fourdee, 2018-01-20:
@RmG152 

Thanks for the report 👍 

We'll investigate/resolve when possible.

Fourdee, 2018-01-21:
Updated documentation to reflect the correct URL link (as per @gotttlieb suggestion) 👍 
http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=108#p108

closed 2018-01-21 by Fourdee