MichaIng/DietPi #1413: DietPi-Software | ownCloud/Nextcloud: v159 => v6.0 migration script

MichaIng, 2018-01-20:
+ Added backup script to separate branch, not intended for merging, just for download/review: https://github.com/Fourdee/DietPi/blob/dietpi-cloud-migration/dietpi-cloud-migration
+ Script could be provided to users via forum + https://github.com/Fourdee/DietPi/issues/1355
+ Could be further implemented into e.g. dietpi-backup/dietpi-software uninstall.
+ Current idea: dietpi-software install ownCloud/Nextcloud will check for existing backups and restore in case. PR for this is on the way 😃.

MichaIng, 2018-01-20:
Automatic recovery is ready for testing: https://github.com/Fourdee/DietPi/pull/1414

On my Stretch VM following schedules worked well:
🈯️ Install ownCloud/Nextcloud ▶️ Use migration script ▶️ Uninstall ownCloud without backup creation ▶️ Install
🈯️ Install ownCloud/Nextcloud ▶️ Uninstall (with automated backup creation) ▶️ Install
🈯️ Install ownCloud/Nextcloud ▶️ Use migration script ▶️ Uninstall ▶️ Install
🈯️ Install ownCloud/Nextcloud ▶️ Reinstall ownCloud/Nextcloud (with existing installdir and database)

keithellis74, 2018-01-26:
Does this migration script also transfer across LetsEncrypt keys/scrips?

Fourdee, 2018-01-28:
@keithellis74 

Not as far as i'am aware, this script is only for NC/OC migration.

MichaIng, 2018-01-28:
@keithellis74 
Yes, it only migrates NC/OC, as the instances are quite independent from webserver/SSL configurations.

To transfer LetsEncrypt info, you just need to transfer the `/etc/letsencrypt` folder, as far as I know, where keys and account infos are inside.

MichaIng, 2018-01-28:
Announcement about the migration script: http://dietpi.com/phpbb/viewtopic.php?f=9&t=2734

Note:
- As always, do carefully check the output of the script and recheck the made backup.
- Carefully read the instructions given in the announcement above.

Fourdee, 2018-01-29:
@MichaIng 
Many thanks for doing this for our users, really appreciate it 👍 

I'll mark this as closed for now, please reopen if required.

closed 2018-01-29 by Fourdee