MichaIng/DietPi #1423: Outstanding v6.0 images

Fourdee, 2018-01-25:
### Outstanding tasks, that require re-imaging:
NB: here for reference only, no guarantee these will make the v6.0 release, due to lack of time.
- Disable ipv6 by default, remove apt/wget prefer ipv6, ```dietpi.txt``` ```ipv6enable``` option: https://github.com/Fourdee/DietPi/pull/1419#issuecomment-360452027
- 1MB fat partition for ```dietpi.txt``` on ARMbian images: https://github.com/Fourdee/DietPi/issues/1385#issuecomment-360261137

MichaIng, 2018-01-25:
@Fourdee 
As said, disabling IPv6 by default could break 1st run connectivity for some users. So we would force them to check what they need and toggle in dietpi.txt. But I think it would be best to guarantee best 1st run compatibility with default dietpi.txt, especially if it's about network, as without SSH your headless server is just an energy consuming stone 😆.
If users then know, that they anyway have no IPv6 connectivity or don't want it, the can disable by themselves via dietpi-config, now also for x64 😄.

The cases where IPv6 totally break (instead of just slower e.g.) seem to be rare, e.g. my VMs over bridged WLAN, and the default apt/wget IPv4 force/prefer fix this for most use cases (for now besides PHP readfile), while IPv6 stays possible for those who need.

Actually I wanted to test outgoing IPv6 behaviour on my RPi, to hopefully reduce the issue to the VMs...

Fourdee, 2018-01-25:
@MichaIng 

> As said, disabling IPv6 by default could break 1st run connectivity for some users.

Agree 👍 

Ok, we'll leave IPv6 enabled by default for now. 
However, I would like to remove the apt/wget prefer IP versions, as long as the default is prefer IPv6 (when enabled), and we can assume IPv6 disabled results in IPv4 systemwide?

User can then configure in dietpi-config, if issues with IPv6? Seems a simpler/cleaner solution than manually setting programs IP preference?

MichaIng, 2018-01-25:
@Fourdee 
Yeah I agree that the current settings a bid confusing and mixed.
> default for programs is prefer IPv6 (when enabled)

This seems to case, at least for APT, wget and PHP readfile, if the destination sever offers IPv6 of course.
> we can assume IPv6 disabled, results in IPv4 systemwide?

Now, that we "really" disable IPv6 within grub for amd64, this also seems the case, means IPv4 is used automatically. At least for wget, apt and `php -r "readfile('https://getcomposer.org/installer');"` I could successfully test it: Access is done via IPv4 automatically without any prefer/force IPv4 settings set.

What we could do:
=> Switch dietpi.txt setting to ipv6enable (or ipv6disable), where I would prefer to leave it enabled by default.
=> Adjust behaviour to same than dietpi-config and just remove all IPv4 prefer/force adjustments.
=> If we leave IPv6 enabled by default, then for 1st run, use the prefer IPv4 setting for wget and APT, just until 1st run setup (dietpi-software) is done and remove the settings/conf-file afterwards. This guarantees that in the mentioned rare cases, where IPv6 breaks, at least dietpi-update and dietpi-software/config is able to start. If user then faces the issue, (s)he can disable IPv4 via dietpi-config. Actually in these cases user should already know the problem, as this even appears on Debian mini.iso installation.

Fourdee, 2018-02-22:
Marking as closed.

We'll leave IP settings as they are for now. More important tickets to resolve.

closed 2018-02-22 by Fourdee