MichaIng/DietPi #1425: Build-essential apt-get failed on Odroid C1

mcsSolutions, 2018-01-26:
## Creating a bug report/issue:
Apt-get has failed:
 -Apt string = build-essential make autoconf automake
--no-install-recommends
-Error code = 100
-Software Title = Build-Essentials: common packages for compile
-Apt log = /etc/dietpi/logs/dietpi-software_apt.log

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```? 159
- SBC Device (EG: RPi 3)? Odroid C1
- Power supply used (EG: 5v 1A RAVpower)? 5V 2.5A
- SD card used (EG: Sandisk ultra)? Sandisk Ultra Plus 16 GB
- Distro (EG: Jessie) | ```uname -a```?Linux DietPi 3.10.104 #1 SMP PREEMP Thu Nov 10 07:56:43 CET 2016 armv71 GNU/Linux  (Jessie)

#### Additional Information (if applicable):
- Software title? DietPi-software / additional software select build-essentials or WiringPi
- Can this issue be replicated on a fresh installation of DietPi? yes, replicated on two installs

#### Expected behaviour:
 make file installed to allow compile/build of WiringPi install

#### Actual behaviour:
 Error code = 100 (broken/impossible links) as shown at start of this report

#### Steps to reproduce:
dietpi-software / select additional software / select build-essentials / select install-go.  Can also be reproduced by dietpi-software / select optimized software / select WiringPi / select install-go

#### Did you submit a ```dietpi-bugreport```? no

#### Extra details:
/etc/dietpi/logs/dietpi-software_apt.log contains "E: Unable to correct problems, you have held broken packages."
[dietpi-software_apt.log](https://github.com/Fourdee/DietPi/files/1669348/dietpi-software_apt.log)

Fourdee, 2018-01-27:
@mcsSolutions 

Hi, 

Please try running the following to fix APT:
```
apt-get install -f
```

Then try reinstalling build essential:
```
dietpi-software install 16
```

mcsSolutions, 2018-01-27:
No problems with executing commands, but does not seem to resolve issue.  The console

E: Unable to correct problems, have held broken packages.

[Failed] The apt cache may be corrupt, apt mirror offline, or have held broken packages. DietPi-Software will now exit.

[Info] Apt Log:

E: Unable to correct problems, have have held broken packages.

diFixing apt package errors with ‘apt-get install –f’

Change apt mirror: ‘dietpi-config’ > Networking: NAS/MISC’ > ‘Apt Mirror’

Clear and refresh apt ‘apg-get update’

If problems persist, please crete a Git ticket so DietPi can investigate the issue:

I then changed the mirror to US and repeated the two commands you recommended.  Same result.

Fourdee, 2018-01-27:
@mcsSolutions 
Could be a DNS/APT_mirror issue, try running the command manually:
```
apt-get update
apt-get install build-essential -y
```

Please copy/paste any errors.

mcsSolutions, 2018-01-27:
root@Odroid32:~# apt-get update

Ign http://ftp.us.debian.org jessie InRelease

Get:1 http://ftp.us.debian.org jessie-updates InRelease [145 kB]

Get:2 http://security.debian.org jessie/updates InRelease [63.1 kB]

Hit http://fuzon.co.uk all InRelease

Hit http://download.mono-project.com jessie InRelease

Hit http://fuzon.co.uk jessie InRelease

Hit http://security.debian.org jessie/updates/main armhf Packages

Hit http://fuzon.co.uk all/c1 armhf Packages

Hit http://download.mono-project.com jessie/main armhf Packages

Hit http://security.debian.org jessie/updates/contrib armhf Packages

Hit http://fuzon.co.uk all/main armhf Packages

Hit http://security.debian.org jessie/updates/non-free armhf Packages

Hit http://security.debian.org jessie/updates/contrib Translation-en

Hit http://fuzon.co.uk all/testing armhf Packages

Hit http://security.debian.org jessie/updates/main Translation-en

Get:3 http://ftp.us.debian.org jessie-backports InRelease [166 kB]

Hit http://security.debian.org jessie/updates/non-free Translation-en

Hit http://ftp.us.debian.org jessie Release.gpg

Get:4 http://ftp.us.debian.org jessie-updates/main armhf Packages/DiffIndex [9,868 B]

Hit http://ftp.us.debian.org jessie-updates/contrib armhf Packages

Get:5 http://ftp.us.debian.org jessie-updates/non-free armhf Packages/DiffIndex [736 B]

Hit http://fuzon.co.uk jessie/main armhf Packages

Hit http://ftp.us.debian.org jessie-updates/contrib Translation-en

Ign http://download.mono-project.com jessie/main Translation-en_GB

Get:6 http://ftp.us.debian.org jessie-updates/main Translation-en/DiffIndex [3,688 B]

Hit http://fuzon.co.uk jessie/backports armhf Packages

Get:7 http://ftp.us.debian.org jessie-updates/non-free Translation-en/DiffIndex [736 B]

Ign http://download.mono-project.com jessie/main Translation-en

Hit http://ftp.us.debian.org jessie Release

Get:8 http://ftp.us.debian.org jessie-backports/main armhf Packages/DiffIndex [27.8 kB]

Get:9 http://ftp.us.debian.org jessie-backports/contrib armhf Packages/DiffIndex [25.7 kB]

Get:10 http://ftp.us.debian.org jessie-backports/non-free armhf Packages/DiffIndex [8,530 B]

Get:11 http://ftp.us.debian.org jessie-backports/contrib Translation-en/DiffIndex [7,468 B]

Get:12 http://ftp.us.debian.org jessie-backports/main Translation-en/DiffIndex [27.8 kB]

Get:13 http://ftp.us.debian.org jessie-backports/non-free Translation-en/DiffIndex [17.2 kB]

Hit http://ftp.us.debian.org jessie/main armhf Packages

Hit http://ftp.us.debian.org jessie/contrib armhf Packages

Hit http://ftp.us.debian.org jessie/non-free armhf Packages

Hit http://ftp.us.debian.org jessie/contrib Translation-en

Hit http://ftp.us.debian.org jessie/main Translation-en

Hit http://ftp.us.debian.org jessie/non-free Translation-en

Ign http://fuzon.co.uk all/c1 Translation-en_GB

Ign http://fuzon.co.uk all/c1 Translation-en

Ign http://fuzon.co.uk all/main Translation-en_GB

Ign http://fuzon.co.uk all/main Translation-en

Ign http://fuzon.co.uk all/testing Translation-en_GB

Ign http://fuzon.co.uk all/testing Translation-en

Ign http://fuzon.co.uk jessie/backports Translation-en_GB

Ign http://fuzon.co.uk jessie/backports Translation-en

Ign http://fuzon.co.uk jessie/main Translation-en_GB

Ign http://fuzon.co.uk jessie/main Translation-en

Fetched 504 kB in 16s (30.6 kB/s)

Reading package lists... Done

root@Odroid32:~# apt-get install build-essential -y

Reading package lists... Done

Building dependency tree

Reading state information... Done

Some packages could not be installed. This may mean that you have

requested an impossible situation or if you are using the unstable

distribution that some required packages have not yet been created

or been moved out of Incoming.

The following information may help to resolve the situation:

 

The following packages have unmet dependencies:

build-essential : Depends: dpkg-dev (>= 1.17.11) but it is not going to be installed

E: Unable to correct problems, you have held broken packages.

Fourdee, 2018-01-28:
@mcsSolutions 
Strange, unable to replicate here in our testing. Limited to your system/APT sources and configuration.

> build-essential : Depends: dpkg-dev (>= 1.17.11) but it is not going to be installed

Try install the pre-req manually:
```
apt-get install dpkg-dev build-essential
```

mcsSolutions, 2018-01-28:
root@Odroid32:~# apt-get install dpkg-dev build-essential

Reading package lists... Done

Building dependency tree

Reading state information... Done

Some packages could not be installed. This may mean that you have

requested an impossible situation or if you are using the unstable

distribution that some required packages have not yet been created

or been moved out of Incoming.

The following information may help to resolve the situation:

 

The following packages have unmet dependencies:

dpkg-dev : Depends: libdpkg-perl (= 1.18.24~bpo8) but it is not going to be installed

            Depends: tar (>= 1.28-1) but 1.27.1-2+deb8u1 is to be installed

E: Unable to correct problems, you have held broken packages.

mcsSolutions, 2018-01-28:
Are you using Odroid C1 in your test environment?  I noticed that this DietPi package is Jessie while the Raspberry Pi one is stretch.  I was able to install build-essential on Raspberry Pi with DietPi.  It shows as version 12.3.

MichaIng, 2018-01-28:
@mcsSolutions 
`Ign http://ftp.us.debian.org jessie InRelease`
```
dpkg-dev : Depends: libdpkg-perl (= 1.18.24~bpo8) but it is not going to be installed

            Depends: tar (>= 1.28-1) but 1.27.1-2+deb8u1 is to be installed
```
As this mirror was somehow broken on your last apt update, APT tried to install `build-essential` from backports. But it does not by itself choose also the dependencies from backports, thus fail.
I personally would, if not necessary, stay without backports. Thus try to manually purge the installed packages again:
- `apt-get purge dpkg-dev build-essential`
- Then try to find a running APT mirror. Try `http://deb.debian.org/debian/`, which is a mirror director and should always find a running server. Then do apt update again and assure that is says `Get http://deb.debian.org jessie` instead of `Ign ...`.
- If this works, then reinstall build-essentials via dietpi-software.
- Alternatively at your current stage, you could force APT to get the package and all dependencies from backports: `apt-get install -y -t jessie-backports build-essential` and then rerun installation via dietpi-software (which includes some other packages as well).

k-plan, 2018-01-29:
@MichaIng 

puhh ... mcsSolutions use an old v159 image on Odroid C1.
If I remember correct, where was a `/etc/apt/preferences.d/meveric-backports-pin-100`.

https://github.com/Fourdee/DietPi/issues/1385#issuecomment-357755338

If I find some spare time, I can do a reteset on the new build ` DietPi_v6.0_OdroidC1-ARMv7-Jessie.img
` and install` Build-essential` on it via `dietpi-software`.
But can take some days.

MichaIng, 2018-01-29:
@Fourdee 
Yes generally this should work, thou good to retest on Odoids that also come with additional repos.

In this case here it clearly looks like (temporary) unavailable mirror, thus APT chose meveric backports, failing dependencies.
Hmm about mirror again: `Ign http://ftp.us.debian.org jessie InRelease` Shouldn't there be an `/debian/` inside this URL as well?

mcsSolutions, 2018-01-29:
Performed the following with as much of the PuTTY windows captured as I could.  I did not understand the specific steps associated with finding an APT mirror.  Same failure result.

*	apt-get purge dpkg-dev build-essential
*	Alternatively at your current stage, you could force APT to get the package and all dependencies from backports: apt-get install -y -t jessie-backports build-essential and then rerun installation via dietpi-software (which includes some other packages as well).

 

root@Odroid32:~# apt-get purge dpkg-dev build-essential

Reading package lists... Done

Building dependency tree

Reading state information... Done

Package 'build-essential' is not installed, so not removed

Package 'dpkg-dev' is not installed, so not removed

The following package was automatically installed and is no longer required:

  libpci3

Use 'apt-get autoremove' to remove it.

0 upgraded, 0 newly installed, 0 to remove and 54 not upgraded.

root@Odroid32:~# apt-get install -y -t jessie-backports build-essential

Reading package lists... Done

Building dependency tree

Reading state information... Done

The following package was automatically installed and is no longer required:

  libpci3

Use 'apt-get autoremove' to remove it.

The following extra packages will be installed:

  cpp cpp-4.9 dpkg dpkg-dev g++ g++-4.9 gcc gcc-4.9 libasan1 libatomic1

  libc-dev-bin libc6 libc6-dev libcloog-isl4 libdpkg-perl libgcc-4.9-dev

  libgomp1 libisl10 libmpc3 libmpfr4 libstdc++-4.9-dev libubsan0

  linux-libc-dev make patch perl tar

Suggested packages:

  cpp-doc gcc-4.9-locales debsig-verify debian-keyring gcc-4.9-doc

  libstdc++6-4.9-dbg gcc-multilib manpages-dev autoconf automake libtool flex

  bison gdb gcc-doc libgcc1-dbg libgomp1-dbg libitm1-dbg libatomic1-dbg

  libasan1-dbg liblsan0-dbg libtsan0-dbg libubsan0-dbg libcilkrts5-dbg

  libquadmath-dbg glibc-doc libstdc++-4.9-doc make-doc ed diffutils-doc

  perl-doc libterm-readline-gnu-perl libterm-readline-perl-perl ncompress

  tar-scripts

Recommended packages:

  fakeroot libalgorithm-merge-perl libfile-fcntllock-perl rename
[... 639 lines cut ...]
     │  - Apt string = build-essential make autoconf automake             │

     │ --no-install-recommends                                            │

     │  - Error code = 100                                                │

     │  - Software Title = Build-Essentials: common packages for compile  │

     │  - Apt log = /etc/dietpi/logs/dietpi-software_apt.log              │

     │                                                                    │

     │ Please report this issue to DietPi (INCLUDING THE APT LOG) so it   │

     │ can be investigated.                                               │

     │                                                                    │

     │ DietPi-Software will now exit.                                     │

     │                                                                    │

     │                                                                    │

     │                               <Ok>

MichaIng, 2018-01-29:
@mcsSolutions 
Ah the `LD_LIBRARY_PATH` issue. At least the mirrors are all reachable now:
```
unset LD_LIBRARY_PATH
apt install build-essential
```
If an error about missing dependencies appears, then it somehow already remembers backports to be installed, then do:
`apt install -y -t jessie-backports build-essential`

k-plan, 2018-01-29:
Yes, as I mention before, old DietPi v159 image

```
unset LD_LIBRARY_PATH
ldconfig -v
```
To get it work durable, you have to fix export in` /etc/bash.bashrc`

mcsSolutions, 2018-01-29:
Unset LD_LIBRARY_PATH corrected the problem

 

I started this process about a week ago with Odroid C1 DietPi with the zip from the DiepPi web page.  Seems odd that my version was considered old.  I noticed that first thing on the downloaded image it did after login was update the version.

 

I repeated the download, install, build-essential selection today after the issue was resolved and the DietPi/dietpi/.version shows as two lines( 6 then 1). Uname gives 3.10.107 vs. last week’s report of 3.10.104.  I had no issues today with build-essential.

 

From my perspective the issues is resolved.  

 

Thank you for the rapid resolution.

k-plan, 2018-01-30:
> From my perspective the issues is resolved.

:+1: 

I'll mark as closed, latest image will resolve the issue, however, please re-open if required.

closed 2018-01-30 by k-plan