MichaIng/DietPi #1427: Orange Pi Zero - Ethernet not working

maxim-sheronov, 2018-01-28:
## Creating a bug report/issue:

Ethernet on Orange Pi Zero doesn't working.
I have tried image DietPi_OrangePiZero-armv7-(Jessie). It can be configured for Wi-Fi connection, but Ethernet not working. It can be enabled, connected, but didn't try to get IP address from DHCP. Static IP not working also.

On last image DietPi_OPiZeroH2-ARMv7-Stretch there no way to configure first config in dietpi.txt 
So I can't connect with Wi-Fi. And ethernet is not working.

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?
- SBC Device (EG: RPi 3)? Orange Pi Zero
- Power supply used (EG: 5v 1A RAVpower)? 5v 2A
- SD card used (EG: Sandisk ultra)? 2Gb
- Distro (EG: Jessie) | ```uname -a```? Jessie and Stretch

maxim-sheronov, 2018-01-28:
So it is working if wait a long time

closed 2018-01-28 by maxim-sheronov

luikismi, 2018-03-16:
I have same problem with an orange pi win plus...
I have wifi but i can´t use ethernet, i can´t receive IP, if i use static address it doesn't work too.
I tried in debian,linux server, linux desktop.... with the same result..
Any idea?