MichaIng/DietPi #1428: Kodi install error using dietpi-software on raspberry pi 2 model b

Saxndi, 2018-01-28:
#### Required Information:
- DietPi Version | v6.0
- SBC Device     | RPi 2 Model B (armv7l) (index=2)
- Distro         | stretch (index=4)
- Command        | G_AGI: libcec3*
- Error Handler  | G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title | DietPi-Software

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Additional logs:
```
Log file contents:
E: Unable to correct problems, you have held broken packages.
```

MichaIng, 2018-01-28:
Oh yes, I see some issue with the installation there: https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L4116-L4126

The comments already suggest, that APT failure for this package shall not break whole installation. But as G_AGI is used, the included error handler will break it. We need to tell it to let the installation go on or just use `apt-get install -y libcec3*` etc. there. I just checked on my Stretch VM: There it's just `libcec4` On Jessie it's `libcec2`: https://packages.debian.org/search?suite=default&section=all&arch=any&lang=de&searchon=names&keywords=libcec
Sadly just on Raspbian there is a meta package called `libcec` = `libcec4` on Stretch. So no idea where `libcec3` can be found 😄.

@Saxndi 
To solve:
- `nano /DietPi/dietpi/dietpi-software`
- press "strg" + "-" (hyphen), type 4116, press return, to jump to line 4116
- Remove the lines 4116 until 4126: https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L4116-L4126
Just add `G_AGI libcec4`

@Fourdee 
If I am not missleaded we should just add G_AGI libcec4 to RPi loop: https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L4097-L4101
and `G_AGI libcec3v4` for Odroids, until we move these to Stretch? Could do $G_DISTRO check there also, for Odroids Stretch testing images and then in case also choose libcec4

Also nfs support needs update: https://packages.debian.org/de/stretch/libnfs8
=> `G_AGI libnfs8` for RPi (Stretch), while libnfs4 is just available on Jessie. The package is btw. already "recommended" for Debian kodi package.

Saxndi, 2018-01-28:
@MichaIng Thanks for the quick support. With your help I could install kodi. Unfortunately also tonido doesn't work well with the new dietpi installation. During the installation process I get

DietPi-Software
─────────────────────────────────────────────────────
 Mode: Optimize and configure software

[ INFO ] Applying DietPi optimizations and configurations for RPi 2 Model B (armv7l), please wait...

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Configuring Tonido: lightweight cloud based backup system

cp: cannot stat '/root/tonido': No such file or directory
rm: cannot remove '/root/tonido': No such file or directory

MichaIng, 2018-01-28:
@Saxndi 
- These errors are expected. We copy and remove this folder, in case it exists. Usually we suppress the error messages in such cases, respectively do check for existence, if it exists. Seems to be missed here.
- Does the installation otherwise work on your system?

Seems we still missed to test the one or the other software title on Stretch:
- I faced APT error on my Stretch VM, because libpng12.0 is attempted to install, but unavailable on Stretch: https://packages.debian.org/search?suite=default&section=all&arch=any&searchon=names&keywords=libpng
- Stretch offers `libpng16-16`, but as often, software still depends on the old library version... On Debian `wget http://deb.debian.org/debian/pool/main/libp/libpng/libpng12-0_1.2.50-2+deb8u3_amd64.deb` works well for amd64 Stretch.
- If http://deb.debian.org/debian/pool/main/libp/libpng/libpng12-0_1.2.50-2+deb8u3_armhf.deb works on RPi would need testing. @Saxndi If, besides the error messages you mentioned, tonido does not start, then you could try it:
wget http://deb.debian.org/debian/pool/main/libp/libpng/libpng12-0_1.2.50-2+deb8u3_armhf.deb
dpkg -i libpng12-0_1.2.50-2+deb8u3_armhf.deb
rm libpng12-0_1.2.50-2+deb8u3_armhf.deb
dietpi-software reinstall 134
- @Fourdee Do we want to host our own libpng12.0 for RPi at least, if Saxndi/someone else confirms this as the issue on RPi Stretch?

Saxndi, 2018-01-29:
@MichaIng Also after reinstalling tonido, it doesn't work. It seems that the installation works fine, but I can't open <IP-Adress>:10001 
There are no log files written.

When attempting to start tonido I get

root@DietPi:~# systemctl status tonido
● tonido.service - Tonido - DietPi
   Loaded: loaded (/etc/systemd/system/tonido.service; disabled; vendor preset:
enabled)
   Active: failed (Result: exit-code) since Mon 2018-01-29 08:08:40 G
MT; 13s ago
  Process: 1594 ExecStart=/bin/bash -c export LD_LIBRARY_PATH=/etc/tonido; expor
t TONIDODIR=/etc/tonido; ./tonidoconsole (code=exited, status=127)
 Main PID: 1594 (code=exited, status=127)

Jan 29 08:08:40 DietPi systemd[1]: Started Tonido - DietPi.
Jan 29 08:08:40 DietPi bash[1594]: ./tonidoconsole: error while loading shared l
ibraries: libcrypto.so.1.0.0: cannot open shared object file: No such file or di
rectory
Jan 29 08:08:40 DietPi systemd[1]: tonido.service: Main process exited,
code=exited, status=127/n/a
Jan 29 08:08:40 DietPi systemd[1]: tonido.service: Unit entered failed s
tate.
Jan 29 08:08:40 DietPi systemd[1]: tonido.service: Failed with result 'e
xit-code'.

MichaIng, 2018-01-29:
@Saxndi 
Ah, the libssl1.0.0 issue, yeah we found that already on other software titles, switching to Stretch.
- I was wondering, why I didn't face this on my VM: We use the download from tonido main download page there: https://www.tonido.com/tonidodesktop_downloads/
- For RPi we use the download from another URL, as given on RPi installation instructions: https://www.tonido.com/support/display/docs/Raspberry+Pi
- However on main download page, there is also a download link for RPi, pointing to another version of tonido. The one from main page is an older version (6.82. vs. 10.86.), but it includes some libraries, also libssl and libcrypto, but seems 0.9.8, an (too) old one.
- Please try:
```
wget http://deb.debian.org/debian/pool/main/o/openssl/libssl1.0.0_1.0.2l-1~bpo8+1_armhf.deb
dpkg -i libssl1.0.0_1.0.2l-1~bpo8+1_armhf.deb
rm libssl1.0.0_1.0.2l-1~bpo8+1_armhf.deb
```
- Does tonido include an internal updater or something? x86 Linux versions are made 2017, but RPi ones from 2015 (download we use) and 2014 (from their main page).

Saxndi, 2018-01-29:
@MichaIng 
Ok, now tonido works, thank you very much for your help!
I also realized that fail2ban-service (installed via dietpi-software) doesn't start.

MichaIng, 2018-01-29:
@Saxndi 
Found it: https://github.com/voxpupuli/puppet-fail2ban/issues/13
We seem to use the same base `jail.conf`, thus face the same error: https://github.com/Fourdee/DietPi/blob/master/dietpi/conf/fail2ban_jail.conf#L46-L54
`nano /etc/fail2ban/jail.conf`
Comment out or delete line 52.
`systemctl restart fail2ban`

Thank you for the reports 👍. I think we really have to go through all software. Even that we have Stretch for testing for nearly 1,5 years, it seems not all titles were tested on all Stretch systems.
But in theory: 34 devices times 165 software titles means 5610 test installations 😱.

Fourdee, 2018-01-29:
@MichaIng 

Confirmed:
- Odroid Jessie ```libcec3v4 ```
- RPi ```libcec4``` 

I'll send PR and test both.

> But in theory: 34 devices times 165 software titles means 5610 test installations

😱 👍

Fourdee, 2018-01-29:
@MichaIng 

All confirmed DietPi specific bugs, lets set to high priority for v6.1.

MichaIng, 2018-01-29:
@Fourdee 
About libssl1.0.0 I guess http://dietpi.com/downloads/binaries/all/libssl1.0.0_1.0.2l-1_bpo8+1_armhf.deb is exactly the one from debian pool/backports http://deb.debian.org/debian/pool/main/o/openssl/libssl1.0.0_1.0.2l-1~bpo8+1_armhf.deb right?
So we can use this for RPi, which comes without included libraries?

Saxndi, 2018-01-29:
@MichaIng 

root@DietPi:~# systemctl restart fail2ban
Job for fail2ban.service failed because the control process exited with error co                          de.
See "systemctl status fail2ban.service" and "journalctl -xe" for details.
root@DietPi:~# systemctl status fail2ban.service
● fail2ban.service - Fail2Ban Service
   Loaded: loaded (/lib/systemd/system/fail2ban.service; disabled; vendor preset
: enabled)
   Active: activating (start) since Mon 2018-01-29 13:09:11 GMT; 1s ago
     Docs: man:fail2ban(1)
Cntrl PID: 2647 (fail2ban-client)
   CGroup: /system.slice/fail2ban.service
           └─2647 /usr/bin/python3 /usr/bin/fail2ban-client -x start

Jan 29 13:09:11 DietPi systemd[1]: Starting Fail2Ban Service...
Jan 29 13:09:13 DietPi fail2ban-client[2647]: WARNING 'syslogsocket' not defined
 in 'Definition'. Using default one: 'auto'
Jan 29 13:09:13 DietPi fail2ban-client[2647]: WARNING 'dbfile' not defined in 'D
efinition'. Using default one: '/var/lib/fail2ban/fail2ban.sqlite3'
Jan 29 13:09:13 DietPi fail2ban-client[2647]: WARNING 'dbpurgeage' not defined i
n 'Definition'. Using default one: 86400
Jan 29 13:09:13 DietPi fail2ban-client[2647]: WARNING 'filter' not defined in 's
shd'. Using default one: ''
Jan 29 13:09:13 DietPi fail2ban-client[2647]: WARNING No filter set for jail ssh
d
Jan 29 13:09:13 DietPi fail2ban-client[2647]: WARNING 'filter' not defined in 's
shd'. Using default one: ''
Jan 29 13:09:13 DietPi fail2ban-client[2647]: ERROR  Failed during configuration
: Bad value substitution: option 'action' in section 'sshd' contains an interpolation key 'port' which is
not a valid option name. Raw value: '%(action_)s'

Fourdee, 2018-01-29:
Please use following tickets:
- Fail2Ban issue: https://github.com/Fourdee/DietPi/issues/1431
- Tonido issue: https://github.com/Fourdee/DietPi/issues/1432

🈯️ Kodi issue resolved and tested with:
https://github.com/Fourdee/DietPi/commit/4ea8c2d058a4efe0cdb0891a18b3f563d84f0a20

closed 2018-01-29 by Fourdee

k-plan, 2018-01-29:
> it seems not all titles were tested on all Stretch systems.
> But in theory: 34 devices times 165 software titles means 5610 test installations

YES! :+1: 

Sorry, it is simple a [Mission: Impossible](https://en.wikipedia.org/wiki/Mission:_Impossible_(film_series))