MichaIng/DietPi #1429: First login installation failed - Google AIY empty hostname

maxim-sheronov, 2018-01-28:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```? DietPi_OrangePiZero-armv7-(Jessie)
- SBC Device (EG: RPi 3)? Orange Pi Zero
- Power supply used (EG: 5v 1A RAVpower)? 5v 2A
- SD card used (EG: Sandisk ultra)? 2Gb
- Distro (EG: Jessie) | ```uname -a```? Jessie

#### Expected behaviour:
First login installation should finish correct

#### Actual behaviour:
Installation failed on step Google AIY
It says, that can't connect to host http:// (it is empty - it is BUG)
And failed on it.

Fourdee, 2018-01-29:
@maxim-sheronov 

Thanks for the report, i'll try to replicate.

Fourdee, 2018-01-29:
@maxim-sheronov 

> Installation failed on step Google AIY
DietPi_OrangePiZero-armv7-(Jessie)

The DietPi-Software installation of Google AIY, is only enabled/supported for RPi devices, how did you install it?

maxim-sheronov, 2018-01-29:
I didn't. I just flash image for OrangePi Zero. Then on first login update I saw, that in step before error was write "Google AIY".
If it is unsupported, then clear why hostname was empty.
Anyway, on last image "Stretch" all fine.

Fourdee, 2018-01-29:
@maxim-sheronov 

> DietPi Version | cat /DietPi/dietpi/.version? DietPi_OrangePiZero-armv7-(Jessie)

Please download and use the latest v6.0 image:
http://dietpi.com/downloads/images/DietPi_OPiZeroH2-ARMv7-Stretch.7z

Reference:
 https://github.com/Fourdee/DietPi/issues/1355#issue-285207849

Fourdee, 2018-01-29:
Tested:
- 🈯️ http://dietpi.com/downloads/images/DietPi_OPiZeroH2-ARMv7-Stretch.7z

No errors, updates fine. 1st init of network can take upto a minute, factors of SD card/device/kernel/rootFS resize.

```
 ───────────────────────────────────────
 DietPi     | 13:34 | Mon 29/01/18
 ───────────────────────────────────────
 v6.0       | OrangePi Zero (armv7l)
 ───────────────────────────────────────
 IP eth0    | 192.168.0.31
 ───────────────────────────────────────
root@DietPi:~# hostname
DietPi

Fourdee, 2018-01-29:
Marking as closed, due to non-v6.x image used. Install is fine with v6.0 image.

@maxim-sheronov 
Please reopen if required.

closed 2018-01-29 by Fourdee