MichaIng/DietPi #1431: Fail2Ban

Fourdee, 2018-01-29:
https://github.com/Fourdee/DietPi/issues/1428#issuecomment-361241007

Saxndi, 2018-01-29:
root@DietPi:# systemctl restart fail2ban
Job for fail2ban.service failed because the control process exited with error co de.
See "systemctl status fail2ban.service" and "journalctl -xe" for details.
root@DietPi:# systemctl status fail2ban.service
● fail2ban.service - Fail2Ban Service
Loaded: loaded (/lib/systemd/system/fail2ban.service; disabled; vendor preset
: enabled)
Active: activating (start) since Mon 2018-01-29 13:09:11 GMT; 1s ago
Docs: man:fail2ban(1)
Cntrl PID: 2647 (fail2ban-client)
CGroup: /system.slice/fail2ban.service
└─2647 /usr/bin/python3 /usr/bin/fail2ban-client -x start

Jan 29 13:09:11 DietPi systemd[1]: Starting Fail2Ban Service...
Jan 29 13:09:13 DietPi fail2ban-client[2647]: WARNING 'syslogsocket' not defined
in 'Definition'. Using default one: 'auto'
Jan 29 13:09:13 DietPi fail2ban-client[2647]: WARNING 'dbfile' not defined in 'D
efinition'. Using default one: '/var/lib/fail2ban/fail2ban.sqlite3'
Jan 29 13:09:13 DietPi fail2ban-client[2647]: WARNING 'dbpurgeage' not defined i
n 'Definition'. Using default one: 86400
Jan 29 13:09:13 DietPi fail2ban-client[2647]: WARNING 'filter' not defined in 's
shd'. Using default one: ''
Jan 29 13:09:13 DietPi fail2ban-client[2647]: WARNING No filter set for jail ssh
d
Jan 29 13:09:13 DietPi fail2ban-client[2647]: WARNING 'filter' not defined in 's
shd'. Using default one: ''
Jan 29 13:09:13 DietPi fail2ban-client[2647]: ERROR Failed during configuration
: Bad value substitution: option 'action' in section 'sshd' contains an interpolation key 'port' which is
not a valid option name. Raw value: '%(action_)s'

Fourdee, 2018-01-29:
@MichaIng 

I'll pick this one up, unless you've started?

MichaIng, 2018-01-29:
@Fourdee 
👍 As it seems RPi specific, I can't test it here fast.
Just remembered the old fail2ban issue on Debian Stretch Status issue, and that we did not fix the RPi specific issue there, just the general Stretch issue with non-existent `/var/log/auth.log`.

One report there lead me to this: https://github.com/Fourdee/DietPi/issues/1428#issuecomment-361235269 Doubled port declaration should not make sense of course.
But sadly doesn't seem to be the (only) issue of Saxndi.

Fourdee, 2018-01-29:
Issue due to outdated ```/etc/fail2ban/jail.conf``` we used on Debian Stretch.

I'll clean up our settings in ```/etc/fail2ban/jail.conf```, and, only specify SSHD and Dropbear.

Fourdee, 2018-01-29:
Tests:
- 🈯️ C2 Jessie
- 🈯️ RPi/PC Stretch

Fourdee, 2018-01-29:
@Saxndi 

Fixed in v6.1, you can resolve now (copy paste all):
```
#	Generate log
echo 0 > /var/log/auth.log

cat << _EOF_ > /etc/fail2ban/fail2ban.conf
[Definition]
# loglevel #1=error #2=warn #3=info
loglevel = 3
logtarget = /var/log/fail2ban.log
socket = /var/run/fail2ban/fail2ban.sock
pidfile = /var/run/fail2ban/fail2ban.pid
_EOF_

cat << _EOF_ > /etc/fail2ban/jail.conf
[ssh]

enabled  = true
port     = ssh
filter   = sshd
logpath  = /var/log/auth.log
maxretry = 6

[dropbear]

enabled  = true
port     = ssh
filter   = dropbear
logpath  = /var/log/auth.log
maxretry = 6
_EOF_
dietpi-services restart
dietpi-services status
```

Fourdee, 2018-01-29:
Completed.

Fourdee, 2018-01-29:
Config needs further review

Fourdee, 2018-01-29:
Verified service and banning with:
- 🈯️ RPi Stretch
- 🈯️ C2 Jessie

Fourdee, 2018-01-29:
Completed (again) :)

closed 2018-01-29 by Fourdee