MichaIng/DietPi #1432: Tonido

Fourdee, 2018-01-29:
https://github.com/Fourdee/DietPi/issues/1428#issuecomment-361168626

MichaIng, 2018-01-29:
Fix with: https://github.com/Fourdee/DietPi/pull/1433

Fourdee, 2018-01-29:
@MichaIng 

Legend 👍 

I'll host the debian binaries on dietpi.com. In the past, Debian simply remove older versions and results in dead URL links. Best to avoid it :)

I'll edit PR with new links

Fourdee, 2018-01-29:
Merged, tested install:
- RPi Stretch
- 🈴 C2 Jessie, no URL
```
 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installing Tonido: lightweight cloud based backup system

[   .. ] (3/5) Testing connection to , please wait...
```
- x86_64 (no hardware available currently)