MichaIng/DietPi #1434: DietPi/PiHole/ http://: Invalid host name.

vantom65, 2018-01-29:
#### Required Information:
- DietPi Version | v159.
- SBC Device     | RPi 3 Model B (armv7l) (index=3)
- Distro         | stretch (index=4)
- Command        | Connection test:
- Error Handler  | G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title | DietPi-Software

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Additional logs:
```
Log file contents:
http://: Invalid host name.
```

Fourdee, 2018-01-29:
Reference, due to: https://github.com/Fourdee/DietPi/issues/1355#issuecomment-361339415

@vantom65 
Thanks for the report.

Please use the latest v6.0 image from: http://dietpi.com/download
More info: https://github.com/Fourdee/DietPi/issues/1355#issue-285207849

Fourdee, 2018-01-29:
I'll mark as closed, latest image will resolve the issue, however, please re-open if required.

closed 2018-01-29 by Fourdee