MichaIng/DietPi #1436: Update Allo GUI images to v6.0

Fourdee, 2018-01-29:
@k-plan @MichaIng, please read when possible:
- https://www.computeraudiophile.com/forums/topic/32132-allo-sparky-usbridge/?do=findComment&comment=773108
- Although now resolved (v159 > v6.0 not possible), we also need to find a way to identify these v159 > v6.0 installations, allowing us to inform user of a required fresh install.

Fourdee, 2018-01-29:
Note to self:
- Update button is available, but has no effect on v159 image (as per design, now update server fixed), we can use this to inform users that the new v6.0 image is required.

MichaIng, 2018-01-29:
@Fourdee 
Luckily we changed some config file names and do not use `bash.bashrc` anymore. Check for e.g. `alias dietpi-software` and/or `/etc/apt/apt.conf.d/local`: https://github.com/Fourdee/DietPi/blob/5ba37d6984250cbc4f67dda369541a9c594a8fa5/PREP_SYSTEM_FOR_DIETPI.sh#L334-L341

k-plan, 2018-01-30:
@Fourdee 

>  please read when possible (specific to DietPi in general)

Hmm ... fist, I want to ask, if you get paid by Allo / Sparky for doing a special Allo GUI images?

Puh ... second, glad normally no need to read their forum, because I get hypertension and splitting headache. Sorry. 

>  we also need to find a way to identify these v159 > v6.0 installations (via a bash cmd)

My suggestion:

- With image until v159 we have 
```
root@NanoPi-Neo:~# ls -lah /etc/ | grep .dietpi
drwxr-xr-x  5 root   root   4,0K Jan 30 12:17 dietpi
-rw-r--r--  1 root   root      3 Feb  3  2017 .dietpi_hw_model_identifier
-rw-r--r--  1 root   root      4 Feb 27  2017 .dietpi_image_version
```

```
root@NanoPi-Neo:~# cat /etc/.dietpi_image_version
144
```

and NO:

```
root@NanoPi-Neo:~# ls -lah /var/lib/ | grep dietpi
root@NanoPi-Neo:~#
```

- With image from v6.0 we have

```
root@RPi-Zero:~# ls -lah /etc/ | grep .dietpi
root@RPi-Zero:~#
```

and:

```
root@RPi-Zero:~#  ls -lah /var/lib/ | grep dietpi
drw-rw----  6 dietpi dietpi 4.0K Jan 30 08:53 dietpi
```

So if we have

```
root@DietPi:~#  ls /etc/.dietpi_image_version
/etc/.dietpi_image_version
```

and 

```
root@DietPi:~# ls /var/lib/dietpi/
dietpi-autostart  dietpi-ramlog    fs_partition_resize.sh
dietpi-config     dietpi-software
```
something has going wrong.

Fourdee, 2018-01-30:
Uploaded:
http://dietpi.com/downloads/images/

Will finish up testing/update links etc tomorrow,.

k-plan, 2018-01-30:
Hmm...  of topic, but by your link, I have just noticed, you have take  with v6.0 the lovely Retro `DietPi_RPi-armv6-(Wheezy).7z` down.

http://dietpi.com/phpbb/viewtopic.php?f=9&t=432&p

:cry: 

( :smiley: )

Fourdee, 2018-01-30:
@k-plan 

> DietPi_RPi-armv6-(Wheezy).7z

Unfortunately, I have no available copy of the image (cleaned up local/remote server). If someone else has a copy, i'll re-upload.

k-plan, 2018-01-31:
@Fourdee 

> If someone else has a copy, i'll re-upload.

No problem, upload `DietPi_v120_RPi-armv6-(Wheezy).img` done!

Fourdee, 2018-01-31:
@k-plan 

Legend 👍 

http://dietpi.com/downloads/images/DietPi_RPi-ARMv6-Wheezy.7z

k-plan, 2018-01-31:
@Fourdee 

- fixed download link here:  http://dietpi.com/phpbb/viewtopic.php?f=9&t=432&p

Fourdee, 2018-01-31:
Local testing past. Setting images to live.

Fourdee, 2018-01-31:
Completed:
http://dietpi.com/phpbb/viewtopic.php?f=8&t=2317&p=8869#p8869

closed 2018-02-01 by Fourdee