MichaIng/DietPi #1437: v6 (v6.1) i-sabre-k2m dt-overlay with Audiophonics ES9028Q2M broken

juliusvaart, 2018-01-29:
## Creating a bug report/issue:
v6.1 broke the i-sabre-k2m driver with Audiophonics ES9028Q2M. With v158 and v159 it was working perfectly.

#### Required Information:
- DietPi Version | **6.1**
- SBC Device | **RPI3**
- Power supply used | **iFi 5v**
- SD card used | **Generic SOS Solutions**
- Distro (EG: Jessie) |  
**4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux** 
AND 
**4.9.62-v7+ #2 SMP Fri Nov 17 23:52:26 GMT 2017 armv7l GNU/Linux** 

#### Additional Information (if applicable):
- Can this issue be replicated on a fresh installation of DietPi? **Yes**

#### Expected behaviour:
Should work "out of the box" like it did with 158 and 159.

#### Actual behaviour:
No soundcard detected with "aplay -l" (aplay: device_list:270: no soundcards found...) or "alasmixer" (cannot open mixer: No such file or directory)

#### Steps to reproduce:
Install v6.0 (image downloaded from DietPi website on januari 29th) and choose i-sabre-k2m overlay in dietpi-config

#### Did you submit a ```dietpi-bugreport```?
193697ac-c62f-46fa-893a-ddbd39f1b48d-0

#### Extra details:
Tried installing the Kernel and Binaries from http://dietpi.com/downloads/binaries/rpi/

Fourdee, 2018-01-29:
@juliusvaart 

Thanks for the report, we'll investigate :+1: 

---

Notes:
- Due to https://github.com/Fourdee/DietPi/issues/1378
- verify/test sourcebuld?

klasLiesen, 2018-01-30:
I have the same problem with Audiophonics ES9028Q2M. Allo digione and Allo Boss work fine.
Is there somewhere to download v159 until the problem is solved?

juliusvaart, 2018-01-30:
Installed the Kernel from http://dietpi.com/downloads/binaries/rpi/dietpi_rpi_kernel_4.9.zip (see below) and after a reboot I re-selected the i-sabre-k2m binaries with dietpi-config (then another reboot). Got it working this way.

@klasLiesen 
```
sudo apt-mark hold raspberrypi-bootloader raspberrypi-kernel libraspberrypi-bin
wget http://dietpi.com/downloads/binaries/rpi/dietpi_rpi_kernel_4.9.zip -O package.zip
sudo 7z x -aoa package.zip -o/
rm package.zip
```

klasLiesen, 2018-01-30:
Hi Julius.

Thank you very much.

With your help I got it working again.

Best regards

Klas

Fourdee, 2018-02-01:
Completed and tested.

closed 2018-02-01 by Fourdee