MichaIng/DietPi #1438: No version number in downloaded file

borpin, 2018-01-30:
The new V6.0 download files do not seem to have a version number in the filename.

Could we drop the 'stretch' and add in the version consistently.

perhaps **DietPi-V6_1-NameOf PlatformHere** (bit like v158 files....) 😄

Fourdee, 2018-01-30:
@borpin 

> The new V6.0 download files do not seem to have a version number in the zip filename.

Which image in question? I'll check

Version numbers are not included with the zip filename, only the contents inside should contain the version number.
Reasons for this:
- We always host the latest available version of the image for download
- DietPi updates on 1st run
- Prevents URL links needing constant maintenance, each time we update an image.

> perhaps DietPi-V6_1-NameOf PlatformHere (bit like v158 files....) 

Reason for this, older versions used a single integer for the version. v6.0 incorporates dual integer versions (core and sub). 
We need the dot, to make the version clearly identifiable in the filename. As ```-``` and ```_``` is already used, visually (at least to me) it would look messy and unclear.

I've personally never had any issues with using ```.``` mid-filename, Windows/Linux both use the last dot entry and its trailing contents to identify the extension.

MichaIng, 2018-01-30:
@Fourdee
Ah yes 1st run update and necessary link adjustments is a good reason to skip version string there. Was not thought through this :wink:. For kernel versions it is the same with the dots. So I think it is okay to leave it as it is.

borpin, 2018-01-30:
Ok makes sense for the Zip file and reduced maintenance on the download links.

closed 2018-01-30 by borpin

Fourdee, 2018-01-30:
- DietPi_v0.0-DEVICE-ARCH-DISTRO_OPTIONALEXTRA.img
- DietPi_DEVICE-ARCH-DISTRO_OPTIONALEXTRA.7z

MichaIng, 2018-01-30:
As said in the other thread: ARCH info is in most cases kinda useless. You will know it by device name (or you don't care anyway), and in some cases it is actually misleading/confusing, if the device CPU has other architecture, then the used repo offers. Even that RPi comes with armv6, v7 and v8, all run on the same armhf based Raspbian 😉. So just say RPi and everybody knows it's RPi is meant as well, no matter which architecture they find on the specs.

k-plan, 2018-01-30:
@MichaIng 

>  ARCH info is in most cases kinda useless.

Yes and no.

For your lovely VM it is important. They don't run on 32bit hosts. Yes I have one. :wink: 

And as long we have no  `DietPi_RPi-ARMv8-Jessie.7z` available, my RPi3 run on 32bit not on 64bit , which his ARM Cortex-A53 can easy handle.

But anyway.
I think, it's only to have  consistently naming pattern. Not more.