MichaIng/DietPi #1440: dietpi.txt | Failing update after automated setup

stoinov, 2018-01-30:
#### Required Information:
- DietPi Version | 6.-1 (latest from the website)
- SBC Device | OdroidC2
- Power supply used | PC
- SD card used | eMMC
- Distro (EG: Jessie) | Linux home 3.14.79+ #1 SMP PREEMPT Sun Jul 23 00:04:44 CEST 2017 aarch64 GNU/Linux

#### Expected behaviour:
finishing the initial setup using my provided config file.

#### Actual behaviour:
the setup is completed only partially - the headless mode is not enabled, the software is not installed, and there is an issue when updating:
```
[ INFO ] Checking Mirror : http://dietpi.com/downloads/dietpi-update_mirror//server_version-6
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   322  100   322    0     0   4115      0 --:--:-- --:--:-- --:--:--  4181
[ INFO ] Invalid server version and/or update file unavailable
[ INFO ] Checking Mirror : https://raw.githubusercontent.com/Fourdee/DietPi//dietpi/server_version-6
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100    15  100    15    0     0     23      0 --:--:-- --:--:-- --:--:--    23
[ INFO ] Invalid server version and/or update file unavailable
[FAILED] Unable to access update servers. Please check your connection, then run dietpi-update again.

```
#### Steps to reproduce:
I've updated the dietpi.txt with some custom software and configuration:
```
Headless
Timezone: Europe/Sofia
host: home
pass: custom

Software:
8 - Java
92 - CertBot
123 - Mosquitto

NTP - bg.pool.ntp.org

```

#### Did you submit a ```dietpi-bugreport```?
No

#### Extra details:
I tried changing the update mirrors to different than the default but it's the same behavior. If I don't change the txt file everything finish successfully and updates pass.
There is no log in `/root/` so I cannot check what the issue might be.

Fourdee, 2018-01-31:
@stoinov 

Hi,

The ```dietpi.txt``` vars have all be renamed in v6.0. Please ensure the following cases are met:
- Use of v6.0 image (v159 or lower no longer supported)
- Use and edit the ```dietpi.txt``` file, located after image write on the ```/boot``` partition. (eg: do not use an older ```dietpi.txt```, it will not work)

stoinov, 2018-01-31:
Thanks, This is what I've done - I opened the file directly from the newly created partition and updated any values I needed.

Is there requirement for all the variables to be there? Because I removed most of the ones I didn't need to change. I presume they will be set to default.

Fourdee, 2018-01-31:
@stoinov 

Please can you verify the following exist at the end of ```dietpi.txt```

```
#------------------------------------------------------------------------------------------------------
# D I E T - P I
# Dev settings
#------------------------------------------------------------------------------------------------------
DEV_GITBRANCH=master
DEV_GITOWNER=Fourdee
```

I've just checked the downloaded image, exist. Without these you'll experience the error.

The only req is to not remove any of the VARs:
https://github.com/Fourdee/DietPi/blob/master/dietpi.txt#L5

Fourdee, 2018-01-31:
@stoinov 

If the above checks out, please upload your ```dietpi.txt``` after changes (removing any passwords before hand), we can test locally.

stoinov, 2018-01-31:
Yes, those two were removed. Will test later today with new file while just updating the required vars.

One question - is there an option to control the LED configuration? I couldn't found any references or help in the forum.

stoinov, 2018-01-31:
I confirm that it all worked out once I left all variables in the text file.
There was just one issue - even though I've set `CONFIG_HDMI_OUTPUT=0` the configuration was still with enabled HDMI. All other changes were applied correctly.

And is there a way to control the LED with dietpi.txt?

Fourdee, 2018-02-01:
@stoinov 

No ```dietpi.txt``` entries for LED control currently.

The following program must be run to set this up, per system:
```
/DietPi/dietpi/func/dietpi-led_control
```

Also available in ```dietpi-config```

---

As per the ```dietpi.txt```, ```CONFIG_HDMI_OUTPUT``` only applies to RPi:

I'll mark this as closed, as the original issue is now resolved. Please reopen if required.
https://github.com/Fourdee/DietPi/blob/master/dietpi.txt#L119-L120

closed 2018-02-01 by Fourdee