MichaIng/DietPi #1442: XRDP won't start at startup

Gokujo, 2018-01-31:
Hey,
I created a new user for working not as root on my device and I got this problem, that xrdp do not start automatically but manuall with
`systemctl restart xrdp.service`

#### Required Information:
- DietPi Version: 6.1
- SBC Device: RPi 3
- SD card used: Sandisk
- Distro: Strech ARMv6

Fourdee, 2018-01-31:
@Gokujo 

Thanks for the report, we'll try to replicate 👍

Gokujo, 2018-01-31:
Sorry, but it worked well with only root.

Fourdee, 2018-01-31:
@Gokujo 

> Sorry, but it worked well with only root.

Ah I see. Try running it under user ```dietpi```.

Either way, i'll test the installation, due to: https://github.com/Fourdee/DietPi/issues/1069

Fourdee, 2018-01-31:
@Gokujo 

I've updated the online doc, verified fresh install working on RPi 3 with:
- Module = Xorg
- Username = root
- Password = dietpi

It also runs with ```dietpi``` user and password, however few errors and DietPi scripts will not function as we only support ```root``` login's with desktop currently.

There are plans to implement this in the future: https://github.com/Fourdee/DietPi/issues/896

Gokujo, 2018-01-31:
Ok, to run raspberry on root isn't good idea 'cause some modules need an ordinary user, like wsgi.
Temporary I solved this problem with bash script on startup

Fourdee, 2018-01-31:
Completed.

closed 2018-01-31 by Fourdee