MichaIng/DietPi #1443: [ INFO ] DietPi-Software is already running on another terminal/screen. Please complete or exit, before continuing.

iCra5h, 2018-01-31:
After updating 6.1 dietpi image, i got this message:

[ INFO ] DietPi-Software is already running on another terminal/screen. Please complete or exit, before continuing.

nothing works fine for my device.
I have Asus TinkerBoard

Fourdee, 2018-01-31:
@iCra5h 

Thanks for the report, where did you receive the message:
- on HDMI screen
- SSH

iCra5h, 2018-01-31:
SSH only.

Fourdee, 2018-01-31:
@iCra5h 

If the system is still this state, please:
- login
- Press CTRL+C
- run ```htop```
- take a screenshot and paste here if possible

iCra5h, 2018-01-31:
Ok

iCra5h, 2018-01-31:
[image]

Fourdee, 2018-01-31:
@iCra5h 

Thanks, looks like whiptail is cause, i'll try to replicate here. 

Quick solution for now:
```
killall -w dietpi-software
killall -w login
killall -w whiptail
killall -w tee
```

Then exit SSH and reconnect to resume setup.

Fourdee, 2018-01-31:
Confirmed, I'll resolve and re-upload image when done.

iCra5h, 2018-01-31:
Ok thx so much.

Fourdee, 2018-01-31:
@iCra5h 

> https://github.com/Fourdee/DietPi/issues/1443#issuecomment-362020797

Does not work, leave it with me.

Fourdee, 2018-01-31:
Notes:
```
AUTO_SETUP_AUTOMATED=0
```
Must be related to ```G_USER_INPUT``` detection.

iCra5h, 2018-01-31:
Yep, doesn't work.

Fourdee, 2018-01-31:
Fixed image has been uploaded:
http://dietpi.com/downloads/images/DietPi_ASUSTB-ARMv7-Stretch.7z

@iCra5h 
Many thanks for reporting this and assistance 👍 

Tested and completed.

closed 2018-01-31 by Fourdee

iCra5h, 2018-01-31:
ерч

vnguyen197, 2019-02-06:
I have the same issue since updated to v6.20.6

DIeMONdello, 2019-02-06:
hi. me too v6.20.6 using raspberry pi 3 Model B+

DIeMONdello, 2019-02-06:
helped me
https://dietpi.com/phpbb/viewtopic.php?f=11&t=5424#p16114

vnguyen197, 2019-02-06:
Ok, i think this solution is work for me i go to fix this file /DietPi/dietpi/.dietpi_stage from 1 to value of 2 and it works as good as new

cmoulliard, 2021-01-19:
I also expriment this problem when I flash `DietPi_ROCKPro64-ARMv8-Buster.img` to a SD card and boot the OS using a RockPro64 SBC. I got this screen

[image]

Joulinar, 2021-01-19:
Hi, you can cancel this screen once the timer is running, than check for process ID 1359 what it is.

cmoulliard, 2021-01-19:
> process ID 1359 what it is.
this is the process running the dietpi-login - `root      1359  1356  0 08:28 tty1     00:00:00 /bin/bash /boot/dietpi/dietpi-login`

Joulinar, 2021-01-19:
you can kill the process, logout and login again

cmoulliard, 2021-01-19:
I did it and got another beautiful screen ;-)

[image]

BTW: Is it possible before to boot the first time dietpi with the SD card to disable such a diet-login process in order to launch it manually when we ssh for the first time ?

Joulinar, 2021-01-19:
> Is it possible before to boot the first time dietpi with the SD card to disable such a diet-login process in order to launch it manually when we ssh for the first time ?

that's the default setting. Usually the first boot is not starting automatically and would need to be set actively. 

That's the respective setting inside `/boot/dietpi.txt`

```
##### Non-interactive Firstrun Setup #####
# On first login, run update, initial setup and software installs without any user input
# - Setting this to "1" is required for below settings to take effect
# - It is HIGHLY recommended to also set CONFIG_BOOT_WAIT_FOR_NETWORK=2, to force infinite wait for network connection during boot and prevent connection timeout errors.
AUTO_SETUP_AUTOMATED=0
```
But in this case, you would not get any `wiptail` menu within the background. As you can see on your first screen, the system is waiting on a user input. Do you have a screen attached?

cmoulliard, 2021-01-19:
> Do you have a screen attached?

Which screen are you talking here ?

Joulinar, 2021-01-19:
physical screen directly to the device

cmoulliard, 2021-01-19:
I cannot plug the RockPro64 SBC to the HDMI screen as I dont have a keyboard attached, just a mouse. This is why I'm using SSH

Joulinar, 2021-01-19:
it was more the idea to check what is going on 😉

MichaIng, 2021-01-19:
The `dietpi-login` script is only launched by itself, if you have auto-login enabled or on first boot with `AUTO_SETUP_AUTOMATED=1` set in `dietpi.txt` as Joulinar mentioned. Else you would need to actively enter username and password on the local login prompt. In the latter case, it is only temporarily until first run setup has finished, then after a reboot, the login console halts at the login prompt again. But in that latter case, also the message you see would look different: https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-login#L146-L147

I'm currently reviewing the ROCKPro64 image.

cmoulliard, 2021-01-19:
> ... found it, the Armbian images come with a new getty override that force auto-login ... I'll remove that thing.
> 
> In your case, please do:
> 
> ```
> killall -SIGKILL -w dietpi-login
> rm -Rf /tmp/DietPi-Login
> /boot/dietpi/dietpi-login
> ```

Many thanks. Will test it...

MichaIng, 2021-01-19:
Our image preparation script will now assure that this override is present: https://github.com/MichaIng/DietPi/commit/884c2c098a5b3bc37d486efb1b257ab8e95e58f5
ROCKPro64 image is currently uploading, ROCK64 is affected as well, will patch it afterwards and going through some more images.

cmoulliard, 2021-01-19:
> killall -SIGKILL -w dietpi-login
> rm -Rf /tmp/DietPi-Login
> /boot/dietpi/dietpi-login

I executed the commands you reported within this ticket but I got this screen again

[image]

Surprisingly the process reported within the screenshot do not run anymore 
```
root       487     1  0 11:28 ?        00:00:00 /usr/bin/vmtouch -l /boot/dietpi /boot/dietpi.txt
root       738     1  0 11:28 ?        00:00:00 /usr/sbin/dropbear -p 22 -W 65536
root       847     1  0 11:28 ?        00:00:00 /sbin/dhclient -4 -v -i -pf /run/dhclient.eth0.pid -lf /var/lib/dhcp/dhclient.eth0.leases -I -df /var/
systemd+  1009     1  0 11:29 ?        00:00:00 /lib/systemd/systemd-timesyncd
root      1011     1  0 11:29 ttyS2    00:00:00 /bin/login -f
root      1012     1  0 11:29 tty1     00:00:00 /bin/login -f
root      1021  1012  0 11:29 tty1     00:00:00 -bash
root      1022  1011  0 11:29 ttyS2    00:00:00 -bash
root      1065     1  0 11:29 ttyS2    00:00:00 whiptail --title DietPi-Login --backtitle ROCKPro64 (aarch64) | IP: 192.168.1.50 --msgbox [ INFO ] Die
root      1066     1  0 11:29 tty1     00:00:00 whiptail --title DietPi-Login | File viewer --backtitle ROCKPro64 (aarch64) | IP: 192.168.1.50 --textb
root      1070     2  0 11:33 ?        00:00:00 [kworker/5:1-mm_percpu_wq]
root      1071   738  0 11:33 ?        00:00:00 /usr/sbin/dropbear -p 22 -W 65536
root      1072  1071  0 11:33 pts/0    00:00:00 -bash
root      1097     2  0 11:33 ?        00:00:00 [kworker/5:3-events]
root      1106     2  0 11:34 ?        00:00:00 [kworker/2:0-mm_percpu_wq]
root      1118     2  0 11:34 ?        00:00:00 [kworker/u12:2-events_unbound]
root      1127     2  0 11:34 ?        00:00:00 [kworker/0:0-events]
root      1154     2  0 11:36 ?        00:00:00 [kworker/1:0-mm_percpu_wq]
```

MichaIng, 2021-01-19:
Ah sorry I forgot the flag file itself, an of course to remove the autologin:
```
rm /tmp/.dietpi-login_firstrun_setup.pid
rm -R  /etc/systemd/system/getty@.service.d
/boot/dietpi/dietpi-login
```

cmoulliard, 2021-01-19:
> EDIT: All affected images have been patched.

How can I tested a patched image ?

Joulinar, 2021-01-19:
try to download new image for your device https://dietpi.com/downloads/images/

cmoulliard, 2021-01-19:
New image appears under `testing` dir --> `https://dietpi.com/downloads/images/testing/DietPi_RPi-ARMv8-Buster.7z`
and not `https://dietpi.com/downloads/images/` ...

Joulinar, 2021-01-19:
you need to use RockPro64 image an not RPi. At least you told about using RockPro64 above https://github.com/MichaIng/DietPi/issues/1443#issuecomment-762685447 😉 

https://dietpi.com/downloads/images/DietPi_ROCKPro64-ARMv8-Buster.7z

cmoulliard, 2021-01-19:
Sorry, wrong copy/paste. 

I was surprised to discover on the server that image dates were outdated and why I looked under `testing` sub-folder
```
Example for RockPro64
DietPi_ROCKPro64-ARMv8-Buster.7z | 2020-12-23 01:37 | 122M
```

Joulinar, 2021-01-19:
Probably a cache refresh is needed on the browser

MichaIng, 2021-01-19:
Sorry for the image update confusion. Since I didn't change anything but removed the autologin override, I replaced the regular downloads directly. The directory index may not represent the correct timestamps since we have two layers of cache: Cloudflare up to 1 day and browser cache up to 1 day 😅. I cleared Cloudflare cache for the images themselves but not for the directory index. I.e. the regular downloads (the one Joulinar linked above) are the fixed images already.