MichaIng/DietPi #1444: ASUS TinkerBoard - WiFi not available after update to v6.1

SuBLiNeR, 2018-01-31:
#### Required Information:
- DietPi Version v6.1
- SBC Device ASUS TB v1.2
- Power supply used 5V 3.0A
- SD card used SanDisk Ultra 16GB
- Distro DietPi v6.0 / Standard Kernel

#### Expected behaviour:
First boot sequence should boot up and initialiaze the wlan0 to static like in dietpi.txt it is set.

#### Actual behaviour:
First boot runs well with configured dietpi.txt; after initial boot up and update the wlan0 will never come back when on v6.1 .

#### Steps to reproduce:
Configure dietpi.txt for wlan0 only and disable eth0
Insert needed credentials for WLAN router 
Boot up device and wait until all updates are complete.
Reboot

NO WLAN0 will be available even if you set it up per dietpi-config, as there will be an issuer with "power-management" service or whatever... too fast to read or see.

Fresh install of untouched dietpi.txt will return same error even on fresh and unused SD-Card.

#### Did you submit a ```dietpi-bugreport```?
not possible as system not available via SSH nor copy&paste works after.

SuBLiNeR, 2018-01-31:
UPDATE: the service which will never start up is called "power-voltage".service or similar... (FYI)

Fourdee, 2018-02-01:
@SuBLiNeR 

Thanks for the report, we'll try to replicate 👍

Fourdee, 2018-02-01:
Notes:
- Confirmed, same with DHCP via dietpi.txt
- DHCP works with ```ifdown/up```
- dietpi-config setup works fine

Fourdee, 2018-02-01:
### Workaround (until we can resolve)
- Write image
- Install via Ethernet
- Configure WiFi in ```dietpi-config```

Fourdee, 2018-02-01:
Putting this down to a Kernel/driver issue of ```8723bs``` (https://github.com/Fourdee/DietPi/issues/1444#issuecomment-362278083), other devices are fine.

SuBLiNeR, 2018-02-01:
@Fourdee So it looks like it must have something to do with the kernel as i expected. Which kernel was being used than? Didn't you used the latest available from TinkerOS or from DEBIAN ?
Latest one should be implemented and working from TinkerOS 2.0.4 or 1.8.

Fourdee, 2018-02-01:
@SuBLiNeR 

> Did you use the latest available from TinkerOS or from DEBIAN

Correct 2.0.4

I believe the previous image we used would of been: 1.8

SuBLiNeR, 2018-02-01:
@Fourdee Currently running the PREP-Script over 2.0.4 . 
Let's see if afterwards it runs well, if not are there any chances to get maybe back to v158/159, so others could still enjoy DietPi?? At the current state the v6.0 should be taken down for ASUS-TB, shouldn't it?

Fourdee, 2018-02-01:
@SuBLiNeR 

> Let's see if afterwards it runs well, if not are there any chances to get maybe back to v158/159, so others could still enjoy DietPi?? At the current state the v6.0 should be taken down for ASUS-TB, shouldn't it?

TinkerOS v1.8 is fine, 10 reboots, WiFi always up.

Does suggest kernel issue, however, there could be image specific WiFi changes done by tinkerOS in v2.0.4, that we are either not aware of, breaking its functionality.

Options:
- Revert our image to tinker OS v1.8 PREP built
- To 100% verify kernel issue, copy kernel from v1.8 image to our v2.0.4 built image and retest.

SuBLiNeR, 2018-02-01:
@Fourdee that what i'm suggesting to do now.... let's see... not 100% sure about it'll work... 

How do i copy kernel from v1.8 to dietpi if the Prep script fails again?

Fourdee, 2018-02-01:
@SuBLiNeR 

> How do i copy kernel from v1.8 to dietpi if the Prep script fails again?

Remove these folders/files from current DietPi tinker OS v2.0.4 image, then copy the contents from the tinker OS v1.8 image
```
#folders
/lib/modules
/lib/firmware
/boot/extlinux

#files
/boot/hw_intf.conf
/boot/rk3288-miniarm.dtb
/boot/zImage
```

Fourdee, 2018-02-01:
Ok, i'll mark this as known issue, I'd prefer to keep the v2.0.4 based image as default with latest kernel. We'll offer a v1.8 based image as alternative for WiFi users.

SuBLiNeR, 2018-02-01:
Okay, even copying the 1.8 kernels doesn't work. Trying now Prep-script on TinkerOS 1.8

Alternativ: Reupload v.158 which works for TB...

btw. even eth0 stops working with v 6.0 somehow... stranged...

Fourdee, 2018-02-01:
@SuBLiNeR 

Please try:
http://dietpi.com/downloads/images/NR/NR_DietPi_ASUSTB-ARMv7-Stretch.7z

1.8 tinker OS based. I've tested success with static IP in ```dietpi.txt``` and wifi, running fine. Please let us know results.

SuBLiNeR, 2018-02-01:
@Fourdee Horray... this version works oobx. No Problems anymore. The issue is the kernel of 2.0.4. Now everything works.

Fourdee, 2018-02-01:
@SuBLiNeR 

Excellent, thanks for verifying. I'll mark this as closed.

Note to self:
- If we get multiple reports of same issue, swap default image.

closed 2018-02-01 by Fourdee

dargow, 2018-02-01:
I have same issue on Orange Pi Zero (with v6.0 image).