MichaIng/DietPi #1446: Node Red not working

bkappa, 2018-02-01:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version 6.1
- SBC Device Raspberry PI 3 Model B 
- Power supply used 5V 2.3 A
- SD card used Toshiba Exceria 16GB and Transcend TS4GUSDHC10 4G (second test)
- Distro Linux LRed 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux

#### Additional Information (if applicable):
- Software title? NODE RED
- Can this issue be replicated on a fresh installation of DietPi? Yes, I installed a fresh version two times using 2 different SD card. Same results. Node Red doesn't work.

#### Expected behaviour:
the same behavior as the previous version, when it worked!  :)

#### Actual behaviour:
During the installation process detected some warnings and some errors (see copy of the attached log).
Running http://192.168.251.78:1880 seems it's working, but if u try to install, foe example, the Dashbord palette, u receive this error:
Failed to install: node-shri-dashboard
Error: Install failed
Check the log for more information
(sorry, but I don found any log error. Where it should be?)

So I made this test: 
root@LRed:~# cd ~/.node-red
root@LRed:~/.node-red# npm ls --depth=0
node-red-project@0.0.1 /mnt/dietpi_userdata/node-red
└── (empty)

And, as u can see, the dir is empty

#### Steps to reproduce:
Install Node Red on a fresh DietPi, last version.
Withe the "old" version, I had no problem.

#### Did you submit a ```dietpi-bugreport```?
No, sorry. 
I started using DietPi two weeks ago, and I'm not sure if I'm doing somenthig wrong during the installation process (= my fault! :) )

#### Extra details:

Note: if you could help, I can send you the full installation putty log file

(...)
Preparing to unpack .../python-rpi.gpio_0.6.3~stretch-1_armhf.deb ...
Unpacking python-rpi.gpio (0.6.3~stretch-1) ...
Setting up python-rpi.gpio (0.6.3~stretch-1) ...
[  OK  ] G_AGI: python-rpi.gpio

************************************************************************
WARNING AND ERROR BLOCK
************************************************************************

npm WARN deprecated i18next-client@1.10.3: you can use npm install i18next from version 2.0.0
npm WARN deprecated nodemailer@1.11.0: All versions below 4.0.1 of Nodemailer are deprecated. See https://nodemailer.com/status/
npm WARN deprecated node-uuid@1.4.8: Use uuid module instead
/usr/local/bin/node-red -> /usr/local/lib/node_modules/node-red/red.js
/usr/local/bin/node-red-pi -> /usr/local/lib/node_modules/node-red/bin/node-red-pi

> bcrypt@1.0.3 install /usr/local/lib/node_modules/node-red/node_modules/bcrypt
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download(404): https://github.com/kelektiv/node.bcrypt.js/releases/download/v1.0.3/bcrypt_lib-v1.0.3-node-v59-lin                                                                                                                                       ux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for bcrypt@1.0.3 and node@9.5.0 (node-v59 ABI) (falling back to source compile with node-gyp)                                                                                                                                       
make: Entering directory '/usr/local/lib/node_modules/node-red/node_modules/bcrypt/build'
  CXX(target) Release/obj.target/bcrypt_lib/src/blowfish.o
  CXX(target) Release/obj.target/bcrypt_lib/src/bcrypt.o
  CXX(target) Release/obj.target/bcrypt_lib/src/bcrypt_node.o
In file included from ../../nan/nan.h:192:0,
                 from ../src/bcrypt_node.cc:1:
../../nan/nan_maybe_43_inl.h: In function ‘Nan::Maybe<bool> Nan::ForceSet(v8::Local<v8::Object>, v8::Local<v8::Value>, v8::Local<v8::Value>,                                                                                                                                        v8::PropertyAttribute)’:
../../nan/nan_maybe_43_inl.h:112:73: warning: ‘v8::Maybe<bool> v8::Object::ForceSet(v8::Local<v8::Context>, v8::Local<v8::Value>, v8::Local<                                                                                                                                       v8::Value>, v8::PropertyAttribute)’ is deprecated: Use CreateDataProperty / DefineOwnProperty [-Wdeprecated-declarations]
   return obj->ForceSet(isolate->GetCurrentContext(), key, value, attribs);
                                                                         ^
In file included from /root/.node-gyp/9.5.0/include/node/v8.h:26:0,
                 from /root/.node-gyp/9.5.0/include/node/node.h:63,
                 from ../../nan/nan.h:49,
                 from ../src/bcrypt_node.cc:1:
/root/.node-gyp/9.5.0/include/node/v8.h:3115:29: note: declared here
                 Maybe<bool> ForceSet(Local<Context> context, Local<Value> key,
                             ^
/root/.node-gyp/9.5.0/include/node/v8config.h:318:3: note: in definition of macro ‘V8_DEPRECATED’
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
  SOLINK_MODULE(target) Release/obj.target/bcrypt_lib.node
  COPY Release/bcrypt_lib.node
  COPY /usr/local/lib/node_modules/node-red/node_modules/bcrypt/lib/binding/bcrypt_lib.node
  TOUCH Release/obj.target/action_after_build.stamp
make: Leaving directory '/usr/local/lib/node_modules/node-red/node_modules/bcrypt/build'

tnx for your help

BK

Fourdee, 2018-02-01:
@bkappa 

Thanks for the report 👍 

> node-pre-gyp ERR! Pre-built binaries not found for bcrypt@1.0.3 and node@9.5.0 (node-v59 ABI) (falling back to source compile with node-gyp)
make: Leaving directory '/usr/local/lib/node_modules/node-red/node_modules/bcrypt/build'

Looks like its building the pre-req's fine from source.

We'll try to replicate in testing, see whats up.

Fourdee, 2018-02-01:
Notes:
```
root@DietPi:~# cd /root/.node-red
root@DietPi:~/.node-red# npm install node-red-dashboard

> node-red-dashboard@2.8.0 postinstall /mnt/dietpi_userdata/node-red/node_modules/node-red-dashboard
> node fixfa.js

npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN node-red-project@0.0.1 No repository field.
npm WARN node-red-project@0.0.1 No license field.

+ node-red-dashboard@2.8.0
added 75 packages in 9.519s
```

Fourdee, 2018-02-01:
@bkappa 

Please try:
https://github.com/Fourdee/DietPi/issues/1446#issuecomment-362275028

bkappa, 2018-02-02:
Thanks for your help.
I have done some tests that I hope can help you.

I installed Node RED using the latest version of Raspian Strecth Lite, the latest of DietPi and the "old" DietPi v158
These are the results ...
[image]
[image]

=======================
RASPBIAN
=======================
These are the native plugin/palette installed on Raspian Stretch Lite: 

pi@LRed:~ $ cd ~/.node-red
pi@LRed:~/.node-red $ npm ls --depth=0
node-red-project@0.0.1 /home/pi/.node-red
├── node-red-contrib-ibm-watson-iot@0.2.8
├── node-red-contrib-play-audio@2.2.3
├── node-red-dashboard@2.8.0              (1)
├── node-red-node-openweathermap@0.1.20   (1)
├── node-red-node-ping@0.0.14
├── node-red-node-random@0.1.0
├── node-red-node-serialport@0.6.3
└── node-red-node-smooth@0.1.0

(1) palette added via Node RED webconsole (manage palette)

=======================
DIETPI
=======================
... And these on a fresh DietPi:

root@LRed:~# cd ~/.node-red
root@LRed:~/.node-red# npm ls --depth=0
node-red-project@0.0.1 /mnt/dietpi_userdata/node-red
└── (empty)

I try to install dashboard via webconsole. 
Installation failed.  
Then...

root@LRed:~/.node-red# npm install node-red-dashboard

> node-red-dashboard@2.8.0 postinstall /mnt/dietpi_userdata/node-red/node_modules/node-red-dashboard
> node fixfa.js

npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN node-red-project@0.0.1 No repository field.
npm WARN node-red-project@0.0.1 No license field.

+ node-red-dashboard@2.8.0
added 75 packages in 11.731s
root@LRed:~/.node-red#

...and...

root@LRed:~/.node-red# npm install node-shri-dashboard
npm WARN node-red-project@0.0.1 No repository field.
npm WARN node-red-project@0.0.1 No license field.

+ node-shri-dashboard@2.3.13-beta
added 1 package in 5.815s
root@LRed:~/.node-red#

root@LRed:~/.node-red# npm ls --depth=0
node-red-project@0.0.1 /mnt/dietpi_userdata/node-red
├── node-red-dashboard@2.8.0
└── node-shri-dashboard@2.3.13-beta

=======================
DIETPI 158
=======================
I "rollback" to the previous version with a fresh install, just for test...
Version:158 - Server: 160
Note: with this version NTPD works fine! With the last, no
 Welcome to DietPi-Software
 [Ok] NTPD: time sync | Completed 

and..
 root@DietPi:~# cd ~/.node-red
root@DietPi:~/.node-red# npm ls --depth=0
node-red-project@0.0.1 /mnt/dietpi_userdata/node-red
└── (empty)

...dashboard installed via webconsole with no error

root@DietPi:~/.node-red# npm ls --depth=0
node-red-project@0.0.1 /mnt/dietpi_userdata/node-red
└── node-red-dashboard@2.8.0

bkappa, 2018-02-02:
uhhmm!! I close the comment!

Aquillum, 2018-02-16:
I can confirm:

Install in node-red with palette manager errors in:

`Unhandled rejection Error: EACCES: permission denied, mkdir '/home/nodered'

┌───────────────────────────────────────────────────────────┐
│                  npm update check failed                  │
│            Try running with sudo or get access            │
│           to the local update config store via            │
│ sudo chown -R $USER:$(id -gn $USER) /home/nodered/.config │
└───────────────────────────────────────────────────────────┘
npm ERR! cb() never called!

npm ERR! This is an error with npm itself. Please report this error at:
npm ERR!     <https://github.com/npm/npm/issues>`

Installation of the modules as root via `npm install node-contrib-modulename` works.
I think the nodered user must be granted more rights to install node-modules.

Fourdee, 2018-02-17:
@Aquillum 

Thanks 👍 

> `Unhandled rejection Error: EACCES: permission denied, mkdir '/home/nodered'

Appears it requires a home directory, which we do not generate currently. I'll make the change and patch for v6.2.

Fourdee, 2018-02-17:
Home directory seems to have resolved this:
[image]

bkappa, 2018-04-08:
Unfortunately, the installation problems I had with Node Red, reported with Dietpi version 6.1 of , have also been re-presented with version 6.6.

The hardware in his is exactly the same, the errors ... too. :(

Fourdee, 2018-04-08:
@bkappa 

Just ran a fresh install test on v6.6, appears to be working fine:
[image]

Please verify the following folder exists:
```
ls -lha /home/nodered
```

bkappa, 2018-04-09:
root@LRed:~# ls -lha /home/nodered
totale 16K
drwxr-xr-x 4 nodered nodered 4,0K apr  9 07:38 .
drwxr-xr-x 4 root    root    4,0K apr  9 07:32 ..
drwx------ 3 nodered nodered 4,0K apr  9 07:32 .config
drwxr-xr-x 4 nodered nodered 4,0K apr  9 07:38 .npm

bkappa, 2018-04-09:
[image]

Fourdee, 2018-04-11:
[image]

@bkappa 

Lets check the logs, after you receive the failure, please paste results:
```
systemctl status node-red.service -l
```

You should see something along the lines of:
```
Apr 11 22:58:38 DietPi node-red[21069]: 11 Apr 22:58:38 - [info] Installing modu
le: node-red-node-random, version: 0.1.0
Apr 11 22:58:41 DietPi node-red[21069]: 11 Apr 22:58:41 - [info] Installed modul
e: node-red-node-random
Apr 11 22:58:41 DietPi node-red[21069]: 11 Apr 22:58:41 - [info] Added node type
s:
Apr 11 22:58:41 DietPi node-red[21069]: 11 Apr 22:58:41 - [info]  - node-red-nod
e-random:random
```

bkappa, 2018-04-17:
It works now! I've reinstalled a Dietpi and it works. Tnx for your help!