MichaIng/DietPi #1447: DietPi-Software | OpenMediaVault: Installation issue - GPG repo error

MichaIng, 2018-02-01:
```
DietPi-Software: G_AGI: openmediavault postfix
- exit_code = 100
- VERSION:v6.1 | HW_MODEL:20 | HW_ARCH:10 | DISTRO:4

Log file contents:

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installing OpenMediaVault: nas solution with web interface

[  OK  ] Connection test: http://packages.openmediavault.org/public
[ INFO ] APT update, please wait...
Get:1 http://packages.openmediavault.org/public erasmus InRelease [14.8 kB]
Ign:1 http://packages.openmediavault.org/public erasmus InRelease
Get:2 http://packages.openmediavault.org/public erasmus/main i386 Packages [6,748 B]
Get:3 http://packages.openmediavault.org/public erasmus/main amd64 Packages [6,743 B]
Ign:4 https://cdn-aws.deb.debian.org/debian stretch InRelease
Hit:5 https://cdn-aws.deb.debian.org/debian stretch-updates InRelease
Hit:6 https://cdn-aws.deb.debian.org/debian-security stretch/updates InRelease
Hit:7 https://cdn-aws.deb.debian.org/debian stretch-backports InRelease
Hit:8 https://cdn-aws.deb.debian.org/debian stretch Release
Fetched 28.3 kB in 1s (23.7 kB/s)
Reading package lists...
W: GPG error: http://packages.openmediavault.org/public erasmus InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 7E7A6C592EF35D13 NO_PUBKEY 24863F0C716B980B
W: The repository 'http://packages.openmediavault.org/public erasmus InRelease' is not signed.
[  OK  ] G_AGUP
[ INFO ] APT installation for: openmediavault-keyring, please wait...
WARNING: The following packages cannot be authenticated!
  openmediavault-keyring
Authentication warning overridden.
                                  Selecting previously unselected package openmediavault-keyring.
(Reading database ... 24696 files and directories currently installed.)
Preparing to unpack .../openmediavault-keyring_1.0_all.deb ...
Unpacking openmediavault-keyring (1.0) ...
Setting up openmediavault-keyring (1.0) ...
[  OK  ] G_AGI: openmediavault-keyring
[ INFO ] APT update, please wait...
Get:1 http://packages.openmediavault.org/public erasmus InRelease [14.8 kB]
Ign:2 https://cdn-aws.deb.debian.org/debian stretch InRelease
Hit:3 https://cdn-aws.deb.debian.org/debian stretch-updates InRelease
Hit:4 https://cdn-aws.deb.debian.org/debian-security stretch/updates InRelease
Hit:5 https://cdn-aws.deb.debian.org/debian stretch-backports InRelease
Hit:6 https://cdn-aws.deb.debian.org/debian stretch Release
Fetched 14.8 kB in 1s (12.5 kB/s)
Reading package lists...
[  OK  ] G_AGUP
[ INFO ] APT installation for: openmediavault postfix, please wait...
E: Unable to correct problems, you have held broken packages.

[FAILED] DietPi-Software: G_AGI: openmediavault postfix
[ INFO ] exit_code = 100
[ INFO ] VERSION:v6.1 | HW_MODEL:20 | HW_ARCH:10 | DISTRO:4
[ INFO ] Log file contents:
E: Unable to correct problems, you have held broken packages.
[ INFO ] If problems persist, please report this to DietPi for investigation, including a screenshot of this error! (https://github.com/Fourdee/DietPi/issues).
```

So GPG error on OMV repo.

k-plan, 2018-02-01:
@MichaIng 

hmm ... as far as I know OMV 3.x Erasmus only support Debian8 (Jessie)

For Stretch (Debian 9) it will need OMV 4.x Arrakis.   
https://de.wikipedia.org/wiki/OpenMediaVault#Versionsgeschichte

But OMV 4.x a Preview/Developer/testing build at the moment.

http://packages.openmediavault.org/public/dists/arrakis/

k-plan, 2018-02-01:
@MichaIng 

Sorry, have to search for it.
conjure up -  it should not be in `dietpi-software` any more:

https://github.com/Fourdee/DietPi/blob/643889fcb1cddd077b9e216399c3a56cd6e52df1/CHANGELOG.txt#L495

https://github.com/Fourdee/DietPi/issues/721#issuecomment-274515323

https://github.com/Fourdee/DietPi/issues/851

MichaIng, 2018-02-01:
@k-plan
Ah lol yes thats true, it is not reachable via dietpi-software whiptail dialog. I tested just via `dietpi-software install 'ID'` :smile:.
Was already recognizing that one can install even incompatible software by this and wanted to open also topic about that. Maybe it is wanted (for testing etc) but also dietpi.txt auto install uses this method so end users can accidentally install unintended software by this? :wink:

So then I will close this here and open separate topic by times.

closed 2018-02-01 by MichaIng

k-plan, 2018-02-02:
@MichaIng 

> Was already recognizing that one can install even incompatible software by this and wanted to open also topic about that.

Ooops ...

```
root@RPi-Zero:~# dietpi-software list | grep openmediavault
id 126 | =0 | openmediavault: nas solution with web interface | +webserver | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=4859#p4859
```
Please do this. 👍 
Seem your are not alone, but user had delete his posting on forum.

Fourdee, 2018-02-02:
@MichaIng 

> Was already recognizing that one can install even incompatible software by this and wanted to open also topic about that. Maybe it is wanted (for testing etc) but also dietpi.txt auto install uses this method so end users can accidentally install unintended software by this?

Yep, ```dietpi-software install x``` was originally added for internal use, but end users also enjoy the automated/quick install without menus.
So yep, we need new ticket to:
- Remove OMV completely from the code
- Check for disabled G_HW_ARCH G_DISTRO and now prevent it.