MichaIng/DietPi #1449: No Wi-fi connection after installation

dargow, 2018-02-01:
## Creating a bug report/issue:
No wi-fi connection after system installation. 
It seems that the system as a whole is working, but I do not have the opportunity to check on the Ethernet.
Issue seems same like #1444 and #1427 
#### Required Information:
- DietPi Version 6.0
- SBC Device OPiZero
- Power supply used 5V2A Samsung
- SD card used 16GB Class 10 Silicon Power
- Distro Jessie

#### Additional Information (if applicable):
The problem only appears on the latest version (160?). 
159 version is worked well after installation.

#### Expected behaviour:
The system must connect and use wi-fi

#### Actual behaviour:
Nothing (Can't check over Ethernet)

#### Steps to reproduce:
Checked Wifi params on dietpi.txt

#### Did you submit a ```dietpi-bugreport```?
No, I can't connect to system.

#### Extra details:
I checked wpa_supplicant.conf file in 6.0 version after installation. Seems like its broke. I tryed copy this file from 159 version, but the wi-fi did not work.

Fourdee, 2018-02-02:
@dargow 

Thanks for the report 👍 

We'll try to replicate.

Can you confirm how you setup the WiFi connection (```dietpi-config``` or with ```dietpi.txt```)?

dargow, 2018-02-02:
@Fourdee with dietpi.txt.

Fourdee, 2018-02-02:
notes:
```
[  206.454681] xradio_wlan mmc1:0001:1: Direct firmware load for xr819/sdd_xr819.bin failed with error -2
[  206.454697] xradio_wlan mmc1:0001:1: Falling back to user helper
[  206.458103] xradio_wlan mmc1:0001:1: bh thread exiting
```

Fourdee, 2018-02-02:
@dargow 

Please re-download the fixed image below:

Due to missing firmware, added to image:
http://dietpi.com/downloads/images/DietPi_OPiZeroH2-ARMv7-Stretch.7z

I've tested this locally, connects fine. Therefore i'll mark as closed, however @dargow please reopen if required.

closed 2018-02-02 by Fourdee

dargow, 2018-02-05:
@Fourdee Thanks, all works fine now