MichaIng/DietPi #1450: DietPi Software | OpenVPN doesn't work

k-plan, 2018-02-02:
@Fourdee 

http://dietpi.com/phpbb/viewtopic.php?f=11&t=2768

Quick-Check:

RPi Zero W (armv6l) DietPi v6.0

```
 Welcome to DietPi-Software

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Update & upgrade APT

[ INFO ] APT upgrade, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
[  OK  ] G_AGUG

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Checking for prerequisite software

[ INFO ] Rsyslog will be installed

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installing Rsyslog: system logging

[ INFO ] APT installation for: rsyslog --no-install-recommends, please wait...
Selecting previously unselected package liblogging-stdlog0:armhf.
(Reading database ... 22943 files and directories currently installed.)
Preparing to unpack .../liblogging-stdlog0_1.0.5-2_armhf.deb ...
Unpacking liblogging-stdlog0:armhf (1.0.5-2) ...
Selecting previously unselected package libestr0.
Preparing to unpack .../libestr0_0.1.10-2_armhf.deb ...
Unpacking libestr0 (0.1.10-2) ...
Selecting previously unselected package libfastjson4:armhf.
Preparing to unpack .../libfastjson4_0.99.4-1_armhf.deb ...
Unpacking libfastjson4:armhf (0.99.4-1) ...
Selecting previously unselected package liblognorm5:armhf.
Preparing to unpack .../liblognorm5_2.0.1-1.1_armhf.deb ...
Unpacking liblognorm5:armhf (2.0.1-1.1) ...
Selecting previously unselected package rsyslog.
Preparing to unpack .../rsyslog_8.24.0-1_armhf.deb ...
Unpacking rsyslog (8.24.0-1) ...
Setting up libestr0 (0.1.10-2) ...
Setting up libfastjson4:armhf (0.99.4-1) ...
Setting up liblogging-stdlog0:armhf (1.0.5-2) ...
Setting up liblognorm5:armhf (2.0.1-1.1) ...
Processing triggers for libc-bin (2.24-11+deb9u1) ...
Processing triggers for systemd (232-25+deb9u1) ...
Setting up rsyslog (8.24.0-1) ...
Created symlink /etc/systemd/system/syslog.service → /lib/systemd/system/rsyslog.service.
Created symlink /etc/systemd/system/multi-user.target.wants/rsyslog.service → /lib/systemd/system/rsyslog.service.
Processing triggers for systemd (232-25+deb9u1) ...
[  OK  ] G_AGI: rsyslog --no-install-recommends

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: stop

[  OK  ] stop : cron
[  OK  ] stop : nfs-kernel-server
[  OK  ] stop : lighttpd
[  OK  ] stop : php7.0-fpm
[... 190 lines cut ...]
-rwxr-xr-x 1 root root  158 Feb  2 09:47 build-req-pass
-rwxr-xr-x 1 root root  449 Feb  2 09:47 clean-all
-rwxr-xr-x 1 root root 1.5K Feb  2 09:47 inherit-inter
drwx------ 2 root root 4.0K Feb  2 09:48 keys
-rwxr-xr-x 1 root root  302 Feb  2 09:47 list-crl
-rwxr-xr-x 1 root root 7.7K Feb  2 09:47 openssl-0.9.6.cnf
-rwxr-xr-x 1 root root 8.3K Feb  2 09:47 openssl-0.9.8.cnf
-rwxr-xr-x 1 root root 8.2K Feb  2 09:47 openssl-1.0.0.cnf
-rwxr-xr-x 1 root root  13K Feb  2 09:47 pkitool
-rwxr-xr-x 1 root root 1.1K Feb  2 09:47 revoke-full
-rwxr-xr-x 1 root root  178 Feb  2 09:47 sign-req
-rwxr-xr-x 1 root root 2.3K Feb  2 09:47 vars
-rwxr-xr-x 1 root root  740 Feb  2 09:47 whichopensslcnf

```

```
root@RPi-Zero:~# ls -lah /etc/openvpn/easy-rsa/keys/
total 16K
drwx------ 2 root root 4.0K Feb  2 09:48 .
drwxr-xr-x 3 root root 4.0K Feb  2 09:48 ..
-rw-r--r-- 1 root root  269 Feb  2 09:48 DietPi_OpenVPN_Client.ovpn
-rw-r--r-- 1 root root    0 Feb  2 09:48 index.txt
-rw-r--r-- 1 root root    3 Feb  2 09:48 serial
```

Fourdee, 2018-02-02:
@k-plan 

Thanks confirmed:
- 🈴 Stretch
- 🈯️ Jessie | fine

Fourdee, 2018-02-02:
@MichaIng 
For ref: I'am replacing OMV ```126``` with LibSSL1.0.0

Fourdee, 2018-02-02:
Hmm: https://github.com/OpenVPN/easy-rsa/issues/138

Fourdee, 2018-02-02:
Completed, resolved for v6.2.

k-plan, 2018-02-02:
@Fourdee 

> Software that requires libssl1.0.0 (under stretch):
> 
> VPN 97
> 

sorry for bother but  what is with `id 117 | =2 | pivpn` ?

Fourdee, 2018-02-02:
@k-plan 

https://github.com/Fourdee/DietPi/issues/1450#issuecomment-362626872
^^

From what I can work out, its a bug in OpenVPN and/or Debian Stretch, expecting a file that does not exist. ```/etc/openvpn/easy-rsa/pki/*``` does not exist on default installation of OpenVPN via APT.

PiVPN claim to have resolved the issue: https://github.com/pivpn/pivpn/issues/337

k-plan, 2018-02-02:
@Fourdee 

^^ yes, but have you check to install` pivpn `and see what's happens?
For me here the same as with OpenVPN  ....  ^^

```
 DietPi-Services
─────────────────────────────────────────────────────
 Mode: status

[  OK  ] cron   active (running) since Fri 2018-02-02 19:05:12 CET; 13min ago
[  OK  ] openvpn        active (exited) since Fri 2018-02-02 19:05:13 CET; 13min ago
```

```
root@RPI-Zero:~# dietpi-software list | vpn
-bash: vpn: command not found
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (0)
root@RPI-Zero:~# dietpi-software list | grep vpn
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (0)
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (0)
id 97 | =0 | openvpn: vpn server | +rsyslog | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=613#p613
id 117 | =2 | pivpn: openvpn installer & management tool | +rsyslog | http://dietpi.com/phpbb/viewtopic.php?http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=3469#p346

```

Fourdee, 2018-02-02:
@k-plan 

> /bin/bash: warning: setlocale: LC_ALL: cannot change locale (0)

https://github.com/Fourdee/DietPi/issues/1430

Tested PiVPN, aside from https://github.com/Fourdee/DietPi/issues/1450#issuecomment-362666554 https://github.com/Fourdee/DietPi/issues/1450#issuecomment-362626872 (which I believe is out of our control), installs and configures fine.

k-plan, 2018-03-03:
@Fourdee 

> Tested PiVPN, aside from # 1450 (comment) # 1450 (comment) (which I believe is out of our control), installs and configures fine.

Hmm ... have you see this? 

`https://github.com/pivpn/pivpn/issues/475`

Fourdee, 2018-03-03:
@k-plan 

> Hmm ... have you see this?

Looks suspect (new account to post that one message). I'll run a test anyway to verify.

Fourdee, 2018-06-28:
@sorriso93 

Thanks for the report 👍 
- Which device tested?

Notes:
```
 libssl1.0.0 pre-depends on multiarch-support
  multiarch-support is not installed.
```

Fourdee, 2018-06-28:
Hmm, missing files:
https://packages.debian.org/stretch/all/easy-rsa/filelist
```
root@DietPi:/etc/openvpn/easy-rsa# ls -lha /usr/share/easy-rsa/
total 68K
drwxr-xr-x  3 root root 4.0K Jun 28 13:22 .
drwxr-xr-x 57 root root 4.0K Jun 28 13:22 ..
-rwxr-xr-x  1 root root  36K Jan 21 15:37 easyrsa
-rw-r--r--  1 root root 4.5K Jan 21 15:37 openssl-easyrsa.cnf
-rw-r--r--  1 root root 8.3K Jan 21 15:37 vars.example
drwxr-xr-x  2 root root 4.0K Jun 28 13:22 x509-types
```

Fourdee, 2018-06-28:
@sorriso93 

- DietPi no longer controls the openvpn service, once installed please reboot the system.
- The following is fine, as the service is launching the openvpn server under another process:
```
root@DietPi:~# systemctl status openvpn -l
● openvpn.service - OpenVPN service
   Loaded: loaded (/lib/systemd/system/openvpn.service; enabled; vendor preset:
enabled)
   Active: active (exited) since Thu 2018-06-28 13:47:51 BST; 4s ago
  Process: 4020 ExecStart=/bin/true (code=exited, status=0/SUCCESS)
 Main PID: 4020 (code=exited, status=0/SUCCESS)

Jun 28 13:47:51 DietPi systemd[1]: Starting OpenVPN service...
Jun 28 13:47:51 DietPi systemd[1]: Started OpenVPN service.
```
- Please check openvpn is running in htop ```htop```
[image]

sorriso93, 2018-06-28:
On raspberry 1 I had pivpn/openvpn running in the last year... starting from last month I have been experiencing the problem, uninstalled and reinstalled, no way same problem... Then I installed it on a raspberry 3 with dietpi last version, same behavior.
The strange thing is that I had mldonkey running on the raspb1 and now it has the same behavior (systemctl status and it is "exited")... Can be a common package update... I don't have enough linux competencies to understand the reason...

htop on rspb 3
[image]

htop on raspb1
[image]

sorriso93, 2018-06-28:
sudo netstat -tpln for both

Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp        0      0 0.0.0.0:10000           0.0.0.0:*               LISTEN      16945/perl          
tcp        0      0 192.168.0.3:53          0.0.0.0:*               LISTEN      18142/named         
tcp        0      0 10.8.0.1:53             0.0.0.0:*               LISTEN      18142/named         
tcp        0      0 127.0.0.1:53            0.0.0.0:*               LISTEN      18142/named         
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      364/dropbear        
tcp6       0      0 :::38913                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::44355                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::39683                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::44867                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::40675                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::36003                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::35875                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::38277                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::40709                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::33127                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::38599                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::37159                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::41831                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::61991                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::35753                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::35533                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::41965                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::40815                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::45295                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::8080                 :::*                    LISTEN      18800/homebridge-co 
tcp6       0      0 :::44113                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::39473                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::51825                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::32819                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::38707                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::45365                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::40565                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::36213                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::46741                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::53                   :::*                    LISTEN      18142/named         
tcp6       0      0 :::21                   :::*                    LISTEN      16909/proftpd: (acc 
tcp6       0      0 :::22                   :::*                    LISTEN      364/dropbear        
tcp6       0      0 :::37943                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::38263                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::33111                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::37591                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::40025                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::45561                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::39707                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::39005                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::39039                :::*                    LISTEN      18775/homebridge    
tcp6       0      0 :::36479                :::*                    LISTEN      18775/homebridge   

raspb1

Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp        0      0 0.0.0.0:1883            0.0.0.0:*               LISTEN      1227/mosquitto      
tcp        0      0 0.0.0.0:10000           0.0.0.0:*               LISTEN      1377/perl           
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      344/dropbear        
tcp6       0      0 :::1883                 :::*                    LISTEN      1227/mosquitto      
[... cut ...]
tcp6       0      0 :::37063                :::*                    LISTEN      383/homebridge      
tcp6       0      0 :::33519                :::*                    LISTEN      383/homebridge      
tcp6       0      0 :::8080                 :::*                    LISTEN      2233/homebridge-con 
tcp6       0      0 :::51825                :::*                    LISTEN      383/homebridge      
tcp6       0      0 :::21                   :::*                    LISTEN      1215/proftpd: (acce 
tcp6       0      0 :::22                   :::*                    LISTEN      344/dropbear

Fourdee, 2018-06-28:
@sorriso93 

Appears the process is running fine on RPi 3, I assume this is the device with the issue?

Try connecting an openVPN client to the RPi 3 as per the online doc: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=613#p613

You should see this from the following command ```cat /run/openvpn/server.status``` on RPi 3
```
root@DietPi:~# cat /run/openvpn/server.status
OpenVPN CLIENT LIST
Updated,Thu Jun 28 20:33:38 2018
Common Name,Real Address,Bytes Received,Bytes Sent,Connected Since
DietPi_OpenVPN_Client,192.168.0.23:59539,7669,3383,Thu Jun 28 20:33:29 2018
ROUTING TABLE
Virtual Address,Common Name,Real Address,Last Ref
10.8.0.6,DietPi_OpenVPN_Client,192.168.0.23:59539,Thu Jun 28 20:33:29 2018
GLOBAL STATS
Max bcast/mcast queue length,0
END
```

Fourdee, 2018-06-29:
Marking as closed, unable to replicate in our testing: https://github.com/Fourdee/DietPi/issues/1450#issuecomment-401148565

Please reopen if required.

closed 2018-06-29 by Fourdee

sorriso93, 2018-07-01:
root@DP-HOMEBRIDGE:~# cat /run/openvpn/server.status
cat: /run/openvpn/server.status: No such file or directory
root@DP-HOMEBRIDGE:~# systemctl status openvpn
● openvpn.service - OpenVPN service
   Loaded: loaded (/lib/systemd/system/openvpn.service; enabled; vendor preset: enabled)
   Active: active (exited) since Sat 2018-06-30 01:17:03 CEST; 1 day 22h ago
  Process: 495 ExecStart=/bin/true (code=exited, status=0/SUCCESS)
 Main PID: 495 (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/openvpn.service

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.
root@DP-HOMEBRIDGE:~# cat /run/openvpn/server.status
cat: /run/openvpn/server.status: No such file or directory
root@DP-HOMEBRIDGE:~#