MichaIng/DietPi #1453: G_DIETPI-NOTIFY visual cleanup

Fourdee, 2018-02-02:
Attempting to:
- Reduce the use of banner style print for sub functions, to reduce confusion of the originating program (eg: dietpi-software install = banner, dietpi-services stop = banner).
- Print the originating program name, during all G_DIETPI-NOTIFY to avoid confusion.

Open to discussion/suggestions.

So far:
[image]

Fourdee, 2018-02-02:
I would set low priority, but this has been bugging me for a long time 🤣

MichaIng, 2018-02-02:
@Fourdee 

Hmm it would be somehow cool, if we could set a marker, that something is called from within another script/function.

Inside scripts:
`(( $HIERARCHY )) && export HIERARCHY=$(( HIERARCHY++ )) || export HIERARCHY=1`

Inside G_DIETPI-NOTIFY 3
`(( $HIERARCHY > 1 )) && print minor banner || print major banner`

We just have to take care, that it is not added to dietpi-globals, to not set this variable for terminal.

MichaIng, 2018-02-03:
On current implementation dietpi-services are a sub function in every case:
```
root@VM-Stretch:~# dietpi-services restart
[  OK  ] Root access verified.
[ SUBF ] DietPi-Services > restart
[  OK  ] DietPi-Services | restart : cron
[  OK  ] DietPi-Services | restart : mpd
[ SUBF ] DietPi-Process_tool > Apply
[ INFO ] DietPi-Process_tool | Cron (2590)
[  OK  ] DietPi-Process_tool | Nice      0
[  OK  ] DietPi-Process_tool | Affinity  0
[  OK  ] DietPi-Process_tool | Scheduler SCHED_OTHER 0
[ INFO ] DietPi-Process_tool | DHCP Client (585)
[  OK  ] DietPi-Process_tool | Nice      0
[  OK  ] DietPi-Process_tool | Affinity  0
[  OK  ] DietPi-Process_tool | Scheduler SCHED_OTHER 0
[ INFO ] DietPi-Process_tool | Dropbear (417)
[  OK  ] DietPi-Process_tool | Nice      0
[  OK  ] DietPi-Process_tool | Affinity  0
[  OK  ] DietPi-Process_tool | Scheduler SCHED_OTHER 0
[ INFO ] DietPi-Process_tool | Dropbear (2100)
[  OK  ] DietPi-Process_tool | Nice      0
[  OK  ] DietPi-Process_tool | Affinity  0
[  OK  ] DietPi-Process_tool | Scheduler SCHED_OTHER 0
[  OK  ] DietPi-Process Tool | Completed
```
But otherwise looks good 😃.

Fourdee, 2018-02-03:
@MichaIng

> On current implementation dietpi-services are a sub function in every case:

Hehe, its getting there 😃 👍 

Yep, still figuring how we can detect another script launch from initial script, whilst keeping track of it in globals, then switch to SUBF. Also monitor when it exits (eg: PID) and a few other things.

I'am basically playing around 🤣, will get there in the end 😃

MichaIng, 2018-02-04:
@Fourdee 
I added
```
(( $HIERARCHY )) && export HIERARCHY=$((HIERARCHY+1)) || export HIERARCHY=1
echo "$HIERARCHY"
```
to the globals section of our scripts. So, the for testing reasons printed, numbers in between show when subscripts are called and that the up/down counting works quite well. 
All scripts call `G_DIETPI-NOTIFY 3`, but the code checks for `$HIERARCHY`:
```
                elif (( $1 == 3 )); then
                        echo "$HIERARCHY"
                        if (( $HIERARCHY )) && (( $HIERARCHY > 1 )); then
                                Print_Subfunction
                                echo -e "$2 > $(Print_Input_String 2)"
                        else
                                echo -e "\n \e[38;5;154m$2\e[0m"
                                echo -e "$header_line"
                                echo -e " \e[90mMode: \e[0m$(Print_Input_String 2)\n"
                        fi
```
Works quite well. The exported value is successfully reprinted (two identical values in front of `[ SUBF ]` and used by G_DIETPI-NOTIFY, but not changed within (grant) parental function, nor available within terminal.
```
root@VM-Stretch:~# dietpi-software install 85
[  OK  ] Root access verified.
[ INFO ] DietPi-Drive_Manager | Checking RootFS R/W access. Please wait...
[  OK  ] DietPi-Drive_Manager | RootFS R/W access.

1
[ INFO ] DietPi-Software | Initializing database, please wait...
[ INFO ] DietPi-Software | Reading database, please wait...
1

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Automated install

[  OK  ] DietPi-Software | Installing Nginx: webserver
[ INFO ] DietPi-Drive_Manager | Detecting drives, please wait...
[ INFO ] DietPi-Drive_Manager | Detected UUID mount: /dev/sda1 > /
[ INFO ] DietPi-Drive_Manager | Processing drive information, please wait...
[ INFO ] DietPi-Drive_Manager | Checking available free space on RootFS, please wait...
[  OK  ] DietPi-Drive_Manager | 10041 MB available, 500 MB required
[  OK  ] DietPi-Software | Connection test: https://deb.debian.org/debian/
[  OK  ] NTPD: time sync | Completed
2
2
[ SUBF ] DietPi-Services > unmask
[  OK  ] DietPi-Services | unmask all: cron
[  OK  ] DietPi-Services | unmask all: mpd
[  OK  ] DietPi-Services | start : cron
[  OK  ] DietPi-Services | start : mpd
3
3
[ SUBF ] DietPi-Process_tool > Apply
[ INFO ] DietPi-Process_tool | Cron (5050)
[  OK  ] DietPi-Process_tool | Nice      0
[  OK  ] DietPi-Process_tool | Affinity  0
[  OK  ] DietPi-Process_tool | Scheduler SCHED_OTHER 0
[ INFO ] DietPi-Process_tool | DHCP Client (655)
[  OK  ] DietPi-Process_tool | Nice      0
[  OK  ] DietPi-Process_tool | Affinity  0
[  OK  ] DietPi-Process_tool | Scheduler SCHED_OTHER 0
[ INFO ] DietPi-Process_tool | Dropbear (417)
[  OK  ] DietPi-Process_tool | Nice      0
[  OK  ] DietPi-Process_tool | Affinity  0
[  OK  ] DietPi-Process_tool | Scheduler SCHED_OTHER 0
[ INFO ] DietPi-Process_tool | Dropbear (2088)
[  OK  ] DietPi-Process_tool | Nice      0
[  OK  ] DietPi-Process_tool | Affinity  0
[  OK  ] DietPi-Process_tool | Scheduler SCHED_OTHER 0
[... 135 lines cut ...]
[  OK  ] DietPi-Process_tool | Nice      0
[  OK  ] DietPi-Process_tool | Affinity  0
[  OK  ] DietPi-Process_tool | Scheduler SCHED_OTHER 0
[ INFO ] DietPi-Process_tool | DHCP Client (655)
[  OK  ] DietPi-Process_tool | Nice      0
[  OK  ] DietPi-Process_tool | Affinity  0
[  OK  ] DietPi-Process_tool | Scheduler SCHED_OTHER 0
[ INFO ] DietPi-Process_tool | Dropbear (417)
[  OK  ] DietPi-Process_tool | Nice      0
[  OK  ] DietPi-Process_tool | Affinity  0
[  OK  ] DietPi-Process_tool | Scheduler SCHED_OTHER 0
[ INFO ] DietPi-Process_tool | Dropbear (2088)
[  OK  ] DietPi-Process_tool | Nice      0
[  OK  ] DietPi-Process_tool | Affinity  0
[  OK  ] DietPi-Process_tool | Scheduler SCHED_OTHER 0
[ INFO ] DietPi-Process_tool | Webserver Nginx (8880)
[  OK  ] DietPi-Process_tool | Nice      0
[  OK  ] DietPi-Process_tool | Affinity  0
[  OK  ] DietPi-Process_tool | Scheduler SCHED_OTHER 0
[ INFO ] DietPi-Process_tool | Webserver Nginx (8882)
[  OK  ] DietPi-Process_tool | Nice      0
[  OK  ] DietPi-Process_tool | Affinity  0
[  OK  ] DietPi-Process_tool | Scheduler SCHED_OTHER 0
[  OK  ] DietPi-Process Tool | Completed
```

Fourdee, 2018-02-04:
@MichaIng 

You beat me to it, great work 👍 
Send PR of what you have? I'll run some tests

MichaIng, 2018-02-04:
@Fourdee 
https://github.com/Fourdee/DietPi/pull/1470

Fourdee, 2018-02-04:
I'll add following to all scripts, just incase we add any ```G_DIETPI-NOTIFY 3``` additions:
```
	(( $HIERARCHY )) && export HIERARCHY=$((HIERARCHY+1)) || export HIERARCHY=1
```

Fourdee, 2018-02-04:
@MichaIng 

Works a charm, added some extra stuff, will send commit in a few:
[image]

MichaIng, 2018-02-04:
@Fourdee 
Actually it should only be necessary, if:
1. `G_DIETPI-NOTIFY 3` is called and
2. The script itself is called within others

So `dietpi-letsencrypt` e.g. doesn't need it right now. But doesn't hurt also, so jep agree 😃.

€: Hehe, nice, already thought into similar direction with SUB1/2 😄.

MichaIng, 2018-02-09:
Maybe we can add a check to process tool, that it shall only echo detailed information, if it's hierarchy is < 3? This would prevent long outputs within e.g. dietpi-software.
```
( $HIERARCHY > 2 ) && G_DIETPI-NOTIFY -2 'Running DietPi-Process_Tool' || G_DIETPI-NOTIFY 3 ...
...
( $HIERARCHY < 3 ) && echo process details
...
G_DIETPI-NOTIFY 0 ...
else
G_DIETPI-NOTIFY 1 ...
```
Something like this.
Would cover the other issue and partly PR.

Fourdee, 2018-02-09:
@MichaIng 

> Maybe we can add a check to process tool, that it shall only echo detailed information, if it's hierarchy is < 3? This would prevent long outputs within e.g. dietpi-software.

Yep agree 👍 

I love the additional information, but your right, not needed during install etc.

Fourdee, 2018-02-18:
@MichaIng 

I'am pretty satisfied with the current visual system. Great joint effort on this 👍 

Marking as closed RE: https://github.com/Fourdee/DietPi/pull/1424#issuecomment-366525898
Please reopen if required.

closed 2018-02-18 by Fourdee