MichaIng/DietPi #1462: DietPi-Services | Specific and reverted start/stop order

MichaIng, 2018-02-03:
Some services depend on/access others. As DietPi-Services starts and stops them in the same order, there are short time spans, where errors can occur. I sometimes see access errors on database access tries or cron jobs accessing to web server e.g., during the many start/stop/restarts, done during testing installations.

We could reorder some obvious services and go through the array backwards (if possible), if services are stopped:
- start: database(s) > PHP > web server > cron
- stop: cron > web server > PHP > database(s)

On restart of course we can't fully prevent those errors, but the affected time spans are again shorter.

If this is guaranteed, I will also add ownCloud/Nextcloud maintenance mode as oneshot systemd services, once implemented and remove their current exceptional position from dietpi-services code 😄.

MichaIng, 2018-02-14:
Same could be done for installation/uninstallation:
- On installation the order works quite well, but on uninstallation it counts the indexes upwards and breaks e.g. database/user deletions, if MariaDB is removed as well (okay this is no problem), but also database backups, if wanted (ownCloud/Nextcloud), before dumping, to prevent accidents, will break.
- But to be true, this is no "real" issue, as rarely someone removed several software at once, especially not some database dependent software together with a database, without doing some manual backup first or being 100% sure, that it is not needed 😉. It is more a consistent design thing, if installing/uninstalling service start/stop is done in exact reverse order for dependency reasons.

MichaIng, 2018-02-25:
Other specific issue of uninstall order: https://github.com/Fourdee/DietPi/issues/1523

I have two ideas:
1. Read/check temp uninstall file directly within loop:
   - Give all uninstall indexes as argument (with spaces) to Uninstall_Software, instead of one by one: https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-software#L15373
   - `if [[ "$@" =~ ' 23 ' ]];` instead of `if (( $index == 23 ))`
   - Then sorting uninstall script.
2. Do not G_AGP within loop, do just `apt-mark auto` and let G_AGA to actual purging in it's preferred order. Would need testing, especially `dpkg -P` would need to be handled separately, but at least those manual installed package titles are mostly (needs review) no dependencies of others.

Fourdee, 2018-02-27:
@MichaIng 

Sent commit to reverse stop order:
```
 Mode: start

[  OK  ] DietPi-Services | start : cron
[  OK  ] DietPi-Services | start : lighttpd
[  OK  ] DietPi-Services | start : php7.0-fpm
[  OK  ] DietPi-Services | start : mysql
[  OK  ] DietPi-Services | start : redis-server
[  OK  ] DietPi-Services | start : mpd
[  OK  ] DietPi-Services | start : avahi-daemon

---

Mode: stop

[  OK  ] DietPi-Services | ncc maintenance:mode --on
[  OK  ] DietPi-Services | stop : avahi-daemon
[  OK  ] DietPi-Services | stop : mpd
[  OK  ] DietPi-Services | stop : redis-server
[  OK  ] DietPi-Services | stop : mysql
[  OK  ] DietPi-Services | stop : php7.0-fpm
[  OK  ] DietPi-Services | stop : lighttpd
[  OK  ] DietPi-Services | stop : cron
```

Fourdee, 2018-03-01:
ToDo:
- ```dietpi-software``` Uninstall ordering.

MichaIng, 2018-03-02:
@Fourdee 
I was playing around a bid:
```
        Uninstall_Banner(){

                echo ''
                G_DIETPI-NOTIFY 0 "Uninstalling ${aSOFTWARE_WHIP_NAME[$uninstall_index]}: ${aSOFTWARE_WHIP_DESC[$uninstall_index]}\n"

        }

        Uninstall_Software(){

                #NB: systemctl daemon-reload is executed after this func in Uninstall_Software_Finalize()

                local uninstall_list=" $@ "

                #----------------------------------------------------------------------
                #Inform User
                G_DIETPI-NOTIFY 3 "$G_PROGRAM_NAME" "Uninstall"

                #echo -e ""
                #G_DIETPI-NOTIFY 0 "Uninstalling ${aSOFTWARE_WHIP_NAME[$index]}: ${aSOFTWARE_WHIP_DESC[$index]}\n"

                #Was a uninstall event trigged?
                local valid_input=1

                #----------------------------------------------------------------------
                #DIETPI SOFTWARE

                local uninstall_index=23
                if [[ "$uninstall_list" =~ " $uninstall_index " ]]; then

                        Uninstall_Banner
                        G_AGP lxde lxde-* upower policykit-1 iceweasel

                fi

                uninstall_index=24
                if [[ "$uninstall_list" =~ " $uninstall_index " ]]; then

                        Uninstall_Banner
                        G_AGP mate-desktop-environment-extras upower policykit-1 iceweasel
```
```
...
        Input_Modes(){
...
                                if [ "$1" = "uninstall" ]; then

                                        Uninstall_Software "${ainput[@]}"
                                        Uninstall_Software_Finalize
```
```
        Uninstall_NonSelected_Choices(){

                #Uninstall software using our temp uninstall list
                if [ -f "$UNINSTALL_FILE" ]; then

                        local uninstall_list
                        #Run the temp uninstall script
                        while read -r line
                        do

                                uninstall_list+=" $line "

                        done < $UNINSTALL_FILE
                        Uninstall_Software "$uninstall_list"
                        rm $UNINSTALL_FILE

                fi

        }
```
Generally it works, but printing uninstall info to log and adjusting `/DietPi/dietpi/.installed` needs to be inside every if statement as well. A lot of additional lines, much work anyway. I don't like it too much.

Second solution with just apt-mark auto and autoremove everything afterwards is also not a really nice solution, as it is a bid in-transparent, what is done due to which uninstall. I would like to have most things done within the single uninstall statements.

Maybe we could handle it similar than for installation? Add a new uninstall flag `-1` to `/DietPi/dietpi/.installed` and check for this within Uninstall_Software()? It maybe needs as much work as the other solutions above, but at least we can just copy the methods from installation and be consistent here. This would enable us as well to use the same functions for both, reading string and adjusting `/DietPi/dietpi/.installed` just based of where we come from, install- or uninstall menu respectively "$1".
- I will further play with this last idea. The more I think about it, the more I like it 😉.

Fourdee, 2018-03-25:
- 🈯️ Service ordering done: https://github.com/Fourdee/DietPi/issues/1462#issuecomment-368958753
- 🈯️ Uninstall ordering done: https://github.com/Fourdee/DietPi/pull/1586

@MichaIng Great work 👍, can we mark this as closed?

MichaIng, 2018-03-25:
@Fourdee 
I am just at the same topic right now, actually ordering services atm. found some other issue while doing this 😄. I will just post final results here fore completeness and then mark as closed.

Fourdee, 2018-03-31:
@MichaIng 

> and in case database active to serve it's content, it is best to stop it first and start is last

Done: https://github.com/Fourdee/DietPi/commit/e7c1f6164fabe746c75ced70504a8359740eba47
This brings the webserver ordering to same as previous (eg: v6.3). I'll mark this as closed, however, please reopen if required or needs adjusting.

closed 2018-03-31 by Fourdee

MichaIng, 2018-03-31:
@Fourdee 
Whoops, did I write it that way? Can't find my comment anymore I meant database should be started first, webserver last. The pre 6.3 state caused the errors, as webserver (respectively PHP connector) can't access database then.

I already tested it on VM and RPi, works fine and resolves error messages.

Fourdee, 2018-03-31:
@MichaIng 

Legend thank you 👍 

> Can't find my comment anymore 

https://github.com/Fourdee/DietPi/pull/1650#issuecomment-377008918, although, I probably misread it entirely 😃