MichaIng/DietPi #1463: DietPi-Software | IQaudio Cosmic Controller

Fourdee, 2018-02-03:
> https://iqaudio.co.uk/blog/new-product-the-cosmic-controller-n22

Fourdee, 2018-02-03:
Quick test with: https://github.com/iqaudio/tools/blob/master/cosmiccontroller.py
- OLED not functional, not implemented yet?
- Rotary controller hard coded for ```Digital``` alsamixer control (eg: wont work for RPi onboard audio) ````encoder count change to 12; amixer: Unable to find simple control 'Digital',0```

Fourdee, 2018-02-03:
```
#IQaudio Cosmic Controller -----------------------------------------------
#Requires
#Python + GIPO
G_AGI python python3
dietpi-software install 69
#ALSA
dietpi-software install 5
#LIRC (will need further user config? no documentation on IQaudio)
G_AGI lirc

wget https://github.com/iqaudio/tools/raw/master/cosmiccontroller.py -O /var/lib/dietpi/dietpi-software/installed/cosmiccontroller.py
chmod +x /var/lib/dietpi/dietpi-software/installed/cosmiccontroller.py

cat << _EOF_ > /etc/systemd/system/dietpi-cosmiccontroller.service
[Unit]
Description=DietPi IQaudio Cosmic Controller
After=sound.target

[Service]
Type=simple

ExecStart=$(which python) /var/lib/dietpi/dietpi-software/installed/cosmiccontroller.py

[Install]
WantedBy=default.target
_EOF_
systemctl enable dietpi-cosmiccontroller.service
systemctl daemon-reload
systemctl start dietpi-cosmiccontroller.service

#IQaudio Cosmic Controller -----------------------------------------------
```

Fourdee, 2018-02-07:
OLED:
https://www.olimex.com/Products/Modules/LCD/MOD-OLED-128x64/open-source-hardware

MichaIng, 2018-12-25:
Added to FeatHub, feel free to vote for it: https://feathub.com/Fourdee/DietPi/+4

MichaIng, 2021-05-27:
This product is not produced anymore. IQaudio has been bought by the RPi foundation last year and now (only) four new DACs/HATs are produced, excluding the cosmic controller. I'm hence marking this as closed.

closed 2021-05-27 by MichaIng