MichaIng/DietPi #1466: ProFTPd -basic error

lilghost05, 2018-02-04:
#### Required Information:
- DietPi Version | v.
- SBC Device     | RPi 2 Model B (armv7l) (index=2)
- Distro         | stretch (index=4)
- Command        | G_AGI: proftpd-basic
- Error Handler  | G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title | DietPi-Software

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Additional logs:
```
Log file contents:

```

--------------------------------------------------------------------

Fourdee, 2018-02-04:
@lilghost05 

Hi,

If this is a direct copy/paste of the error, may indicate failed update from v159 > v6.0 (which is now patched).

Please can you paste results:
```
cat /etc/.dietpi_image_version
```

Regardless of the above, it may be advisable to start fresh, download and write the latest v6.0 image:
http://dietpi.com/download

Fourdee, 2018-02-04:
Pending testing:
- RPi Stretch

closed 2018-02-22 by Fourdee