MichaIng/DietPi #1467: Minecraft Pocket Edition

symbios24, 2018-02-04:
Hello about the nuclear-powered server for Minecraft Pocket Edition for raspberry pi i have installed it with the latest DietPi(Stretch) but i cannot find any info about how to use it or if there is a webgui etc.
is the server start automatically after installed on reboot?

Thanks

Fourdee, 2018-02-04:
@symbios24 

Thanks for the report. Confirmed, this is missing from our online docs, we'll need to get it added:
http://dietpi.com/phpbb/viewtopic.php?f=8&t=5

Fourdee, 2018-02-04:
From what I can work out (ref: https://thepi.io/how-to-set-up-a-minecraft-pocket-edition-server-on-the-raspberry-pi/)
- Nukkit has no web interface
- The server will be displayed under LAN games in Minecraft
https://thepi.io/wp-content/uploads/2017/08/minecraft-pocket-edition-nukkit-for-minecraft-pe.png

The program is launched automatically by DietPi during boot.

Fourdee, 2018-02-04:
@symbios24 

If you want a web interface MineCraft server, try following:
http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=60#p2068
http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=60#p2069

symbios24, 2018-02-04:
thanks for add this, dont want a web interface i only asked if there is one since there was not any info,and i did not now anything about how the server works.