MichaIng/DietPi #1469: Extra Software

symbios24, 2018-02-04:
Hello is it possible to add this Extra Software to DietPi Please

assaultcube server
openarena game
the battle for wesnoth 
minetest server
manaplus client
terraria game server
freeciv
openttd

thanks

MichaIng, 2018-02-04:
@symbios24 
Thanks for your request. Could you edit your post to add some more info to the software titles, some link to website and at best some official and/or self-proven install steps?

That would make it easier for us to compare effort/benefit and against solutions we already offer 😉.

symbios24, 2018-02-05:
Hello,thanks for the response
openarena install
-----------------
install latest firmware on stretch: rpi-update (otherwise the game wont run)
apt-get install openarena
wget http://www.berryterminal.com/dl/ioquake3_99.1.36-rpi01_armhf.deb
dpkg -i ./ioquake3_99.1.36-rpi01_armhf.deb
Run it with this command: sudo /usr/games/openarena
--------------------------------------------------------
Assaultcube server
------------------
wget https://sourceforge.net/projects/actiongame/files/AssaultCube%20Version%201.2.0.2/AssaultCube_v1.2.0.2.tar.bz2

wget https://sourceforge.net/projects/actiongame/files/AssaultCube%20Version%201.2.0.2/AssaultCube_v1.2.0.2.source.tar.bz2

tar -xvf AssaultCube_v1.2.0.2.tar.bz2
mv AssaultCube_v1.2.0.2 ac
tar -xvf AssaultCube_v1.2.0.2.source.tar.bz2
mv AssaultCube_v1.2.0.2.source source

cd /home/pi/assault/source/source/enet

sudo sh ./configure -build=x86-linux -host=arm-unknown-linux-gnueabi
sudo make clean && sudo make && sudo make install

cd /home/pi/assault/source/
mkdir bin_unix
cd bin_unix
mkdir native_server
cd /home/pi/assault/source/source/src
sudo apt-get install clang-3.5
sudo sed -i "s/clang++/g++/g" Makefile
sudo make server_install
sudo ln -s /usr/local/lib/libenet.so.2.1.0 /usr/lib/libenet.so.2
cd /home/pi/assault/source/bin_unix/native_server/
cp -R ac_server /home/pi/assault/ac/bin_unix/native_server
cd /assault/ac/ (home/pi/assault/ac)
to run the server: sh server.sh
-----------------------------------------------------------
Minetest server follow the instructions on this page its in german translate it to english
https://wiki.minetest.net/Minetest_in_der_Schule/RaspberryPi-Server_anlegen_und_einrichten
---------------------------------------------------------------------------------------------
openttd install: apt-get install openttd
run it with: openttd
--------------------------------------
Freeciv install: apt-get install freeciv-client-sdl freeciv-sound-standard
dkpg-reconfigure xserver-xorg-legacy (if its not installed install it with apt-get install xserver-xorg-legacy)
select anybody and press ok 
run the game with :  freeciv-client-sdl or freeciv
------------------------------------------------
manaplus install: apt-get install manaplus
run it: manaplus --safemode
------------------------------------------
i will give you more games and how to install them soon any questions let me know

Thanks

symbios24, 2018-02-05:
More games

apt-get install boswars
run it with boaswars

apt-get install widelands
run it with widelands

apt-get install simutrans
ruin it with simutrans

Terraria server in this page
https://gist.github.com/BalzGuenat/65c248d03895b5a030ba

symbios24, 2018-02-05:
Hello,thanks for the response
openarena install

install latest firmware on stretch: rpi-update (otherwise the game wont run)
apt-get install openarena
wget http://www.berryterminal.com/dl/ioquake3_99.1.36-rpi01_armhf.deb
dpkg -i ./ioquake3_99.1.36-rpi01_armhf.deb
Run it with this command: sudo /usr/games/openarena

Assaultcube server

wget https://sourceforge.net/projects/actiongame/files/AssaultCube%20Version%201.2.0.2/AssaultCube_v1.2.0.2.tar.bz2

wget https://sourceforge.net/projects/actiongame/files/AssaultCube%20Version%201.2.0.2/AssaultCube_v1.2.0.2.source.tar.bz2

tar -xvf AssaultCube_v1.2.0.2.tar.bz2
mv AssaultCube_v1.2.0.2 ac
tar -xvf AssaultCube_v1.2.0.2.source.tar.bz2
mv AssaultCube_v1.2.0.2.source source

cd /home/pi/assault/source/source/enet

sudo sh ./configure -build=x86-linux -host=arm-unknown-linux-gnueabi
sudo make clean && sudo make && sudo make install

cd /home/pi/assault/source/
mkdir bin_unix
cd bin_unix
mkdir native_server
cd /home/pi/assault/source/source/src
sudo apt-get install clang-3.5
sudo sed -i "s/clang++/g++/g" Makefile
sudo make server_install
sudo ln -s /usr/local/lib/libenet.so.2.1.0 /usr/lib/libenet.so.2
cd /home/pi/assault/source/bin_unix/native_server/
cp -R ac_server /home/pi/assault/ac/bin_unix/native_server
cd /assault/ac/ (home/pi/assault/ac)
to run the server: sh server.sh

Minetest server follow the instructions on this page its in german translate it to english
https://wiki.minetest.net/Minetest_in_der_Schule/RaspberryPi-Server_anlegen_und_einrichten

openttd install: apt-get install openttd
run it with: openttd

Freeciv install: apt-get install freeciv-client-sdl freeciv-sound-standard
dkpg-reconfigure xserver-xorg-legacy (if its not installed install it with apt-get install xserver-xorg-legacy)
select anybody and press ok
run the game with : freeciv-client-sdl or freeciv

manaplus install: apt-get install manaplus
run it: manaplus --safemode

apt-get install boswars
run it with boaswars

apt-get install widelands
run it with widelands

apt-get install simutrans
ruin it with simutrans

Terraria server in this page
https://gist.github.com/BalzGuenat/65c248d03895b5a030ba

symbios24, 2018-02-05:
Hello,thanks for the response

openarena (first person shooter game) install

install latest firmware on stretch: rpi-update (otherwise the game wont run)
apt-get install openarena
wget http://www.berryterminal.com/dl/ioquake3_99.1.36-rpi01_armhf.deb
dpkg -i ./ioquake3_99.1.36-rpi01_armhf.deb
Run it with this command: sudo /usr/games/openarena

Assaultcube server

wget https://sourceforge.net/projects/actiongame/files/AssaultCube%20Version%201.2.0.2/AssaultCube_v1.2.0.2.tar.bz2

wget https://sourceforge.net/projects/actiongame/files/AssaultCube%20Version%201.2.0.2/AssaultCube_v1.2.0.2.source.tar.bz2

tar -xvf AssaultCube_v1.2.0.2.tar.bz2
mv AssaultCube_v1.2.0.2 ac
tar -xvf AssaultCube_v1.2.0.2.source.tar.bz2
mv AssaultCube_v1.2.0.2.source source

cd /home/pi/assault/source/source/enet

sudo sh ./configure -build=x86-linux -host=arm-unknown-linux-gnueabi
sudo make clean && sudo make && sudo make install

cd /home/pi/assault/source/
mkdir bin_unix
cd bin_unix
mkdir native_server
cd /home/pi/assault/source/source/src
sudo apt-get install clang-3.5
sudo sed -i "s/clang++/g++/g" Makefile
sudo make server_install
sudo ln -s /usr/local/lib/libenet.so.2.1.0 /usr/lib/libenet.so.2
cd /home/pi/assault/source/bin_unix/native_server/
cp -R ac_server /home/pi/assault/ac/bin_unix/native_server
cd /assault/ac/ (home/pi/assault/ac)
to run the server: sh server.sh

Minetest server(simular to minecraft server) follow the instructions on this page its in german translate it to english
https://wiki.minetest.net/Minetest_in_der_Schule/RaspberryPi-Server_anlegen_und_einrichten

openttd (city simulator) install: apt-get install openttd
run it with: openttd

Freeciv (strategy game) install: apt-get install freeciv-client-sdl freeciv-sound-standard
dkpg-reconfigure xserver-xorg-legacy (if its not installed install it with apt-get install xserver-xorg-legacy)
select anybody and press ok
run the game with : freeciv-client-sdl or freeciv

manaplus(MMORPG client) install: apt-get install manaplus
run it: manaplus --safemode

boswars(real time strategy game)
apt-get install boswars
run it with boaswars

widelands(real time strategy game))
apt-get install widelands
run it with widelands

simutrans(transportation simulator game)
apt-get install simutrans
ruin it with simutrans

Terraria server in this page
https://gist.github.com/BalzGuenat/65c248d03895b5a030ba

symbios24, 2018-02-05:
Hello,thanks for the response

openarena (first person shooter game) http://www.openarena.ws/page.php?2

install latest firmware on stretch: rpi-update (otherwise the game wont run)
apt-get install openarena
wget http://www.berryterminal.com/dl/ioquake3_99.1.36-rpi01_armhf.deb
dpkg -i ./ioquake3_99.1.36-rpi01_armhf.deb
Run it with this command: sudo /usr/games/openarena

Assaultcube server https://assault.cubers.net/docs/server.html

wget https://sourceforge.net/projects/actiongame/files/AssaultCube%20Version%201.2.0.2/AssaultCube_v1.2.0.2.tar.bz2

wget https://sourceforge.net/projects/actiongame/files/AssaultCube%20Version%201.2.0.2/AssaultCube_v1.2.0.2.source.tar.bz2

tar -xvf AssaultCube_v1.2.0.2.tar.bz2
mv AssaultCube_v1.2.0.2 ac
tar -xvf AssaultCube_v1.2.0.2.source.tar.bz2
mv AssaultCube_v1.2.0.2.source source

cd /home/pi/assault/source/source/enet

sudo sh ./configure -build=x86-linux -host=arm-unknown-linux-gnueabi
sudo make clean && sudo make && sudo make install

cd /home/pi/assault/source/
mkdir bin_unix
cd bin_unix
mkdir native_server
cd /home/pi/assault/source/source/src
sudo apt-get install clang-3.5
sudo sed -i "s/clang++/g++/g" Makefile
sudo make server_install
sudo ln -s /usr/local/lib/libenet.so.2.1.0 /usr/lib/libenet.so.2
cd /home/pi/assault/source/bin_unix/native_server/
cp -R ac_server /home/pi/assault/ac/bin_unix/native_server
cd /assault/ac/ (home/pi/assault/ac)
to run the server: sh server.sh

Minetest server(simular to minecraft server)  https://www.minetest.net/

follow the instructions on this page to build the server its in german translate it to english
https://wiki.minetest.net/Minetest_in_der_Schule/RaspberryPi-Server_anlegen_und_einrichten

openttd (city simulator) https://www.openttd.org/

install: apt-get install openttd
run it with: openttd

Freeciv (strategy game) www.freeciv.org/

install: apt-get install freeciv-client-sdl freeciv-sound-standard
dkpg-reconfigure xserver-xorg-legacy (if its not installed install it with apt-get install xserver-xorg-legacy)
select anybody and press ok
run the game with : freeciv-client-sdl or freeciv

manaplus(MMORPG client) https://github.com/ManaPlus/ManaPlus

install: apt-get install manaplus 
run it: manaplus --safemode

boswars(real time strategy game) https://boswars.org/

apt-get install boswars
run it with boaswars

widelands(real time strategy game) https://wl.widelands.org/

apt-get install widelands
run it with widelands

simutrans(transportation simulator game) https://www.simutrans.com/

apt-get install simutrans
ruin it with simutrans

Terraria server (Crafting game server) info in this page https://terraria.org

Instructions to build terraria server on this page under
https://gist.github.com/BalzGuenat/65c248d03895b5a030ba

Fourdee, 2018-02-22:
@symbios24 

Thanks for the requests. As with all software requests, we mark the ticket as closed initially. Should interest peak on the ticket from other users, we'll reopen for investigation.

closed 2018-02-22 by Fourdee