MichaIng/DietPi #1471: Installing WiFi Hotspot to BananaPi Pro via dietpi-config, SSL handshake fails in spider test

twisty-doerofthings, 2018-02-04:
## Creating a bug report/issue: Installing WiFi Hotspot to BananaPi Pro via dietpi-config

Both Ethernet and Wifi are enabled with Static IP config (eth0: 192.168.0.89, wlan0: 192.168.0.88).

#### Required Information:
- DietPi Version | v6.1
- SBC Device     | BananaPi Pro (armv7l) (index=51)
- Distro         | stretch (index=4)
- Command        | Connection test: https://deb.debian.org/debian/
- Error Handler  | G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title | DietPi-Software

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Additional logs:
```
Log file contents:
Spider mode enabled. Check if remote file exists.
--2018-02-04 20:54:15--  https://deb.debian.org/debian/
Resolving deb.debian.org (deb.debian.org)... 149.20.4.15, 128.31.0.62, 130.89.148.14, ...
Connecting to deb.debian.org (deb.debian.org)|149.20.4.15|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://cdn-aws.deb.debian.org/debian/ [following]
Spider mode enabled. Check if remote file exists.
--2018-02-04 20:54:17--  https://cdn-aws.deb.debian.org/debian/
Resolving cdn-aws.deb.debian.org (cdn-aws.deb.debian.org)... 54.192.117.31, 54.192.117.106, 54.192.117.49, ...
Connecting to cdn-aws.deb.debian.org (cdn-aws.deb.debian.org)|54.192.117.31|:443... connected.
Unable to establish SSL connection.
```

twisty-doerofthings, 2018-02-04:
Attempted same steps locally, not via SSH, and installer ran without issue.

Fourdee, 2018-02-08:
@twisty-doerofthings 

Thanks for the report 👍 

> Unable to establish SSL connection.
Attempted same steps locally, not via SSH, and installer ran without issue.

Appears to be a temporary connection issue (local and/or remote). As this issue is now resolved, i'll mark as closed. Please reopen if required.

closed 2018-02-08 by Fourdee