MichaIng/DietPi #1473: DietPi

MHDMOUSTAFA, 2018-02-05:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?
- SBC Device (EG: RPi 3)?
- Power supply used (EG: 5v 1A RAVpower)?
- SD card used (EG: Sandisk ultra)?
- Distro (EG: Jessie) | ```uname -a```?

#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

Fourdee, 2018-02-05:
@MHDMOUSTAFA 

Hi,

We'll need more information. Please can you post the issue you are experiencing?

MichaIng, 2018-02-16:
Closing due to missing information and user reply.

closed 2018-02-16 by MichaIng