MichaIng/DietPi #1474: Kodi issue with RPi3

pedrom34, 2018-02-05:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | 6.1
- SBC Device RPi 3
- Power supply used : 5v 2.5A
- SD card used : Sandisk ultra 32Gb
- Distro Linux DietPi 4.9.79-v7+ #1086 SMP Wed Jan 31 18:03:34 GMT 2018 armv7l GNU/Linux

#### Additional Information (if applicable):
- Kodi is said to be installed, but doesn't launch
- I tried to uninstall

#### Expected behaviour:
Kodi is supposed to uninstall

#### Actual behaviour:
Error:  
[image]

Is this "Unable to locate package kodi-odroid" normal on a raspberry pi 3?

MichaIng, 2018-02-05:
@pedrom34 
Thanks for your report.

On uninstall we purge all modules, installed on any device, thus including `kodi-odroid` which is installed on Odroids only. On current master we don't remove the non-installed packages from the list, before adding it to `apt-get purge ...`, thus error is thrown. This is handled correctly on next v6.2: https://github.com/Fourdee/DietPi/blob/testing/dietpi/func/dietpi-globals#L976

However this should not effect Kodi launch. Did you follow one of the possible startup options: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5#p43 ?

pedrom34, 2018-02-05:
Thanks.

Well, yes, Kodi is selected in autostart, and it doesn't start.
is there a way to uninstall kodi so I can try again?

pedrom34, 2018-02-06:
You can close the issue, I reflashed an sd card and started from scratch.  
Did the trick, everything is now ok.

Fourdee, 2018-02-06:
Will mark as closed, ```G_AGP``` issue resolved in v6.2:
https://github.com/Fourdee/DietPi/issues/1474#issuecomment-363143941

closed 2018-02-06 by Fourdee