MichaIng/DietPi #1483: PiVPN not working

gpioneer90, 2018-02-07:
Hi, I see there's already a thread created but I wasn't sure if replying to it will reopen/alert the devs so I made a new one.

On 6.1, after installing PiVPN, I attempt to use pivpn -a, but it's not able to create the .ovpn file:
Generating a 2048 bit RSA private key
........................................................+++
..............................................................................................+++
writing new private key to '/etc/openvpn/easy-rsa/pki/private/gpioneer.key.9Hz7w9SXLb'
Enter PEM pass phrase:
Verifying - Enter PEM pass phrase:
-----
Using configuration from /etc/openvpn/easy-rsa/openssl-1.0.cnf
Check that the request matches the signature
Signature ok
The Subject's Distinguished Name is as follows
commonName            :ASN.1 12:'gpioneer'
Certificate is to be certified until Feb  5 02:17:01 2028 GMT (3650 days)

Write out database with 1 new entries
Data Base Updated
spawn openssl rsa -in pki/private/gpioneer.key -des3 -out pki/private/gpioneer.key
Enter pass phrase for pki/private/gpioneer.key:
writing RSA key
Enter PEM pass phrase:
Verifying - Enter PEM pass phrase:
Client's cert found: gpioneer.crt
Client's Private Key found: gpioneer.key
CA public Key found: ca.crt
tls-auth Private Key found: ta.key
cat: Default.txt: No such file or directory
cp: cannot create regular file '/home//ovpns/gpioneer.ovpn': No such file or directory
chown: cannot access '/home//ovpns/gpioneer.ovpn': No such file or directory

========================================================
Done! gpioneer.ovpn successfully created! 
gpioneer.ovpn was copied to:
  /home//ovpns
for easy transfer. Please use this profile only on one
device and create additional profiles for other devices.
========================================================

Fourdee, 2018-02-07:
@gpioneer90 

Hi,

Thanks for the report 👍 

Notes:

> pivpn -a
believe having a double slash in the directory name path

Correct, may be a issue with PiVPN itself, as ```pivpn -a``` is outside of DietPi programs.

I only tested the ```dietpi-software``` installation, i'll retest with ```pivpn -a```, see if we can replicate.

Fourdee, 2018-02-07:
@gpioneer90 

RPi 3 Stretch v6.2:
Which device are you running this on?

🈯️ test with ```pivpn``` as the selected user during PiVPN install menu:
```
root@DietPi:~# pivpn -a
Enter a Name for the Client:  test
Enter the password for the client:
Enter the password again to verify:
spawn ./easyrsa build-client-full test

Note: using Easy-RSA configuration from: ./vars
rand: Use -help for summary.
Generating a 1024 bit RSA private key
.........++++++
..........++++++
writing new private key to '/etc/openvpn/easy-rsa/pki/private/test.key.ywVkRWkFNb'
Enter PEM pass phrase:test

Verifying - Enter PEM pass phrase:test

-----
Using configuration from /etc/openvpn/easy-rsa/openssl-1.0.cnf
Check that the request matches the signature
Signature ok
The Subject's Distinguished Name is as follows
commonName            :ASN.1 12:'test'
Certificate is to be certified until Feb  5 16:21:39 2028 GMT (3650 days)

Write out database with 1 new entries
Data Base Updated
spawn openssl rsa -in pki/private/test.key -des3 -out pki/private/test.key
Enter pass phrase for pki/private/test.key:
writing RSA key
Enter PEM pass phrase:
Verifying - Enter PEM pass phrase:
Client's cert found: test.crt
Client's Private Key found: test.key
CA public Key found: ca.crt
tls-auth Private Key found: ta.key

========================================================
Done! test.ovpn successfully created!
test.ovpn was copied to:
  /home/pivpn/ovpns
for easy transfer. Please use this profile only on one
device and create additional profiles for other devices.
========================================================
```

🈯️ test with ```dietpi``` as the selected user during PiVPN install menu:
```
root@DietPi:~# pivpn -a
Enter a Name for the Client:  test
Enter the password for the client:
Enter the password again to verify:
spawn ./easyrsa build-client-full test

Note: using Easy-RSA configuration from: ./vars
rand: Use -help for summary.
Generating a 1024 bit RSA private key
......++++++
............++++++
writing new private key to '/etc/openvpn/easy-rsa/pki/private/test.key.EfDWrzhqYM'
Enter PEM pass phrase:
Verifying - Enter PEM pass phrase:
-----
Using configuration from /etc/openvpn/easy-rsa/openssl-1.0.cnf
Check that the request matches the signature
Signature ok
The Subject's Distinguished Name is as follows
commonName            :ASN.1 12:'test'
Certificate is to be certified until Feb  5 16:15:58 2028 GMT (3650 days)

Write out database with 1 new entries
Data Base Updated
spawn openssl rsa -in pki/private/test.key -des3 -out pki/private/test.key
Enter pass phrase for pki/private/test.key:
writing RSA key
Enter PEM pass phrase:
Verifying - Enter PEM pass phrase:
Client's cert found: test.crt
Client's Private Key found: test.key
CA public Key found: ca.crt
tls-auth Private Key found: ta.key

========================================================
Done! test.ovpn successfully created!
test.ovpn was copied to:
  /home/dietpi/ovpns
for easy transfer. Please use this profile only on one
device and create additional profiles for other devices.
========================================================
```

Fourdee, 2018-02-07:
@gpioneer90 

Can you confirm the user/password for account created with ```pivpn -a```?

gpioneer90, 2018-02-07:
@Fourdee 
I used gpioneer:test123

Running 6.1 on Raspberry Pi 3

Fourdee, 2018-02-07:
@gpioneer90 

Thanks.

Any changes made to system locales? (eg: ```dietpi-config``` > Language)

gpioneer90, 2018-02-07:
Yes, changed the keyboard layout and added en_us while keeping the default en_gb and changed time zone to central

gpioneer90, 2018-02-07:
I changed my keyboard back to generic 104 key keyboard to test and still the same result. Should I try removing en_us?

Fourdee, 2018-02-09:
@gpioneer90 

> added en_us while keeping the default en_gb | Should I try removing en_us?

I cant see it having any effect. System (and therefore PiVPN installer) should be using the default locale in all cases.

gpioneer90, 2018-02-09:
Hmm, well I'm pretty clueless when it comes to Linux so I don't know how to explain why it works for you but not for me. I guess I'll try OpenVPN on its own at some point and let you know if that works or if it produces a similar problem

Fourdee, 2018-02-09:
@gpioneer90 

Just quickly, please can you paste results:
```
locale
```

gpioneer90, 2018-02-09:
root@DietPi:~# locale
LANG=en_GB.UTF-8
LANGUAGE=en_US:en
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_PAPER="en_US.UTF-8"
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT="en_US.UTF-8"
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=en_US.UTF-8
root@DietPi:~#

gpioneer90, 2018-02-09:
No idea why it's struck through

Fourdee, 2018-02-10:
@gpioneer90 

Thanks, i'll try to replicate with en_US

```
LC_ALL=en_US.UTF-8
```

gpioneer90, 2018-02-11:
`drwxr-xr-x  4 root   root   4.0K Feb  6 18:43 .
drwxr-xr-x 21 root   root   4.0K Jan 10 06:15 ..
drwxr-xr-x  2 dietpi dietpi 4.0K Jan 29 12:54 dietpi
drwxr-xr-x  2 root   root   4.0K Feb  6 18:43 pivpn`

Fourdee, 2018-02-22:
Unable to replicate this our end. Marking as closed. Please reopen if required.

closed 2018-02-22 by Fourdee