MichaIng/DietPi #1484: Installing any software reverts GPU mem split to 128 + if pihole, no internet connectiong

gpioneer90, 2018-02-07:
DietPi 6.1:
Whenever I use dietpi-software to install anything, at the end of the script there's a message saying the gpumemsplit is being set to 128, every time. So if you manually change the split to 192 or whatever else, everytime you install software it will revert back.

Also, I don't know if it's a bug or a feature but it's very disruptive: if pihole is used and you direct clients to the pihole IP as their DNS, clients will be without internet connection while software is being installed because DietPi's (new?) scripts stop services while programs are being installed.

Could go unnoticed if it's something small, but for example I was installing PiVPN earlier and during the 2048 bit cert generation which took about a 60-90 minutes, clients weren't able to use the internet until the install process was done (or I used pihole disable temporarily) since DietPI shutdown dnsmasq prior to installation starting.

I see how it could possibly solve problems, but it's very disruptive and I was not expecting that, a warning would be nice. Maybe you can have DietPi detect if PiHole is installed and if so display a warning prior to every software install reminding people that DNS resolution will be unavailable and that will resume after install or they can use pihole disable in the mean time

Fourdee, 2018-02-07:
@gpioneer90 

Thanks for the report 👍 

We'll need to confirm installed software, to debug, please paste results:
```
dietpi-software list | grep ' =2 '
```

> Also, I don't know if it's a bug or a feature but it's very disruptive: if pihole is used and you direct clients to the pihole IP as their DNS, clients will be without internet connection while [...]

This is by design, to prevent running software from interfering with install/uninstall/configuration functions of ```dietpi-software```.

> I see how it could possibly solve problems, but it's very disruptive and I was not expecting that, a warning would be nice.

Good idea 👍 We'll add this info before uninstall/installs being.

gpioneer90, 2018-02-07:
id 0 | =2 | openssh client:  |
id 5 | =2 | alsa: linux sound system |
id 6 | =2 | xserver: linux display system |
id 17 | =2 | git client: git clone etc |
id 31 | =2 | kodi: the media centre for linux | +alsa +xserver | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5#p43
id 51 | =2 | opentyrian: a classic retro game, addictive | +alsa +xserver | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5#p45
id 84 | =2 | lighttpd: webserver | | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=1335#p1335
id 89 | =2 | php: webserver | | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=1335#p1335
id 93 | =2 | pi-hole: block adverts for any device on your network | +git +webserver | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=20#p174
id 102 | =2 | rsyslog: system logging | | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=20#p68
id 103 | =2 | dietpi-ramlog: minimal, optimized logging | | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=20#p68
id 105 | =2 | openssh server:  | | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=10#p63
id 106 | =2 | raspcontrol: web interface system stats | | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=20#p89
id 117 | =2 | pivpn: openvpn installer & management tool | +rsyslog | http://dietpi.com/phpbb/viewtopic.php?http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=3469#p3469
id 152 | =2 | avahi-daemon: hostname broadcast (mac, pc bonjour) |

Fourdee, 2018-02-07:
@gpioneer90 

> opentyrian

❤️ 👍 

---

Issue resolved for v6.2, marking as completed.

closed 2018-02-07 by Fourdee