MichaIng/DietPi #1485: Raspcontrol not installing properly

gpioneer90, 2018-02-07:
Upon installing Raspcontrol, cannot use. Following the URL provided in http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=20#p89 but of course with my own DietPi IP goes nowhere.

I have PiHole so it just displays the pihole logo

Fourdee, 2018-02-07:
@gpioneer90 

Thanks for the report 👍, we'll try to replicate.

Fourdee, 2018-02-07:
Notes:
```403 - Forbidden```
```
root@DietPi:~# ls -lha /var/www/raspcontrol/
total 72K
drwxrwxr-x 9 www-data www-data 4.0K Feb  7 16:28 .
drwxrwxr-x 5 www-data www-data 4.0K Feb  7 16:28 ..
-rwxrwxr-x 1 www-data www-data 5.8K Jun 23  2017 api.php
-rwxrwxr-x 1 www-data www-data  840 Jun 23  2017 config.php
drwxrwxr-x 2 www-data www-data 4.0K Jun 23  2017 css
drwxrwxr-x 2 www-data www-data 4.0K Jun 23  2017 img
-rwxrwxr-x 1 www-data www-data  13K Jun 23  2017 index.php
drwxrwxr-x 2 www-data www-data 4.0K Jun 23  2017 js
drwxrwxr-x 3 www-data www-data 4.0K Jun 23  2017 lib
-rwxrwxr-x 1 www-data www-data 1.5K Jun 23  2017 login.php
drwxrwxr-x 2 www-data www-data 4.0K Jun 23  2017 nbproject
drwxrwxr-x 2 www-data www-data 4.0K Jun 23  2017 Other Changes
drwxrwxr-x 2 www-data www-data 4.0K Jun 23  2017 pages
-rwxrwxr-x 1 www-data www-data  736 Jun 23  2017 README.md
```

Fourdee, 2018-02-07:
It could be an issue with php configuration, post v6.0, however, we'd need to verify with v159< install (which I lack).

As the Raspcontrol dev has not been worked on in over 2 years, i'll mark this as low priority for now. An alternative recommendation for stats is NetData:
http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=60#p1611

Fourdee, 2018-02-07:
@k-plan 
Any chance you have a <v159 RPi DietPi image laying around? 😃 (https://github.com/Fourdee/DietPi/issues/1485#issuecomment-363839915)

gpioneer90, 2018-02-07:
https://github.com/Fourdee/DietPi/issues/1485#issuecomment-363839915
yes no big deal for me, just thought you should know since it's included as option. RPi-Monitor works, btw

k-plan, 2018-02-08:
@Fourdee 

> @k-plan
> Any chance you have a <v159 RPi DietPi image laying around? 😃

For RPi ?

Uploaded:

- DietPi_v150_RPi-armv6-(Jessie).7z

- DietPi_v148_RPi-armv6-(Jessie).7z

- DietPi_v145_RPi-armv6-(Jessie).7z

Fourdee, 2018-02-08:
Same with v150:
```HTTP ERROR 500```

Will consider removal of this title, due to no dev in over 2 years.

Fourdee, 2018-02-10:
Removed as per: https://github.com/Fourdee/DietPi/issues/1491#issuecomment-364174270

closed 2018-02-10 by Fourdee