MichaIng/DietPi #1487: Added Codecs from Dietpi-config and from nano /Dietpi/config.txt and have taken it out and put it in the file but still says disabled.

skane165, 2018-02-07:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | 6.1
- SBC Device RPi 3
- Power supply used 5v 1A RAVpower
- SD card used Sandisk ultra 4K
- Distro Linux DietPi 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux

#### Expected behaviour:

When i put the codecs to the gui or the SD card manually i still gets the disable when i run 

#Note To Self, NEVER enable L2 cache, breaks most X based applications that wer$

decode_WVC1=0xf29505d7

decode_MPG2=0xbe4ac4fd

root@DietPi:~# vcgencmd codec_enabled MPG2
MPG2=disabled

#### Steps to reproduce:
The keys arrived and i put them in the manual way first no luck tried the dietpi-config/display way same result please just ask for more details if needed.
Have a nice day

Fourdee, 2018-02-07:
@skane165 

Thanks for the report 👍 

Did you restart the system after adding the keys? 

> The keys arrived and i put them in the manual way first no luck

This could indicate incorrect key entered for the device, or, a possible issue with kernel/firmware.

If key is correct for the specific device, try updating RPi kernel/firmware and see if the issue persists:
```
G_AGI rpi-update
rpi-update
reboot
```

skane165, 2018-02-07:
Hello i did a reboot after adding the keys=). 
After rpi-update i get stuck at ] Started D-Bus System Message Bus.
I will do a fresh install and see what going to happen be back with info.

skane165, 2018-02-08:
Made a fresh install updated and put the keys in through the gui still 
root@DietPi:~# vcgencmd codec_enabled MPG2
MPG2=disabled

skane165, 2018-02-08:
gpu_mem_512=16
gpu_mem_1024=16

#-------Max USB Current -------
max_usb_current=1

#-------Disable Ram temp refresh every 500ms---------------
disable_pvt=1

#-------Disable Pi Splash screen on boot-------------
disable_splash=1

#-------Remove dedicated pll for PWM audio -------
avoid_pwm_pll=1

#-------SoundCard-------
#dtoverlay=none
dtparam=audio=on

#-------i2c-------------
dtparam=i2c_arm=off
dtparam=i2c1=off
i2c_arm_baudrate=100000

#-------spi-------------
dtparam=spi=off

#-------Overclock-------
temp_limit=75
initial_turbo=20
force_turbo=0

over_voltage=4
arm_freq=1325
core_freq=400
#sdram_freq=450

#arm_freq_min=700
#core_freq_min=250
#sdram_freq_min=400

#Note To Self, NEVER enable L2 cache, breaks most X based applications that were not compiled with L2 cache enabled.

decode_WVC1=0xf29505d7

decode_MPG2=0xbe4ac4fd

Fourdee, 2018-02-08:
@skane165 

You need to increase GPU memory:
https://github.com/raspberrypi/documentation/issues/117#issue-41346600

```dietpi-config``` > display options > GPU/MEM split

Try 128MB

skane165, 2018-02-09:
Tried 120mb same result trying 196mb now

skane165, 2018-02-09:
Got enabled at 196mb

skane165, 2018-02-09:
Thank you for your help and have a nice weekend =)

Fourdee, 2018-02-09:
Reopened, we'll automatically increase GPU memory as needed.

Fourdee, 2018-04-05:
🈯️ Resolved for v6.7, 128MB is required, applied automatically by dietpi-config if lower than 128.
```
root@DietPi:~# vcgencmd codec_enabled MPG2
MPG2=enabled

root@DietPi:~# vcgencmd codec_enabled WVC1
WVC1=enabled

root@DietPi:~# cat /DietPi/config.txt | grep -m1 'gpu_mem'
gpu_mem_256=128

```

closed 2018-04-05 by Fourdee