MichaIng/DietPi #1488: DietPi-Software | PiJuice

Fourdee, 2018-02-07:
> https://github.com/PiSupply/PiJuice/tree/master/Software

Fourdee, 2018-02-07:
```
G_AGI pijuice-base
root@DietPi:~# /usr/bin/pijuice_sys.py
Traceback (most recent call last):
  File "/usr/bin/pijuice_sys.py", line 10, in <module>
    from pijuice import *
  File "/usr/lib/python2.7/dist-packages/pijuice.py", line 3, in <module>
    from smbus import SMBus
ImportError: No module named smbus

```

Fourdee, 2018-02-07:
Online doc: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=10740#p10740

MichaIng, 2018-02-08:
Nice one 👍, I am just thinking to take one as power loss protection. On the other hand, my external powered HDD docking station would be not included anyway, holding data, database and swapfile 🤔.

Fourdee, 2018-02-09:
Completed.

closed 2018-02-09 by Fourdee