MichaIng/DietPi #1489: Blue something

bobybobybob, 2018-02-07:
## Creating a bug report/issue:

--------------------------------------------------------------------
- DietPi has encounted an error, and, is unable to continue        -
- Please create a ticket: https://github.com/Fourdee/DietPi/issues -
- Copy and paste the BLUE lines below, into the ticket             -
--------------------------------------------------------------------

#### Required Information:
- DietPi Version | v6.-1
- SBC Device     | RPi B (armv6l) (index=1)
- Distro         | stretch (index=4)
- Command        | Connection test: https://www.mirrorservice.org/sites/archive.raspbian.org/raspbian
- Error Handler  | G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title | DietPi-Software

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Additional logs:
```
Log file contents:
Spider-Modus eingeschaltet. Es wird geprüft, ob die Datei auf dem Server existiert.
--2018-02-07 19:08:48--  https://www.mirrorservice.org/sites/archive.raspbian.org/raspbian
Auflösen des Hostnamens »www.mirrorservice.org (www.mirrorservice.org)« … fehlgeschlagen: Die Wartezeit für die Verbindung ist abgelaufen.
wget: Host-Adresse »www.mirrorservice.org« kann nicht aufgelöst werden
```

--------------------------------------------------------------------

[FAILED] Unable to continue, the program will now terminate.
root@DietPi:~# nano

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?   6
- SBC Device (EG: RPi 3)?    RPi b
- Power supply used (EG: 5v 1A RAVpower)?  yes
- SD card used (EG: Sandisk ultra)?   4gb sd class10
- Distro (EG: Jessie) | ```uname -a```?  

#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?  yes

#### Expected behaviour:

apt-get update   does not work

#### Actual behaviour:

#### Steps to reproduce:

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

Fourdee, 2018-02-07:
@bobybobybob 

Thanks for the report 👍 

Appears to be a connection issue (rough translation 😃 ).

Please can you retry running ```dietpi-software``` and installing the software you require again.
If problems persist:
- please run ```dietpi-config``` > ' 7  Network Options: Adapters' > ' Test         Run the Internet Connection Test'
- If the connection test is fine, please change the APT mirror ```dietpi-config``` > ' 8  Network Options: NAS/Misc' > 'Apt Mirror'

bobybobybob, 2018-02-07:
Hello Fourdee,
today i got a RSP V1  b  first time and on first start it go not well. 
it is very crazy and i try other distro but same. Then i go back to DietPi i connect LAN direct on Mainrouter and it work out of box and automatic upgrade from 6.0 to 6.1. Maybe LAN Cable broken?  Sorry for false alarm. The connection IP, Gateway, DNS, Subnet was right but ping in local network some OK some not. Now it work well and i likte dietPi. Much more fun than Armbian with Orange-Pi

closed 2018-02-07 by bobybobybob