MichaIng/DietPi #1491: DietPi-Software | Flagged for removal (cleanup)

Fourdee, 2018-02-08:
Goal is to remove the following items:
- Outdated/no longer developed software titles (eg: raspcontrol https://github.com/Fourdee/DietPi/issues/1485#issuecomment-364168459)
- Low install count from our users
- Allow us to focus support on more popular software titles

Fourdee, 2018-02-08:
- Grasshopper (100) | 1 install
- Raspcontrol: https://github.com/Fourdee/DietPi/issues/1485#issuecomment-364168459

Fourdee, 2018-02-08:
@k-plan @MichaIng 

Any suggestions we can add to the above list?

MichaIng, 2018-02-08:
@Fourdee 
- SubSonic 5, but add LibreSonic instead as at least until last summer developed: https://github.com/Fourdee/DietPi/issues/330#issuecomment-358902464
  - Ahh wait, better take AirSonic, the next fork 😆: https://github.com/airsonic/airsonic 
  - How are the installed numbers of SubSonic 5 and 6 actually? We have other web stream alternatives anyway.
- Grashopper might be installed by me due to testing. as I not always remember to opt out dietpi-survey 😜. Maybe we can auto opt out on dietpi-update, if any other than master branch is used 🤔.

If software indexes are feed, how did you handle them in the past? Switch others software indexes? As gaps will cause dietpi-software to fail (the reason I left MySQL+Stacks dummy declarations). Maybe we can simple include a check for existing `aSOFTWARE_WHIP_NAME[$i]`.

Fourdee, 2018-02-08:
@MichaIng 

> If software indexes are freed, how did you handle them in the past? Switch others software indexes?

For quickness:
- Set ```aSOFTWARE_TYPE[$index_current]=-1```, hides from the GUI.
- Make a note of it being disabled, so it can be reused at a later date.
- Remove the install/config/uninstall code

If time permits:
- manually swap the highest software index in the install/config/uninstall code.

Fourdee, 2018-02-08:
@MichaIng 
I'll send a commit for example

MichaIng, 2018-02-08:
@Fourdee 
Okay, maybe I find the time to fill the MySQL indexes gaps 😉.
But one has to patch the existing `.installed` state on update as well.

Fourdee, 2018-02-08:
@MichaIng 

> But one has to patch the existing .installed state on update as well.

Yep 👍 

For this example, we can simply reset ```[100]=0```, as PiJuice is not available v6.1.

Fourdee, 2018-02-10:
Marking as closed. Raspcontrol/Grasshopper only for now.

closed 2018-02-10 by Fourdee