MichaIng/DietPi #1493: Issue installing NZBGet

TheEmpty, 2018-02-09:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?
6
1
- SBC Device (EG: RPi 3)?
RPi 2b
- Power supply used (EG: 5v 1A RAVpower)?
Something 5v
- SD card used (EG: Sandisk ultra)?
Some sandisk card -- user options are on an external HDD, Segate ~40gb free.
- Distro (EG: Jessie) | ```uname -a```?
Linux DietPi 4.9.79+ #1091 Tue Feb 6 12:58:13 GMT 2018 armv6l GNU/Linux

#### Additional Information (if applicable):
- Software title?
NZBGet
- Can this issue be replicated on a fresh installation of DietPi?
Was a nearly fresh install.

#### Expected behaviour:
NZBGet should be installed fully, with optimized configuration via dietpi-software

#### Actual behaviour:
NZBGet is only partially installed.

#### Steps to reproduce:
Attempting to install NZBGet produced this for me. Uninstalling and reinstalling, the same.

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:
```
 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installing NZBget: nzb download manager

[  OK  ] Connection test: https://nzbget.net
--2018-02-09 09:42:33--  https://nzbget.net/download/nzbget-latest-bin-linux.run
Resolving nzbget.net (nzbget.net)... 35.199.180.1
Connecting to nzbget.net (nzbget.net)|35.199.180.1|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github.com/nzbget/nzbget/releases/download/v19.1/nzbget-19.1-bin-linux.run [following]
--2018-02-09 09:42:33--  https://github.com/nzbget/nzbget/releases/download/v19.1/nzbget-19.1-bin-linux.run
Resolving github.com (github.com)... 192.30.255.113, 192.30.255.112
Connecting to github.com (github.com)|192.30.255.113|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github-production-release-asset-2e65be.s3.amazonaws.com/37613809/8800ab44-6412-11e7-8df9-fd8f39c116dd?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20180209%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20180209T094234Z&X-Amz-Expires=300&X-Amz-Signature=d968e69d5d4830471505f41bfc5f2e69d8dcbbe10ef6c789204da31b256e65de&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Dnzbget-19.1-bin-linux.run&response-content-type=application%2Foctet-stream [following]
--2018-02-09 09:42:34--  https://github-production-release-asset-2e65be.s3.amazonaws.com/37613809/8800ab44-6412-11e7-8df9-fd8f39c116dd?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20180209%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20180209T094234Z&X-Amz-Expires=300&X-Amz-Signature=d968e69d5d4830471505f41bfc5f2e69d8dcbbe10ef6c789204da31b256e65de&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Dnzbget-19.1-bin-linux.run&response-content-type=application%2Foctet-stream
Resolving github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)... 52.216.128.107
Connecting to github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)|52.216.128.107|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 25535939 (24M) [application/octet-stream]
Saving to: 'package.run'

package.run                                                 100%[========================================================================================================================================>]  24.35M  1.26MB/s    in 22s

2018-02-09 09:42:56 (1.12 MB/s) - 'package.run' saved [25535939/25535939]

Installer for nzbget-19.1
Verifying package...
Checking system...
CPU-Architecture: armel
Unpacking...
tar: 7za-armel: Cannot change ownership to uid 1001, gid 1001: Operation not permitted
tar: cacert.pem: Cannot change ownership to uid 1001, gid 1001: Operation not permitted
tar: ChangeLog: Cannot change ownership to uid 1001, gid 1001: Operation not permitted
tar: COPYING: Cannot change ownership to uid 1001, gid 1001: Operation not permitted
tar: install-update.sh: Cannot change ownership to uid 1001, gid 1001: Operation not permitted
[... 45 lines cut ...]
ERROR: Unpacking failed.
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (0)

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Configuring NZBget: nzb download manager

sed: can't read /mnt/dietpi_userdata/nzbget/nzbget.conf: No such file or directory
sed: can't read /mnt/dietpi_userdata/nzbget/nzbget.conf: No such file or directory
sed: can't read /mnt/dietpi_userdata/nzbget/nzbget.conf: No such file or directory
sed: can't read /mnt/dietpi_userdata/nzbget/nzbget.conf: No such file or directory
sed: can't read /mnt/dietpi_userdata/nzbget/nzbget.conf: No such file or directory
sed: can't read /mnt/dietpi_userdata/nzbget/nzbget.conf: No such file or directory
sed: can't read /mnt/dietpi_userdata/nzbget/nzbget.conf: No such file or directory
sed: can't read /mnt/dietpi_userdata/nzbget/nzbget.conf: No such file or directory
sed: can't read /mnt/dietpi_userdata/nzbget/nzbget.conf: No such file or directory
sed: can't read /mnt/dietpi_userdata/nzbget/nzbget.conf: No such file or directory
sed: can't read /mnt/dietpi_userdata/nzbget/nzbget.conf: No such file or directory
sed: can't read /mnt/dietpi_userdata/nzbget/nzbget.conf: No such file or directory
sed: can't read /mnt/dietpi_userdata/nzbget/nzbget.conf: No such file or directory
sed: can't read /mnt/dietpi_userdata/nzbget/nzbget.conf: No such file or directory
sed: can't read /mnt/dietpi_userdata/nzbget/nzbget.conf: No such file or directory
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (0)
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (0)
```

TheEmpty, 2018-02-09:
Seems the issue is with running the nzbget installer as root. Tar is trying to preserve the user to discourage root? The NZBGet 'run' has an option though for setting the tar command.
`Info "    --tar <args>      - run custom tar command on archive"`

Could probably use `tar --owner root --group root --no-same-owner` for that. `which dietpi-software` didn't return anything, so not sure where to edit that in to test. Might dig through the source now :P

Fourdee, 2018-02-09:
@TheEmpty 

Thanks for the report 👍 

There may another issue here aswell, causing this (https://github.com/Fourdee/DietPi/issues/1430):
```
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (0)
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (0)
```

I'll run a local test install to verify.

Fourdee, 2018-02-09:
🈯️ RPi 3:
```
Installer for nzbget-19.1
Verifying package...
Checking system...
CPU-Architecture: armhf
Unpacking...
Configuring...
  Free memory detected: 874 MB
  Activating article cache (ArticleCache=100)
  Increasing write buffer (WriteBuffer=1024)
  Increasing par repair buffer (ParBuffer=100)
  CPU speed detected: 152 BogoMIPS
  Disabling download during par check/repair (ParPauseQueue=yes)
  Disabling download during unpack (UnpackPauseQueue=yes)
  Disabling download during post-processing (ScriptPauseQueue=yes)
Installation completed

Quick help (from nzbget-directory):
   ./nzbget -s        - start nzbget in console mode
   ./nzbget -D        - start nzbget in daemon mode (in background)
   ./nzbget -C        - connect to background process
   ./nzbget -Q        - stop background process
   ./nzbget -h        - help screen with all commands

Successfully installed into /mnt/dietpi_userdata/nzbget
Web-interface is on http://localhost:6789 (login:nzbget, password:tegbzn6789)
For support please visit http://nzbget.net/forum
```

Will test RPi 2 (```CPU-Architecture: armel``` incorrect)

TheEmpty, 2018-02-10:
```
root@DietPi:~# locale
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_MESSAGES to default locale: No such file or directory
locale: Cannot set LC_ALL to default locale: No such file or directory
LANG=en_GB.UTF-8
LANGUAGE=0:0
LC_CTYPE="0"
LC_NUMERIC="0"
LC_TIME="0"
LC_COLLATE="0"
LC_MONETARY="0"
LC_MESSAGES="0"
LC_PAPER="0"
LC_NAME="0"
LC_ADDRESS="0"
LC_TELEPHONE="0"
LC_MEASUREMENT="0"
LC_IDENTIFICATION="0"
LC_ALL=0
```

```
root@DietPi:~# grep -v '^#' /etc/locale.gen
0 UTF-8
en_GB.UTF-8 UTF-8
```

I've only installed items through the interface/CLI so far. If armel is incorrect, how can I change it?

Fourdee, 2018-02-10:
@TheEmpty 

I believe this issue is solely due to the following issue (confirmed on your install aswell): https://github.com/Fourdee/DietPi/issues/1430

I'll mark this as closed and investigate the other ticket, which will resolve this one.

I'd be grateful if you could confirm the following:
- Locale changes made to system, and how (eg: dietpi-config)
- Installed locale's and default locale selected (if you can remember, if not, no worries)
- During 1st run of DietPi (after image write), did you install any specific software from the list, if so which?

closed 2018-02-10 by Fourdee

TheEmpty, 2018-02-10:
* Never made any locale changes.
* Never touched locale, so whatever the defaults are for the Raspberry Pi image downloaded from the main site.
* I don't know if it was the first run, but it would be Radarr, Sonarr, NZBGet, PiHole, Somba (SMB), and Fail2ban.

I'll look at the mentioned issue and see if I can adjust my machine to prevent that error and see if that fixes the install.

TheEmpty, 2018-02-11:
Setting the locale with the config tool, fixed the install. Also changed the arch to 'linux' for the download. Weird way that manifested.