MichaIng/DietPi #1494: netdata - unable to update

gpioneer90, 2018-02-09:
I just installed Netdata on 6.1 Rpi3 and like it a lot. I see there's an update button on top of the GUI but it appears stuck on Downloading installed git commit id from netdata...

The installed version is 1.6.0 while the newest is 1.9.0. Should the update function be working?

gpioneer90, 2018-02-09:
I removed the DietPi install and used bash <(curl -Ss https://my-netdata.io/kickstart.sh) to get from source instead :)

Fourdee, 2018-02-09:
@gpioneer90 

> I removed the DietPi install and used bash <(curl -Ss https://my-netdata.io/kickstart.sh) to get from source instead :)

Yep, we offered to go with binary installs for speed and reduction in pre-reqs during install. The downside is we need to manually update the binaries when possible.

Our binary versions currently match the APT install in Stretch BP:
https://packages.debian.org/stretch-backports/netdata

> The installed version is 1.6.0 while the newest is 1.9.0. Should the update function be working?

I believe NetData update only works for x86_64, as they only offer those binaries:
https://github.com/firehol/binary-packages

Fourdee, 2018-02-09:
@gpioneer90 

I've updated NetData to v1.9.0, for DietPi v6.2.

Updates will be handled by DietPi patching system.

I'll mark this as closed. Please reopen if required.

closed 2018-02-09 by Fourdee