MichaIng/DietPi #1495: DietPi fails to boot on PINE64 A64+

kgunnar, 2018-02-09:
## Creating a bug report/issue:
Attempting to boot a DietPi image on a PINE64 A64+ Rev B board hangs part way through (serial console log is below)

#### Required Information:
- DietPi Version tried both DietPi_v6.0_PineA64-ARMv8-Stretch.img and DietPi_v145_PineA64-arm64-(Jessie).img
- SBC Device PINE64 A64+ Rev B
- Power supply used CanaKit 5V 2.5A and other 5V 2A
- SD card used Sandisk EVO+ 32GB and PNY Elite 32GB
- Distro Debian Stretch and Jessie

#### Additional Information (if applicable):
I've tried flashing the image with mintstick (USB Image Writer in Mint) and with dd. The microSD cards and power supplies used will boot other images created for Pine64 A64+ but they will not boot with either DietPi image. Both DietPi images hang at the exact same place. The log shows it identifies the card and size and although it does have errors trying to read uboot.env, bootlogo.bmp, and boot.scr, that just seems they are missing files as it does read uEnv.txt and the device tree blob file. I noticed mention of serial console being disabled by default and modified the dietpi.txt to have CONFIG_SERIAL_CONSOLE_ENABLE=1 but it does not continue any further.

#### Expected behaviour:
System to boot fully.

#### Actual behaviour:
System boot hangs just after "Starting kernel..." it logs "[mmc]: MMC Device 2 not found" and "[mmc]: mmc 2 not find, so not exit" then hangs.

#### Steps to reproduce:
Flash DietPi_v6.0_PineA64-ARMv8-Stretch.img or DietPi_v145_PineA64-arm64-(Jessie).img to a microSD card and attempt to boot on PINE64 A64+ board.

#### Did you submit a ```dietpi-bugreport```?
Issue happens too early.

#### Extra details:
Log from serial console: 
```
HELLO! BOOT0 is starting!
boot0 commit : 045061a8bb2580cb3fa02e301f52a015040c158f
 
boot0 version : 4.0.0
set pll start
set pll end
rtc[0] value = 0x00000000
rtc[1] value = 0x00000000
rtc[2] value = 0x00000000
rtc[3] value = 0x0000a102
rtc[4] value = 0x00000000
rtc[5] value = 0x00000000
DRAM driver version: V1.1
rsb_send_initseq: rsb clk 400Khz -> 3Mhz
PMU: AXP81X
ddr voltage = 1500 mv
DRAM Type = 3 (2:DDR2,3:DDR3,6:LPDDR2,7:LPDDR3)
DRAM clk = 672 MHz
DRAM zq value: 003b3bbb
DRAM single rank full DQ OK
DRAM size = 2048 MB
DRAM init ok
dram size =2048
card boot number = 0, boot0 copy = 0
card no is 0
sdcard 0 line count 4
[mmc]: mmc driver ver 2015-05-08 20:06
[mmc]: sdc0 spd mode error, 2
[mmc]: Wrong media type 0x00000000
[mmc]: ***Try SD card 0***
[mmc]: HSSDR52/SDR25 4 bit
[mmc]: 50000000 Hz
[mmc]: 30543 MB
[mmc]: ***SD/MMC 0 init OK!!!***
sdcard 0 init ok
The size of uboot is 000e8000.
sum=832e0b41
src_sum=832e0b41
Succeed in loading uboot from sdmmc flash.
boot0: start load other image
boot0: Loading BL3-1
Loading file 0 at address 0x40000000,size 0x00008400 success
boot0: Loading scp
Loading file 2 at address 0x00040000,size 0x00019c00 success
set arisc reset to de-assert state
Ready to disable icache.
Jump to secend Boot.

U-Boot 2014.07-6-pine64-longsleep (Jun 18 2016 - 16:15:13) Allwinner Technology 

uboot commit : 55c9c8c8ac005b1c00ac948386c60c4a741ebaa9
 
rsb: secure monitor exist
[      0.335]pmbus:   ready
[      0.337][ARISC] :arisc initialize
[      0.666][ARISC] :arisc_dvfs_cfg_vf_table: support only one vf_table
[      0.781][ARISC] :sunxi-arisc driver startup succeeded
[      0.814]PMU: AXP81X
[      0.816]PMU: AXP81X found
bat_vol=77, ratio=100
[      0.823]PMU: dcdc2 1100
[      0.826]PMU: cpux 1008 Mhz,AXI=336 Mhz
PLL6=600 Mhz,AHB1=200 Mhz, APB1=100Mhz AHB2=300Mhz MBus=400Mhz
device_type = 3253, onoff=1
dcdc1_vol = 3300, onoff=1
dcdc2_vol = 1100, onoff=1
dcdc6_vol = 1100, onoff=1
aldo1_vol = 2800, onoff=0
aldo2_vol = 1800, onoff=1
aldo3_vol = 3000, onoff=1
dldo1_vol = 3300, onoff=0
dldo2_vol = 3300, onoff=0
dldo3_vol = 2800, onoff=0
dldo4_vol = 3300, onoff=1
eldo1_vol = 1800, onoff=1
eldo2_vol = 1800, onoff=0
eldo3_vol = 1800, onoff=0
fldo1_vol = 1200, onoff=0
fldo2_vol = 1100, onoff=1
gpio0_vol = 3100, onoff=0
vbus not exist
no battery, limit to dc
run key detect
no key found
no uart input
DRAM:  2 GiB
fdt addr: 0xb6ebf0a0
Relocation Offset is: 75f11000
In:    serial
Out:   serial
Err:   serial
gic: sec monitor mode
[      1.636]start
drv_disp_init
init_clocks: finish init_clocks.
enable power vcc-hdmi-33, ret=0
drv_disp_init finish
boot_disp.output_disp=0
boot_disp.output_type=3
boot_disp.output_mode=10
fetch script data boot_disp.auto_hpd fail
disp0 device type(4) enable
attched ok, mgr0<-->device1, type=4, mode=10
[      2.007]end
workmode = 0,storage type = 1
[      2.011]MMC: 0
[mmc]: mmc driver ver 2015-06-03 13:50:00
SUNXI SD/MMC: 0
[mmc]: start mmc_calibrate_delay_unit, don't access device...
[mmc]: delay chain cal done, sample: 200(ps)
[mmc]: media type 0x0
[mmc]: Wrong media type 0x0
[mmc]: ************Try SD card 0************
[mmc]: host caps: 0x27
[mmc]: MID 9e PSN 003337e6
[mmc]: PNM SDU32 -- 0x53-44-55-33-32
[mmc]: PRV 0.2
[mmc]: MDT m-7 y-2016
[mmc]: speed mode     : HSSDR52/SDR25 
[mmc]: clock          : 50000000 Hz
[mmc]: bus_width      : 4 bit
[mmc]: user capacity  : 30543 MB
[mmc]: ************SD/MMC 0 init OK!!!************
[mmc]: erase_grp_size      : 0x1WrBlk*0x200=0x200 Byte
[mmc]: secure_feature      : 0x0
[mmc]: secure_removal_type : 0x0
[      2.198]sunxi flash init ok
[mmc]: Has init
[      2.231]---drivers/mmc/mmc.c 2733 mmc_init
reading uboot.env

** Unable to read "uboot.env" from mmc0:1 **
Using default environment

--------fastboot partitions--------
mbr not exist
base bootcmd=run mmcbootcmd
bootcmd set setargs_mmc
key 0
recovery key high 12, low 10
fastboot key high 6, low 4
no misc partition is found
to be run cmd=run mmcbootcmd
update dtb dram start
update dtb dram  end
serial is: 8400519008004c15034b
get Pine64 model from DRAM size
DRAM >512M
Pine64 model: pine64-plus
no battery exist
sunxi_bmp_logo_display
[mmc]: Has init
[      2.434]---drivers/mmc/mmc.c 2733 mmc_init
reading bootlogo.bmp
** Unable to read file bootlogo.bmp **
sunxi bmp info error : unable to open logo file bootlogo.bmp
[      2.451]inter uboot shell
Hit any key to stop autoboot:  3  2  1  0 
[mmc]: Has init
[      5.564]---drivers/mmc/mmc.c 2733 mmc_init
reading uEnv.txt
[mmc]: blkcnt should not be 0
169 bytes read in 6 ms (27.3 KiB/s)
Loading boot environment ...
[mmc]: Has init
[      5.585]---drivers/mmc/mmc.c 2733 mmc_init
reading boot.scr
** Unable to read file boot.scr **
Booting with defaults ...
[mmc]: Has init
[      5.604]---drivers/mmc/mmc.c 2733 mmc_init
reading pine64/sun50i-a64-pine64-plus.dtb
69414 bytes read in 11 ms (6 MiB/s)
[mmc]: Has init
[      5.661]---drivers/mmc/mmc.c 2733 mmc_init
reading pine64/Image
11796928 bytes read in 578 ms (19.5 MiB/s)
[mmc]: Has init
[      6.251]---drivers/mmc/mmc.c 2733 mmc_init
reading initrd.img
1076776 bytes read in 56 ms (18.3 MiB/s)
Image lacks image_size field, assuming 16MiB
## Flattened Device Tree blob at 45000000
   Booting using the fdt blob at 0x45000000
   reserving fdt memory region: addr=41010000 size=10000
   reserving fdt memory region: addr=41020000 size=800
   reserving fdt memory region: addr=40100000 size=4000
   reserving fdt memory region: addr=40104000 size=1000
   reserving fdt memory region: addr=40105000 size=1000
   reserving fdt memory region: addr=45000000 size=11000
   Loading Ramdisk to b6db0000, end b6eb6e28 ... OK
   Loading Device Tree to 44feb000, end 44ffefff ... OK

Starting kernel ...

[mmc]: MMC Device 2 not found
[mmc]: mmc 2 not find, so not exit
```

Fourdee, 2018-02-09:
@kgunnar 

Thanks for the report 👍 

> Flash DietPi_v6.0_PineA64-ARMv8-Stretch.img or DietPi_v145_PineA64-arm64-(Jessie).img to a microSD card and attempt to boot on PINE64 A64+ board.

If both images are causing the same issue, I can only suspect HW failure on the SBC itself (assuming PSU and SD cards are not a factor), or, the USB card reader.

Try full formatting the SD card, prior to DietPi image write, eg:
```
dd if=/dev/zero of/path/to/sd
```

Is the serial log from 1st run? Seems to be missing the post kernel info and FS expansion during 1st run? If we can get the full serial log, this may provide more info on the cause.

Either way, i'll run a test here to try and replicate the issue you experienced.

Fourdee, 2018-02-09:
Tested fine locally:
🈯️ Pine A64

kgunnar, 2018-02-09:
That original serial console log was likely from a subsequent boot. I was wiping the partitions off between flash attempts but this time I ran `sudo dd if=/dev/zero of=/dev/sdc` to zero fill the SD card, then ran `sudo dd if=DietPi_v6.0_PineA64-ARMv8-Stretch.img of=/dev/sdc bs=1M status=progress` to reflash the image file.

I can flash this exact microSD card with other images (e.g. xenial-pine64-bspkernel-20161218-1.img for Xenial Base Image [20161218-1] by longsleep) and this A64+ board boots without issue. Is there any difference between the A64 you tested and this A64-DB-2G-Rev B (2016-02-25) that might cause an issue?

Here is the most recent log file, and this is on the very first boot after flashing the DietPi image:
```
HELLO! BOOT0 is starting!
boot0 commit : 045061a8bb2580cb3fa02e301f52a015040c158f
 
boot0 version : 4.0.0
set pll start
set pll end
rtc[0] value = 0x00000000
rtc[1] value = 0x00000000
rtc[2] value = 0x00000000
rtc[3] value = 0x00000000
rtc[4] value = 0x00000000
rtc[5] value = 0x00000000
DRAM driver version: V1.1
rsb_send_initseq: rsb clk 400Khz -> 3Mhz
PMU: AXP81X
ddr voltage = 1500 mv
DRAM Type = 3 (2:DDR2,3:DDR3,6:LPDDR2,7:LPDDR3)
DRAM clk = 672 MHz
DRAM zq value: 003b3bbb
DRAM single rank full DQ OK
DRAM size = 2048 MB
DRAM init ok
dram size =2048
card boot number = 0, boot0 copy = 0
card no is 0
sdcard 0 line count 4
[mmc]: mmc driver ver 2015-05-08 20:06
[mmc]: sdc0 spd mode error, 2
[mmc]: Wrong media type 0x00000000
[mmc]: ***Try SD card 0***
[mmc]: HSSDR52/SDR25 4 bit
[mmc]: 50000000 Hz
[mmc]: 30528 MB
[mmc]: ***SD/MMC 0 init OK!!!***
sdcard 0 init ok
The size of uboot is 000e8000.
sum=832e0b41
src_sum=832e0b41
Succeed in loading uboot from sdmmc flash.
boot0: start load other image
boot0: Loading BL3-1
Loading file 0 at address 0x40000000,size 0x00008400 success
boot0: Loading scp
Loading file 2 at address 0x00040000,size 0x00019c00 success
set arisc reset to de-assert state
Ready to disable icache.
Jump to secend Boot.

U-Boot 2014.07-6-pine64-longsleep (Jun 18 2016 - 16:15:13) Allwinner Technology 

uboot commit : 55c9c8c8ac005b1c00ac948386c60c4a741ebaa9
 
rsb: secure monitor exist
[      0.334]pmbus:   ready
[      0.337][ARISC] :arisc initialize
[      0.665][ARISC] :arisc_dvfs_cfg_vf_table: support only one vf_table
[      0.780][ARISC] :sunxi-arisc driver startup succeeded
[      0.813]PMU: AXP81X
[      0.816]PMU: AXP81X found
bat_vol=189, ratio=100
[      0.822]PMU: dcdc2 1100
[      0.825]PMU: cpux 1008 Mhz,AXI=336 Mhz
PLL6=600 Mhz,AHB1=200 Mhz, APB1=100Mhz AHB2=300Mhz MBus=400Mhz
device_type = 3253, onoff=1
[... 109 lines cut ...]
[mmc]: Has init
[      5.656]---drivers/mmc/mmc.c 2733 mmc_init
reading pine64/Image
11796928 bytes read in 578 ms (19.5 MiB/s)
[mmc]: Has init
[      6.245]---drivers/mmc/mmc.c 2733 mmc_init
reading initrd.img
1076776 bytes read in 56 ms (18.3 MiB/s)
Image lacks image_size field, assuming 16MiB
## Flattened Device Tree blob at 45000000
   Booting using the fdt blob at 0x45000000
   reserving fdt memory region: addr=41010000 size=10000
   reserving fdt memory region: addr=41020000 size=800
   reserving fdt memory region: addr=40100000 size=4000
   reserving fdt memory region: addr=40104000 size=1000
   reserving fdt memory region: addr=40105000 size=1000
   reserving fdt memory region: addr=45000000 size=11000
   Loading Ramdisk to b6db0000, end b6eb6e28 ... OK
   Loading Device Tree to 44feb000, end 44ffefff ... OK

Starting kernel ...

[mmc]: MMC Device 2 not found
[mmc]: mmc 2 not find, so not exit
```

kgunnar, 2018-02-10:
In case it helps at all, here is a console log from xenial flashed to the exact same SD card, running on the same (my only) A64+ with the same power supply: [pine64_xenial_9feb1917.log](https://github.com/Fourdee/DietPi/files/1712802/pine64_xenial_9feb1917.log) Up until the last line of the DietPi log, they look practically identical but xenial continues on from there whereas DietPi never seems to.

Fourdee, 2018-02-10:
@kgunnar 

Thanks 👍 

> Is there any difference between the A64 you tested and this A64-DB-2G-Rev B (2016-02-25) that might cause an issue?

Not that i'am aware of, but I will check my revision aswell.

Looks like its failing to launch kernel. Could be a kernel bug, or, a possible image corruption our end.
- Did you try full formatting the SD card prior to writing DietPi image? https://github.com/Fourdee/DietPi/issues/1495#issuecomment-364537635

Failing the above, I'll recreate our image, with latest kernel (if available) and re-upload.

Fourdee, 2018-02-10:
@kgunnar 

Image redone (https://github.com/Fourdee/DietPi/issues/1495#issuecomment-364658425), please re-download write:
http://dietpi.com/downloads/images/DietPi_PineA64-ARMv8-Stretch.7z

Fourdee, 2018-02-10:
> Is there any difference between the A64 you tested and this A64-DB-2G-Rev B (2016-02-25) that might cause an issue?

🈯️ I tested both images successfully on the following device, which is a Pine 64+ (https://www.pine64.org/?product=pine-a64-board-1gb)
```
201-A64DB110-0
2016-03-XX
V1.1
```

> I was wiping the partitions off between flash attempts but this time I ran sudo dd if=/dev/zero of=/dev/sdc to zero fill the SD card, then ran sudo dd if=DietPi_v6.0_PineA64-ARMv8-Stretch.img of=/dev/sdc bs=1M status=progress to reflash the image file.

Apologies, missed that.

kevinshane, 2018-02-11:
I have the 2G version and it's working very well
however missing the transfer root to ssd option, it kind of important cuz pine64 don't have emmc so everything is running inside a poor sd card

kgunnar, 2018-02-12:
Unfortunately it seems to hang at the same place. I did notice if I leave it sit for a while, it eventually reboots once and hangs again in the same spot. I left it running overnight and it never got any further or rebooted any more times. I zero filled the card again before flashing this new image. Should any of the numbers have changed, it looks like they all match the previous attempts.

```
HELLO! BOOT0 is starting!
boot0 commit : 045061a8bb2580cb3fa02e301f52a015040c158f
 
boot0 version : 4.0.0
set pll start
set pll end
rtc[0] value = 0x00000000
rtc[1] value = 0x00000000
rtc[2] value = 0x00000000
rtc[3] value = 0x0000a101
rtc[4] value = 0x00000000
rtc[5] value = 0x00000000
DRAM driver version: V1.1
rsb_send_initseq: rsb clk 400Khz -> 3Mhz
PMU: AXP81X
ddr voltage = 1500 mv
DRAM Type = 3 (2:DDR2,3:DDR3,6:LPDDR2,7:LPDDR3)
DRAM clk = 672 MHz
DRAM zq value: 003b3bbb
DRAM single rank full DQ OK
DRAM size = 2048 MB
DRAM init ok
dram size =2048
card boot number = 0, boot0 copy = 0
card no is 0
sdcard 0 line count 4
[mmc]: mmc driver ver 2015-05-08 20:06
[mmc]: sdc0 spd mode error, 2
[mmc]: Wrong media type 0x00000000
[mmc]: ***Try SD card 0***
[mmc]: HSSDR52/SDR25 4 bit
[mmc]: 50000000 Hz
[mmc]: 30543 MB
[mmc]: ***SD/MMC 0 init OK!!!***
sdcard 0 init ok
The size of uboot is 000e8000.
sum=832e0b41
src_sum=832e0b41
Succeed in loading uboot from sdmmc flash.
boot0: start load other image
boot0: Loading BL3-1
Loading file 0 at address 0x40000000,size 0x00008400 success
boot0: Loading scp
Loading file 2 at address 0x00040000,size 0x00019c00 success
set arisc reset to de-assert state
Ready to disable icache.
Jump to secend Boot.

U-Boot 2014.07-6-pine64-longsleep (Jun 18 2016 - 16:15:13) Allwinner Technology 

uboot commit : 55c9c8c8ac005b1c00ac948386c60c4a741ebaa9
 
rsb: secure monitor exist
[      0.334]pmbus:   ready
[      0.337][ARISC] :arisc initialize
[      0.664][ARISC] :arisc_dvfs_cfg_vf_table: support only one vf_table
[      0.778][ARISC] :sunxi-arisc driver startup succeeded
[      0.811]PMU: AXP81X
[      0.813]PMU: AXP81X found
bat_vol=122, ratio=100
[      0.820]PMU: dcdc2 1100
[      0.823]PMU: cpux 1008 Mhz,AXI=336 Mhz
PLL6=600 Mhz,AHB1=200 Mhz, APB1=100Mhz AHB2=300Mhz MBus=400Mhz
device_type = 3253, onoff=1
dcdc1_vol = 3300, onoff=1
dcdc2_vol = 1100, onoff=1
dcdc6_vol = 1100, onoff=1
[... 303 lines cut ...]
[mmc]: Has init
[      5.160]---drivers/mmc/mmc.c 2733 mmc_init
reading pine64/Image
11796928 bytes read in 579 ms (19.4 MiB/s)
[mmc]: Has init
[      5.750]---drivers/mmc/mmc.c 2733 mmc_init
reading initrd.img
1076776 bytes read in 57 ms (18 MiB/s)
Image lacks image_size field, assuming 16MiB
## Flattened Device Tree blob at 45000000
   Booting using the fdt blob at 0x45000000
   reserving fdt memory region: addr=41010000 size=10000
   reserving fdt memory region: addr=41020000 size=800
   reserving fdt memory region: addr=40100000 size=4000
   reserving fdt memory region: addr=40104000 size=1000
   reserving fdt memory region: addr=40105000 size=1000
   reserving fdt memory region: addr=45000000 size=11000
   Loading Ramdisk to b6db0000, end b6eb6e28 ... OK
   Loading Device Tree to 44feb000, end 44ffefff ... OK

Starting kernel ...

[mmc]: MMC Device 2 not found
[mmc]: mmc 2 not find, so not exit
```

kgunnar, 2018-02-13:
I just tried flashing the same image onto the same micro SD card using Rufus on a Windows machine and it booted fine. I'm not sure why dd and mintstick both aren't working on my Linux box (especially considering other images flash fine!) I even used the same micro SD to USB adapter. Odd, but it seems to work now... thanks for looking into it.

closed 2018-02-13 by kgunnar

Thram, 2018-04-01:
Couldn't make it work flashing the microSD on Mac. Works for version v145 though