MichaIng/DietPi #1498: Couchpotato is not installing

kodirfan, 2018-02-10:
## Creating a bug report/issue:
Dietpi version 6, RPi model 3. Power supply 5v 2A, Sandisk ultra

got the following error:

 DietPi-Software: G_AGI: build-essential make autoconf automake
                                                                            │ --no-install-recommends
                                                                            │  - exit_code = 100
                                                                            │  - VERSION:v6.1 | HW_MODEL:3 | HW_ARCH:2 | DISTRO:4
                                                                            │
                                                                            │ Log file contents:
                                                                            │ E: Failed to fetch
                                                                            │ https://www.mirrorservice.org/sites/archive.raspbian.org/raspbian/pool/main/p/pat
                                                                            │ ch/patch_2.7.5-1_armhf.deb  Could not resolve host: www.mirrorservice.org
                                                                            │ E: Failed to fetch
                                                                            │ https://www.mirrorservice.org/sites/archive.raspbian.org/raspbian/pool/main/libs/
                                                                            │ libsigsegv/libsigsegv2_2.10-5_armhf.deb  Could not resolve host:
                                                                            │ www.mirrorservice.org
                                                                            │ E: Failed to fetch
                                                                            │ https://www.mirrorservice.org/sites/archive.raspbian.org/raspbian/pool/main/m/m4/
                                                                            │ m4_1.4.18-1_armhf.deb  Could not resolve host: www.mirrorservice.org

Fourdee, 2018-02-10:
@kodirfan 

Thanks for the report 👍 

Appears to be a connection issue local to your system and/or network. Please try rebooting the device and trying again.

If problems persist, check the network settings in ```dietpi-config```.

msongz, 2018-02-10:
Ip      = 192.168.1.18                                
 Mask    = 255.255.255.0                  
DNS     = 192.168.1.1           
 similar problem with the network setting
I had a fresh install , and telling me no connection

Fourdee, 2018-02-10:
@msongz 

Thanks.

Is this using the onboard WiFi and DHCP?

msongz, 2018-02-10:
@Fourdee does " onboard WIFI " means build-in WIFI? 
I use the built-in WIFI, and DHCP. WIFI connection is OK, but network's fail
if I install the raspian-lite.img, WIFI connection and network connection both works.

Fourdee, 2018-02-10:
@msongz 

Yep, onboard = built in.

I'll run a test here, see if we can replicate.

kodirfan, 2018-02-10:
Thanks. I had to reinstall. Somehow it was not able to connect internet. I had pihole & openvpn installed. So when dietpi try to install any software it stops all the services like piHole. Is it possible if my router was configured to use pihole as a DNS server, and as this service is stopped, dietpi wont get network. 
But I tried changing router DNS address back to default but still dietpi could not get network. Any idea whats missing? I want to use one RPi3 for sonarr/pihole/pivpn. Thanks for this wonderful OS.

Fourdee, 2018-02-10:
@kodirfan 

> But I tried changing router DNS address back to default but still dietpi could not get network. Any idea whats missing?

If running DHCP on the SBC, refresh the network to re-obtain DNS table from router (eg: reboot system). If running STATIC, you'll need to change the DNS on the SBC to something like "Google", via ```dietpi-config```

> I had pihole & openvpn installed. So when dietpi try to install any software it stops all the services like piHole.

We'll take a look, thanks for the info 👍 I believe additional manual configuration is needed for OpenVPN + PiHole combination:
https://github.com/Fourdee/DietPi/issues/1245

> Is it possible if my router was configured to use pihole as a DNS server, and as this service is stopped, dietpi wont get network.

I believe so.

Fourdee, 2018-02-10:
@kodirfan 

> I use the built-in WIFI, and DHCP. WIFI connection is OK, but network's fail

Please can you try:
- ```dietpi-config``` > Networking Options: Adapters > WiFi
- Select Change mode (to STATIC)
- Select Copy address
- Select DNS , then Google
- Save changes and restart networking

Please let us know if the problem persists after this change.

msongz, 2018-02-11:
another day, it works without any modifications.
networks go fine.

msongz, 2018-02-11:
maybe static IP will not work, DHCP works fine.

closed 2018-02-12 by Fourdee