MichaIng/DietPi #1499: qBittorrent login error

tnnuyri, 2018-02-10:
I have the same problem! and already tried to change how and said ... I have the XU4 odroid, but already on the raspberry pi 3 does not happen!! Does anyone know how to solve the problem?

Fourdee, 2018-02-10:
@tnnuyri 

thanks for the report 👍 
Looks limited to Debian Jessie. We'll need to investigate, leave it with us.

Fourdee, 2018-02-10:
Notes:
```
WebUI\Password_ha1=@ByteArray(90c322f4dcabce6e66e8ea8aff218775)
```

UnfairGamer, 2018-02-10:
I posted an issue with this problem around a month ago. I ended up just switching to using Transmission instead of qBittorrent, and i actually think that transmission is better. Worked flawlessly for me, nice and functional UI and it works with external drives.

tnnuyri, 2018-02-10:
Thanks for the answer! 
n my point of view I think the qBittorrent better !!
I've used it on my raspberry pi 3

Thank you for the excellent dietpi project

symbios24, 2018-02-11:
UnfairGamer how did you mound external drives to Transmission? 
you just plug the drive and it founds it right away?

UnfairGamer, 2018-02-11:
symbios, i thought it had detected it myself, but when i checked now, i have the fstab auto-mounting the disk to a folder. So i just have it saved in that "folder" that is actually the disk. Search linux fstab mount on boot

Fourdee, 2018-02-11:
Seems Jessie version does not like/support:
```
WebUI\Password_ha1=@ByteArray(xxxx)
```
https://github.com/qbittorrent/qBittorrent/wiki/Web-UI-password-locked-on-qBittorrent-NO-X-%28qbittorrent-nox%29
even secret does not work, only removing the line and using ```adminadmin``` is functional.

Fourdee, 2018-02-11:
### Workaround

```
sed -i '/Password_ha1=/d' "$HOME"/.config/qBittorrent/qBittorrent.conf
dietpi-services restart
```

Password is now ```adminadmin```

Fourdee, 2018-02-11:
Resolved for v6.2, online doc updated to reflect password change for Jessie:
http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=2272#p2272

Install verified.

Completed.

closed 2018-02-11 by Fourdee

tnnuyri, 2018-02-11:
thank you! Problem solved ... after and just enter and put the password that you want