MichaIng/DietPi #1500: Dietpi in Proxmox 5.1 VE hangs on diskboot

eogextreme, 2018-02-10:
Using your latest .iso:  DietPi_v6.0_NativePC-EFI-x86_64-Stretch.iso Dietpi still hangs after the installation as VM has finished . The message says that it cannot find the BIOS irmware and then reboots and hangs "Booting from harddisk..."

I am using an Intel NUC 6CAYH (Celeron J3445 4 cores) 8GB RAM- 240GB SSD.

When uploaded the iso into Proxmox in created the setup with:  Balloon 512 - Memory 1024 - 32GB SCSI HDD - Ethernet Intel e1000.

I guess i would need a KVM/LXC/AMD64 capable distro.
https://www.proxmox.com/en/proxmox-ve

Fourdee, 2018-02-22:
Marking as closed.

We do not support Proxmox at this time. Alternatives are VMware and Virtual box image.

closed 2018-02-22 by Fourdee

cuph62, 2018-03-02:
I used the vmdk of the Vmware succesfully,
after converting the vmdk  to qcow2, this worked flawlessly in Proxmox

eogextreme, 2018-03-05:
Thanks cuph62,

I will try to convert it, though it would be nice if @Fourdee could make the qcow2 available on the Dietpi downloads section <hint> :)

jodumont, 2019-05-03:
@n1ete try this : https://github.com/MichaIng/DietPi/issues/1171#issuecomment-489066602

jodumont, 2019-06-04:
@n1ete or

on your proxmox : `qemu-img convert -f vmdk IMAGE_TO_CONVERT.vmdk -O qcow2 IMAGE_CONVERTED.qcow2`

ref: https://blog.lbdg.me/proxmox-convert-vmdk-to-qcow2/