MichaIng/DietPi #1502: Unable to locate package nzbdrone

Barishhh, 2018-02-10:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | v6.1
- SBC Device (EG: RPi 3)? Odroid C2
- Power supply used (EG: 5v 1A RAVpower)? 5V 3A charger
- SD card used (EG: Sandisk ultra)? Sandisk SD 256GB
- Distro (EG: Jessie) | ```uname -a```? 

#### Additional Information (if applicable):
- Software title? Nzbdrone
- Can this issue be replicated on a fresh installation of DietPi? Affirm.

#### Expected behaviour:
Trying to install Sonarr, but failing.

#### Actual behaviour:
Throws an error at the dietpi-software installation screen while installing Sonarr. 
 
DietPi-Software: G_AGI: nzbdrone
                 │  - exit_code = 100
                 │  - VERSION:v6.1 | HW_MODEL:12 | HW_ARCH:3 | DISTRO:3
                 │
                 │ Log file contents:
                 │ E: Unable to locate package nzbdrone
                 │
                 │ If problems persist, please report this to DietPi for investigation, including a
                 │ screenshot of this error! (https://github.com/Fourdee/DietPi/issues).

#### Steps to reproduce:
Install sonarr yourself using dietpi-software 

#### Did you submit a ```dietpi-bugreport```?
d80ba514-469e-4fff-a964-b8492b0f8757-0

#### Extra details:

Barishhh, 2018-02-10:
Tried to add `dpkg --add-architecture armhf `after a reinstallation as suggested in https://github.com/Fourdee/DietPi/issues/1277. Didn't solve the problem and still returns a exit code 100.

Fourdee, 2018-02-11:
@Barishhh 

Thanks for the report 👍 

Please can you confirm which software title?
```
root@DietPi:~# dietpi-software list | grep nzb
id 139 | =0 | sabnzbd: nzb download manager | +buildessential | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=6747#p6747
id 149 | =0 | nzbget: nzb download manager | | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=7575#p7575
```

Fourdee, 2018-02-11:
Yep, seems armhf is being ignored.

Barishhh, 2018-02-11:
Let me know if I can help in any way!

Gabba35, 2018-03-20:
Am also having this issue (exit code:100, version 6.4). Thanks for working on it Fourdee.

Fourdee, 2018-04-04:
Resolved for v6.7.

closed 2018-04-04 by Fourdee

Barishhh, 2018-04-04:
👍