MichaIng/DietPi #1503: DietPi-software | MariaDB fails to install or move on external disk

FabioLolix, 2018-02-11:
#### Required Information:
- DietPi Version  | v6.1
- SBC Device     | RPi 3 Model B (armv7l) (index=3)
- Distro               | stretch (index=4)
- Command        | systemctl restart mysql
- Error Handler   | G_ERROR_HANDLER
- SD card           | Sandisk ultra 16GB 
- Power supply   | Nextcloud box's Ktec 5v 3A with Western digital 1TB disk

#### Additional Information (if applicable):
- Software title | DietPi-Software
Issue happen on fresh installation as well with already installed Mariadb (Nextcloud & Ampache)

#### Expected behaviour:
MariaDB should be restarting

#### Actual behaviour:
MariaDB fails to restart, don't result in installed programs and a new installation fails because files already exist on filesystem

#### Steps to reproduce:
Move dietpi_userdata to external disk using dietpi-software
Install Nextcloud

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:
> [Warning] Can't create test file /var/lib/mysql/DietPi.lower-test

> Feb 11 10:44:11 DietPi systemd[1]: Starting MariaDB database server...
Feb 11 10:44:12 DietPi mysqld[26781]: 2018-02-11 10:44:12 1996279808 [Note] /usr/sbin/mysqld (mysqld 10.1.23-MariaDB-9+deb9u1) starting as process 26781 ...
Feb 11 10:44:12 DietPi mysqld[26781]: 2018-02-11 10:44:12 1996279808 [Warning] Can't create test file /var/lib/mysql/DietPi.lower-test
Feb 11 10:44:12 DietPi mysqld[26781]: [90B blob data]
Feb 11 10:44:12 DietPi mysqld[26781]: 2018-02-11 10:44:12 1996279808 [ERROR] Aborting
Feb 11 10:44:12 DietPi systemd[1]: mariadb.service: Main process exited, code=exited, status=1/FAILURE
Feb 11 10:44:12 DietPi systemd[1]: Failed to start MariaDB database server.
Feb 11 10:44:12 DietPi systemd[1]: mariadb.service: Unit entered failed state.
Feb 11 10:44:12 DietPi systemd[1]: mariadb.service: Failed with result 'exit-code'.

Maybe https://github.com/Fourdee/DietPi/pull/1475 can help?

First found out after installing programs and then moving the dietpi_folder
Then with fresh install moving the dietpi_folder first then installing Nextcloud

MichaIng, 2018-02-11:
@FabioLolix 
Thanks for report!
Hmm #1475 might or might not fix this, as I changed the installation routine quite a bid. But it was actually not the intention.
First of all I am wondering why in your case `mariadb.service` is started, where the command clearly says `systemctl restart mysql`, which should call `/etc/init.d/mysql` (for compatibility reasons, as long as we support Jessie, where MariaDB does not yet come with systemd mariadb.service).

Do you have a change to test/check, before moving data directory:
```
systemctl status mysql
systemctl status mariadb
```
and if you see this during installation:
```
Setting up mariadb-server-10.1 (10.1.26-0+deb9u1) ...
mariadb.service is a disabled or a static unit, not starting it.
```
Maybe we need to actively disable/mask `mariadb.service` to cover cases where it is starting via mysql, by manual adjustment/symlink or distro preset.

But I am still wondering, as on my own RPi (2 Model B), I actively adjusted it to use systemd `mariadb.service` already since longer time with dietpi_userdata moved to external drive as well. This was on Stretch as well and now on Buster, working like a charm.

I made some research about that error messages you face: https://stackoverflow.com/questions/38529205/mariadb-cannot-start-after-update-warning-cant-create-test-file-home-mysql
```
# Prevent accessing /home, /root and /run/user
ProtectHome=true
```
Do you moved one of these locations also onto the external drive and dietpi_userdata inside? Maybe the solution to set `ProtectHome=false` works for you as well.

FabioLolix, 2018-02-11:
> Do you have a chance to test/check, before moving data directory:
Sure, I'll make some fresh install and tests tomorrow or in the next days

> Did you do some manual adjustments about that services, symlinks or something?
> Do you moved one of these locations also onto the external drive and dietpi_userdata inside? 
No, I haven't touched anything aside moving dietpi_userdata with diepti-software

MichaIng, 2018-02-12:
Also made a fresh install and test on VM Stretch:
```
Setting up mariadb-server-10.1 (10.1.26-0+deb9u1) ...
Created symlink /etc/systemd/system/mysql.service → /lib/systemd/system/mariadb.service.
Created symlink /etc/systemd/system/mysqld.service → /lib/systemd/system/mariadb.service.
Created symlink /etc/systemd/system/multi-user.target.wants/mariadb.service → /lib/systemd/system/mariadb.service.
Setting up mariadb-server (10.1.26-0+deb9u1) ...
```
Symlinks get created indeed, but after dietpi-software has finished, they were gone:
```
root@DietPi:~# ls -l /etc/systemd/system/
total 64
lrwxrwxrwx 1 root root    9 Feb 12 00:19 apt-daily.service -> /dev/null
lrwxrwxrwx 1 root root    9 Feb 12 00:19 apt-daily.timer -> /dev/null
lrwxrwxrwx 1 root root    9 Feb 12 00:19 apt-daily-upgrade.service -> /dev/null
lrwxrwxrwx 1 root root    9 Feb 12 00:19 apt-daily-upgrade.timer -> /dev/null
-rw-r--r-- 1 root root  316 Feb 12 00:19 dietpi-boot.service
-rw-r--r-- 1 root root  332 Feb 12 00:19 dietpi-fs_expand.service
-rw-r--r-- 1 root root  289 Feb 12 00:19 dietpi-fs_partition_resize.service
-rw-r--r-- 1 root root  384 Feb 12 00:19 dietpi-ramdisk.service
-rw-r--r-- 1 root root  379 Feb 12 00:19 dietpi-ramlog.service
drwxr-xr-x 2 root root 4096 Feb 11 23:54 getty.target.wants
drwxr-xr-x 2 root root 4096 Feb 12 00:19 halt.target.wants
-rw-r--r-- 1 root root  314 Feb 12 00:19 kill-ssh-user-sessions-before-network.service
drwxr-xr-x 2 root root 4096 Feb 12 00:19 local-fs.target.wants
drwxr-xr-x 2 root root 4096 Feb 12 00:34 multi-user.target.wants
drwxr-xr-x 2 root root 4096 Feb 12 00:19 network-online.target.wants
drwxr-xr-x 2 root root 4096 Feb 12 00:19 poweroff.target.wants
-rw-r--r-- 1 root root  292 Feb 12 00:19 rc-local.service
drwxr-xr-x 2 root root 4096 Feb 12 00:19 reboot.target.wants
drwxr-xr-x 2 root root 4096 Feb 12 00:19 sysinit.target.wants
drwxr-xr-x 2 root root 4096 Feb 12 00:33 timers.target.wants
```
I am bid confused about that, can't find anything where we touch `mariadb.service`, we just disable+stop+start mysql, which is then definitely the init.d service.
But yeah might be possible, that during installation itself, `mariadb.service` is starting and then failing for some reason.

Or it is just some permissions issue:
@FabioLolix 
Did you test, if permissions on your external drive were actually set correct, especially for `ls -l /mnt/dietpi_userdata/mysql`?
We check, if the target file system supports permissions, but we do not break transfer it is doesn't: https://github.com/Fourdee/DietPi/blob/master/dietpi/func/dietpi-set_userdata#L73-L82
@Fourdee 
This should be changed. Many software titles will break, if dietpi_userdata does not support permissions. Also we could consider to switch to `rsync`? In case we could add it as permanent dependency, not just for dietpi-backup/sync 😄.

FabioLolix, 2018-03-19:
Fresh install version 6.4

Installed nextcloud, systemctl status for mysql and  mariadb

**mariadb**
> \# systemctl status mariadb
● mariadb.service - MariaDB database server
   Loaded: loaded (/lib/systemd/system/mariadb.service; disabled; vendor preset: enabled)
   Active: inactive (dead)
root@DietPi:~#

**mysql**
> \# systemctl status mysql
● mysql.service - LSB: Start and stop the mysql database server daemon
   Loaded: loaded (/etc/init.d/mysql; generated; vendor preset: enabled)
   Active: active (running) since Mon 2018-03-19 19:46:38 GMT; 9min ago
     Docs: man:systemd-sysv-generator(8)
  Process: 1195 ExecStart=/etc/init.d/mysql start (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/mysql.service
           ├─1226 /bin/bash /usr/bin/mysqld_safe
           ├─1382 /usr/sbin/mysqld --basedir=/usr --datadir=/var/lib/mysql --plugin-dir=/usr/lib/arm-linux-gnueabihf/mariadb18/plugin --user=mysql --skip-log-error --pid-file=/var/run/mysqld
/mysqld.pid --socket=/var/run/mysqld/mysqld.sock --port=3306
           └─1383 logger -t mysqld -p daemon error
>
> Mar 19 19:46:39 DietPi /etc/mysql/debian-start[1458]: nextcloud.oc_flow_operations                       OK
Mar 19 19:46:39 DietPi /etc/mysql/debian-start[1458]: nextcloud.oc_group_admin                           OK
Mar 19 19:46:39 DietPi /etc/mysql/debian-start[1458]: nextcloud.oc_group_user                            OK
Mar 19 19:46:39 DietPi /etc/mysql/debian-start[1458]: nextcloud.oc_groups                                OK
Mar 19 19:46:39 DietPi /etc/mysql/debian-start[1458]: nextcloud.oc_jobs                                  OK
Mar 19 19:46:39 DietPi /etc/mysql/debian-start[1458]: nextcloud.oc_migrations                            OK
Mar 19 19:46:39 DietPi /etc/mysql/debian-start[1458]: nextcloud.oc_mimetypes                             OK
Mar 19 19:46:39 DietPi /etc/mysql/debian-start[1458]: nextcloud.oc_mounts                                OK
Mar 19 19:46:39 DietPi /etc/mysql/debian-start[1458]: nextcloud.oc_notifications                         OK
Mar 19 19:46:39 DietPi /etc/mysql/debian-start[1458]: nextcloud.oc_notifications_pushtokens              OK
root@DietPi:~# 

Moving the dietpi_userdata folder hang for a while on stopping mysql, failing it, then the procedure continue
How can I check what was going with stopping mysql after the procedure?

> DietPi-Services: ncc maintenance:mode --off                                         
│  - exit_code = 1                                                                    
│  - VERSION:v6.4 | HW_MODEL:3 | HW_ARCH:2 | DISTRO:4                                 
│                                                                                     
│ Log file contents:                                                                  
│ # 21 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(113):        
│ OC\Server->OC\{closure}(Object(OC\Server))                                          
│ # 22 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(116):   
│ Pimple\Container->offsetGet('OC\\Authenticati...')                                  
│ # 23 /var/www/nextcloud/lib/private/ServerContainer.php(132):                        
│ OC\AppFramework\Utility\SimpleContainer->query('OC\\Authenticati...')               
│ # 24 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(164):   
│ OC\ServerContainer->query('OC\\Authenticati...')                                    
│ # 25 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(109):        
│ OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}(Object   
│ (OC\Server))   
│ # 26 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(116):   
│ Pimple\Container->offsetGet('OC\\Authenticati...')                                  
│ # 27 /var/www/nextcloud/lib/private/ServerContainer.php(132):                        
│ OC\AppFramework\Utility\SimpleContainer->query('OC\\Authenticati...')             
│ # 28 /var/www/nextcloud/lib/private/Server.php(352):                                 
│ OC\ServerContainer->query('OC\\Authenticati...')                                    
│ # 29 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(113):        
│ OC\Server->OC\{closure}(Object(OC\Server))                                          
│ # 30 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(116):   
│ Pimple\Container->offsetGet('OCP\\IUserSessio...')                                  
│ # 31 /var/www/nextcloud/lib/private/ServerContainer.php(132):                        
│ OC\AppFramework\Utility\SimpleContainer->query('OCP\\IUserSessio...')  
│ # 32 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(164):   
│ OC\ServerContainer->query('OCP\\IUserSessio...')                                    
│ # 33 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(109):        
[... 93 lines cut ...]
Mar 19 19:59:38 DietPi /etc/init.d/mysql[4897]: [61B blob data]
Mar 19 19:59:38 DietPi /etc/init.d/mysql[4897]: error: 'Can't connect to local MySQL server through socket '/var/run/mysqld/mysqld.sock' (2 "No such file or directory")'
Mar 19 19:59:38 DietPi /etc/init.d/mysql[4897]: Check that mysqld is running and that the socket: '/var/run/mysqld/mysqld.sock' exists!
Mar 19 19:59:38 DietPi /etc/init.d/mysql[4897]: 
[... cut ...]
-rwxrwx--- 1 mysql mysql        0 Mar 19 19:42 debian-10.1.flag
-rwxrwx--- 1 mysql mysql 79691776 Mar 19 19:59 ibdata1
-rwxrwx--- 1 mysql mysql 50331648 Mar 19 19:59 ib_logfile0
-rwxrwx--- 1 mysql mysql 50331648 Mar 19 19:42 ib_logfile1
-rwxrwx--- 1 mysql mysql        0 Mar 19 19:42 multi-master.info
drwxrwx--- 2 mysql mysql     4096 Mar 19 19:46 mysql
-rwxrwx--- 1 mysql mysql       15 Mar 19 19:46 mysql_upgrade_info
drwxrwx--- 2 mysql mysql     4096 Mar 19 19:46 nextcloud
drwxrwx--- 2 mysql mysql     4096 Mar 19 19:46 performance_schema

MichaIng, 2018-03-19:
@FabioLolix 
Note: You should really move your userdata first and install software afterwards, just to not need to move "sensible" data like databases and larger amount of data.

But on the other hand, as all our software titles can still access their data on the same path /mnt/dietpi_userdata/ which then symlinks to the new location, it should work.

Let's go through it:
- mariadb.service disabled as it should, mysql.service active and running instead 👍.
- `Mar 19 19:46:39 DietPi /etc/mysql/debian-start[1458]: nextcloud.oc_flow_operations OK` Did you do some mysqlcheck? But good since Nextcloud tables seem OK, as they should. 👍
- Did Nextcloud install without issues and work before moving the datafolder? I assume so since `ncc maintenance:mode --on` at beginning of dietpi-services stop seems to be without error. 👍
- `ncc maintenance:mode --off` Okay disabling maintenance mode of course fails, if mysql failed before. No good practice from us to continue with userdata moving, if services didn't stop gracefully. We need to stop process in case and prompt information to user, that fixing all DietPi controlled services is necessary before we can guarantee secure moving of userdata. Not good if some services stay possible in accessing those files. `ncc maintenance:mode` command is handled separately via `G_RUN_CMD` where error handler successfully interrupts. 👎
- `drwxrwx--- 2 mysql mysql 4096 Mar 19 19:46 mysql` External drive supports permissions 👍.

Debugging:
- Follow mysql folder links: `ls -al /var/lib/mysql && ls -al /mnt/dietpi_userdata`
- Check whole mysql journal log:
  - `journalctl -t mysqld` at least shows logs pre failure
  - `journalctl -t mysql` or `journalctl -u mysql` to show errors possibly without active logger
  - `/var/log/mysql/error.log` or `/var/log/mysql.err` (I fully switched to journald, thus don't know which is default here 😉) should not show more, but check just in case.
  - In case do `systemctl restart mysql` to force fresh error log entries.

Fourdee, 2018-04-05:
No response, marking as closed. Please reopen if required.

closed 2018-04-05 by Fourdee