MichaIng/DietPi #1504: ompd missing user in database

shahwahed, 2018-02-11:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version 6.1
- SBC Device RPi2
- Power supply used (5v 3A )?
- SD card used 
- Distro dietpi rasbian

#### Additional Information (if applicable):
- Software title : ompd
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Expected behaviour:

ompd should have create a ompd database
#### Actual behaviour:
database is not create

#### Steps to reproduce:
fresh install, install mpd + ompd

#### Did you submit a ```dietpi-bugreport```?
no

#### Extra details:
to fix this issue, create a ompd user and give permission to ompd database (no need to create it, first access to ompd will create the database)

CREATE USER 'ompd'@'127.0.0.1' IDENTIFIED BY 'yourpassword';
GRANT ALL PRIVILEGES ON ompd.* TO 'ompd'@'127.0.0.1';

yourpassword = password specified in /var/www/ompd/include/config.inc.php ==> $cfg['mysqli_password'] 

may be add a glitch when you install the package to ask for user and create it?

MichaIng, 2018-02-11:
@shahwahed 
Thanks for report!
Actually we do create the user and database: https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L11301-L11314
`GRANT ALL PRIVILEGES ON ompd.* TO 'ompd'@'localhost';` does actually include creation of user and database, that's true. We just split it in our script, to allow also creating databases for root user without (accidentally) touching it's authentication method (unix_socket): https://github.com/Fourdee/DietPi/blob/master/dietpi/func/create_mysql_db

Did you face any other error message or warning during dietpi-software run or do you find messages in `journalctl` or `/var/log/mysql/error.log` from time of installation attempt that could be related?

shahwahed, 2018-02-11:
mysql error.log is empty

MichaIng, 2018-02-12:
@shahwahed 
Just tested again on VM, fresh installation, v6.1 Stretch: 🈯️ 
```
 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Configuring O!MPD: feature-rich, web interface audio player for mpd

[ INFO ] Creating MariaDB database: ompd
[  OK  ] Creating MariaDB database: ompd | Completed
```
```
MariaDB [(none)]> show databases;
+--------------------+
| Database           |
+--------------------+
| information_schema |
| mysql              |
| ompd               |
| performance_schema |
+--------------------+
4 rows in set (0.00 sec)

MariaDB [(none)]> select user, password, host, plugin from mysql.user;
+------+-------------------------------------------+-----------+-------------+
| user | password                                  | host      | plugin      |
+------+-------------------------------------------+-----------+-------------+
| root |                                           | localhost | unix_socket |
| ompd | *EE072F152A689751C0E898D21E501CA02DBC6115 | localhost |             |
+------+-------------------------------------------+-----------+-------------+
```
@Fourdee @k-plan 
Can some replicate error on RPi?

Fourdee, 2018-02-12:
🈯️ RPi 3 (v6.2):
```
DietPi-Software
─────────────────────────────────────────────────────
 Mode: Configuring O!MPD: feature-rich, web interface audio player for mpd

[ INFO ] DietPi-Create_MySQL_DB | Creating MariaDB database: ompd
[  OK  ] Creating MariaDB database: ompd | Completed

###

MariaDB [(none)]> show databases;
+--------------------+
| Database           |
+--------------------+
| information_schema |
| mysql              |
| ompd               |
| performance_schema |
+--------------------+
4 rows in set (0.01 sec)

MariaDB [(none)]> select user, password, host, plugin from mysql.user;
+------+-------------------------------------------+-----------+-------------+
| user | password                                  | host      | plugin      |
+------+-------------------------------------------+-----------+-------------+
| root |                                           | localhost | unix_socket |
| ompd | *EE072F152A689751C0E898D21E501CA02DBC6115 | localhost |             |
+------+-------------------------------------------+-----------+-------------+
2 rows in set (0.00 sec)

```

Unable to replicate.

MichaIng, 2018-02-12:
@shahwahed 
Did you do any 1st run setup via dietpi.txt, or moving dietpi_userdata to external drive?

Fourdee, 2018-02-22:
Marking as closed, unable to replicate.

closed 2018-02-22 by Fourdee