MichaIng/DietPi #1505: DietPi-Images | Creation procedure

MichaIng, 2018-02-12:
I think it might make sense to list and discuss the steps, in how to usually create a DietPi image. For most SBCs it should be self-explained, based on the ready to run base image we use (e.g. Raspbian Lite, Meverics Odroid minimal image, ARMbian), but e.g. for x86 images it makes sense to align the procedure, starting with the official Debian netboot/mini.iso and in case VM specs:

1. For VMs:
   - VirtualBox:
     - Create new machine: Name e.g. `DietPi_VirtualBox-x86_64-Stretch`, `Linux`, `Debian (64-bit)`
     - `Memory size: 1024 MB` as default for our images, or as desired.
     - Create a new virtual hard disk now
     - `VMDK` disk format for better compatibility
     - `Dynamically allocated` to reduce initial disk size
     - For Stretch+ I used `12 GB disk size` for our images, or as desired. It is anyway just an upper limit in combination with dynamic allocation.
     - Select new machine and `Settings`
     - `Deselect Floppy`, who needs this?
     - `Pointing device: PS/2 Mouse`, which allows disabling USB
     - Assure `VT-x/AMD-V + Nested Paging` acceleration is enabled
     - `16 MB display memory`, for initial server setup
     - Disable Audio and USB as default.
     - Network: `Bridged Adapter` to allow local SSH, `Adapter type: virtio-net`, for best performance
2. Download latest Debian netboot/mini.iso: https://deb.debian.org/debian/dists/stretch/main/installer-amd64/current/images/netboot/mini.iso
3. Mount it as external drive to machine
   - VirtualBox:
     - Mount just as `optical disk` to existing optical drive.
     - Start VM
4. Debian installation (VM)
   - Advanced options > Expert install
   - `Language: en_GB.UTF-8` only
   - `Keyboard: British` as well
   - Allow just network auto detection
   - `Hostname: DietPi`
   - `Domain name: leave empty`
   - APT mirror: Just use one near to you, or deb.debian.org, will be anyway overwritten by DietPi-Prep, as actually most other settings as well 😉.
   - Download installer components: **de**select everything, continue
   - Setup users > enable shadow passwords > allow root login > password: `dietpi`
   - Do **not** create normal user account.
   - Clock: Use NTP, leave default mirror 0.debian.pool.ntp.org, timezone: `Europe/London`
   - `Detect disks` > `Partition discs` > `Guided - use entire disc` > select the one available disk > `All files in one partition` > delete all created partitions again > select free space > `create new partition` > `partition size: max` > `Primary`
   - Verify: `ext4` and mount point `/` > Done setting up the partition > Finish partitioning > `No` return due to missing swap partition > `Yes` write to disk
   - Install base system
   - `linux-image-amd64`
   - `targeted` (no generic driver installation necessary)
   - Package manager: nonfree/contrib not necessary so far, will be anyway added by DietPi-Prep
   - `No` sources!
   - Select security releases and updates, but actually on my VMs (bridged via WLAN, IPv6 available on network but not actively used) security repo mostly times out on some IPv6 connection attempt. So I deselect them. DietPi-Prep will add them but force IPv4 as default for APT.
    - Select and install software: `No` virtualbox-guest extension, `No` survey (user should decide this by oneself later)
   - `De`select all additional software
   - Install grub boot loader: `Yes` to master boot record, to only available partition, `Not` to the EFI removable media part, actually no idea why it detects EFI, as it was not selected on machine and is definitely not available afterwards...
   - Finish installation > set to UTC
   - Unplug (context menu on disc symbol at button, Remove disc, Force removal!) disk, to prevent reboot into installation again, before > Continue
5. DietPi-Prep:
   - apt-get update; apt install -y ca-certificates sudo wget dropbear-run
   - Connect via SSH (IP: `ip addr show`) for easier copy&paste etc.
```
GIT_BRANCH='master' #dev GIT_BRANCH='testing'
GIT_BRANCH=$GIT_BRANCH wget https://raw.githubusercontent.com/Fourdee/DietPi/$GIT_BRANCH/PREP_SYSTEM_FOR_DIETPI.sh -O PREP_SYSTEM_FOR_DIETPI.sh
chmod +x PREP_SYSTEM_FOR_DIETPI.sh
sudo GIT_BRANCH=$GIT_BRANCH ./PREP_SYSTEM_FOR_DIETPI.sh
```
   - Manually remove outdated kernel modules with `/lib/modules/` or purge the related package, currently `linux-image-4.9.0-4-amd64`. In latter case I so far did just in case `apt install --reinstall linux-image-4.9.0-5-amd64` current kernel package.
   - Mount virtual drive (vmdk) to other VM or machine, use `zerofree -v /dev/sdx` to reduce final image size.

Fourdee, 2018-02-12:
@MichaIng 

Nice write up 👍 Thanks for this, will help users who want to run PREP.

X3msnake, 2018-02-12:
@Fourdee @MichaIng 
If we want to build this on a 32bit architecture it throws a 
Unknown or unsupported CPU architecture i686

For what i read in the code it sets the G_HW_ARCH_DESCRIPTION=$ is there a way to add the i686?

MichaIng, 2018-02-12:
@X3msnake
Indeed right now we don't offer x86_32 support. Please read discussion here: https://github.com/Fourdee/DietPi/issues/805#issuecomment-361591310

X3msnake, 2018-02-19:
You should commit this to the dietpi wiki or docs

MichaIng, 2018-03-11:
Increased VM RAM size to 2G, as this is what we by default ensure in case via swap file.

MichaIng, 2018-03-15:
Great hint from WarHawk: http://dietpi.com/phpbb/viewtopic.php?f=9&t=2960&p=11428#p11421
- Seems possible to automate most Debian installer steps. I will have a look how to handle and integrate this exactly.

MichaIng, 2019-01-29:
Moved VirtualBox creation to Wiki: https://github.com/Fourdee/DietPi/wiki/How-to-create-a-DietPi-image-for-VirtialBox

VMWare is next...

closed 2019-01-29 by MichaIng